esp8266/Arduino #7967: ESP32 has esp_arduino_version.h now, port this to ESP8266

dok-net, 2021-04-08:
Since ESP32 v 2.0.0, `Arduino.h` includes the afore-mentioned file. Is there support in the release process yet to update `ESP_ARDUINO_VERSION_MAJOR`, `ESP_ARDUINO_VERSION_MINOR`, and `ESP_ARDUINO_VERSION_PATCH` reliably?

```
// Copyright 2019 Espressif Systems (Shanghai) PTE LTD
//
// Licensed under the Apache License, Version 2.0 (the "License");
// you may not use this file except in compliance with the License.
// You may obtain a copy of the License at
//
//     http://www.apache.org/licenses/LICENSE-2.0
//
// Unless required by applicable law or agreed to in writing, software
// distributed under the License is distributed on an "AS IS" BASIS,
// WITHOUT WARRANTIES OR CONDITIONS OF ANY KIND, either express or implied.
// See the License for the specific language governing permissions and
// limitations under the License.

#pragma once

#ifdef __cplusplus
extern "C" {
#endif

/** Major version number (X.x.x) */
#define ESP_ARDUINO_VERSION_MAJOR   2
/** Minor version number (x.X.x) */
#define ESP_ARDUINO_VERSION_MINOR   0
/** Patch version number (x.x.X) */
#define ESP_ARDUINO_VERSION_PATCH   0

/**
 * Macro to convert ARDUINO version number into an integer
 *
 * To be used in comparisons, such as ESP_ARDUINO_VERSION >= ESP_ARDUINO_VERSION_VAL(2, 0, 0)
 */
#define ESP_ARDUINO_VERSION_VAL(major, minor, patch) ((major << 16) | (minor << 8) | (patch))

/**
 * Current ARDUINO version, as an integer
 *
 * To be used in comparisons, such as ESP_ARDUINO_VERSION >= ESP_ARDUINO_VERSION_VAL(2, 0, 0)
 */
#define ESP_ARDUINO_VERSION  ESP_ARDUINO_VERSION_VAL(ESP_ARDUINO_VERSION_MAJOR, \
                                             ESP_ARDUINO_VERSION_MINOR, \
                                             ESP_ARDUINO_VERSION_PATCH)

#ifdef __cplusplus
}
#endif
```

d-a-v, 2021-04-08:
FWIW, we already have [esp8266CoreVersion](https://github.com/esp8266/Arduino/blob/da6ec83b5fdbd5b02f04cf143dcf8e158a8cfd36/cores/esp8266/core_esp8266_version.h#L48L70). Every commit is differentiated from the others.

dok-net, 2021-04-11:
What I am looking for is a solution to something like this, for example necessitated by PR #7148:
```
#if defined(HAVE_ESP_SUSPEND)
    void __esp_suspend();

    // disable CONT suspend, resume by esp_schedule pattern
    void esp_suspend()
    {
        auto self = CoopTaskBase::self();
        if (self) CoopTaskBase::yield(self);
        else __esp_suspend();
    }
#else
    void __esp_yield();
 
    // disable CONT suspend, resume by esp_schedule pattern
    void esp_yield()
    {
        auto self = CoopTaskBase::self();
        if (self) CoopTaskBase::yield(self);
        else __esp_yield();
    }
#endif
```
Naturally, this should be available for master commits, too, not only for release versions of Arduino Core for ESP8266.