esp8266/Arduino #7992: No such file or directory  "wl_definitions.h"

feviro, 2021-04-16:
----------------------------- Delete above -----------------------------

### Basic Infos

- [X] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [X] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [X] I have tested that the issue is present in current master branch (aka latest git).
- [X] I have searched the issue tracker for a similar issue.
- [X] If there is a stack dump, I have decoded it.
- [X] I have filled out all fields below.

#### Platform

- Hardware:			[ESP-12E]
- Core Version:			[1.8.13]
- Development Env:		[Arduino IDE]
- Operating System:		[Windows]

### Settings in IDE

- Module:  			[Nodemcu 1.0]
- Flash Mode:			[qio|dio|other]
- Flash Size        [4MB]
- lwip Variant:   [v2 Lower Memory]
- Reset Method:		[ck|nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz]
- Upload Using:		[SERIAL]
- Upload Speed:   [115200)

### Problem Description

No such file included:      wl_definitions.h

Compiling:
...........
C:\Users\.......\Arduino\libraries\ESP8266WiFi\ESP8266WiFi.h:28:28: fatal error: wl_definitions.h: No such file or directory
 #include <wl_definitions.h>

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp

#include <ESP8266WiFi.h>
#include <ESPAsyncWebServer.h>
#include <FS.h>
#include <WebSocketsServer.h>

#include "config.h"
#include "Websocket.hpp"
#include "Server.hpp"
#include "ESP8266_Utils.hpp"
#include "ESP8266_Utils_AWS.hpp"

void setup(void)
{
	Serial.begin(115200);
	SPIFFS.begin();
	
	ConnectWiFi_STA();

	InitWebSockets();
	InitServer();
}

void loop(void)
{
	ws.textAll(GetMillis());
}
```

### Debug Messages

```
C:\.........\Arduino\libraries\ESP8266WiFi\ESP8266WiFi.h:28:28: fatal error: wl_definitions.h: No such file or directory
 #include <wl_definitions.h>

```

feviro, 2021-04-17:
Sorry please dear David !
Obviously I'am a beginner !
I found the "external" library in:

https://github.com/arduino-libraries/WiFiLink-Library/blob/master/src/utility/wl_definitions.h

I hope this serves as a response for others like me searching solution for this !

No more than really thanks your hard job !

Fernando

earlephilhower, 2021-04-17:
Your solution is not correct.  The file `wl_definitions.h` is part of the main core and if it's not found on your system, then there is an installation problem of some sort and you should consider removing and reinstalling things.  Good luck.

closed 2021-04-17 by earlephilhower

feviro, 2021-04-18:
Yes Earl
You're true
Keep asking for more libraries
I will reinstall arduino ide
Thanks

inopya, 2023-07-08:
Yes, it could be a real error with the libraries or the IDE installation, but it is also **very possible that you have selected the wrong board** when compiling, for example that your program is for ESP32 or WEMOS and you have selected a UNO, Nano .... or similar that do not have WiFi.