esp8266/Arduino #7997: Wifi doesn't work after several deep sleep-wake cycles

manj9501, 2021-04-24:
I'm having the same issue, with my esp12f modules with 'esp8266mod' written on the shielding. The deep sleep-wake cycle works for number of times initially, but then the esp wakes up but is unable to connect to the WiFi. I tried using 47uf electrolytic caps between Vout and GND and Vin and GND but that didn't help. I'm using MCP1700. I was trying to emulate this project( https://www.instructables.com/Solar-Powered-WiFi-Weather-Station-V20/ ), but my plans have been jeopardized because 2 of my modules are displaying the same anomalous behavior. Any help would be appreciated!

TD-er, 2021-04-24:
Have you tried to call WiFi.disconnect() before making an attempt to connect?
Also call disconnect (and wait a bit using delay())  before going to deep sleep.

Tech-TX, 2021-04-26:
If you're always doing Deep Sleep and skipping the RF_CAL this may eventually happen,  I saw it after 15-20 Sleep cycles myself.   

Without an MCVE we have nothing to work with.

mhightower83, 2021-04-26:
An observation: 

To me, it looks like the MCP1700 is being asked to operate above the "Maximum Output Current: 250mA."

In the [Espressif's Hardware Design Guideline, section 1.4.1.2.](https://www.espressif.com/sites/default/files/documentation/esp8266_hardware_design_guidelines_en.pdf#page=11) the notes specify **"When using a single power supply, the recommended output current is 500 mA."** Not emphasized in their description is that the 500 mA is for  _peak_ demand (transient load) not continuous. 

While the typical transmit power consumption of the ESP8266 is documented as [170mA/140mA/120mA for operating in 802.11b/g/n mode](https://bbs.espressif.com/viewtopic.php?t=133), I never found any details in Espressif documents on the characteristics of this transient startup load. However, I did find an article at [Ondrovo.com "Determining the ESP8266 power consumption"](https://www.ondrovo.com/a/20170207-esp-consumption/), which shows around 430mA at startup. I am not sure his value had the aid of a low ESR CAPS around the ESP8266.

devyte, 2021-06-02:
Closing due to lack of response and no MCVE.

closed 2021-06-02 by devyte