esp8266/Arduino #7998: fatal error: sys/_tz_structs.h: No such file or directory

markg85, 2021-04-24:
Hi,

I'm on a new adventure to compile code for the ESP8266 with C++17.. Like, why not ;)
It's amazing that you folks made it work! Really a cool job!

I'm hitting just this one issue:
`xtensa-lx106-elf-g++ -o .pio/build/esp12e/FrameworkArduino/time.cpp.o -c -std=c++17 -fno-rtti -std=gnu++17 -fno-exceptions -fext-numeric-literals -Os -mlongcalls -mtext-section-literals -falign-functions=4 -U__STRICT_ANSI__ -ffunction-sections -fdata-sections -Wall -DPLATFORMIO=50101 -DESP8266 -DARDUINO_ARCH_ESP8266 -DARDUINO_ESP8266_ESP12 -DATOMIC_FS_UPDATE -DF_CPU=80000000L -D__ets__ -DICACHE_FLASH -DARDUINO=10805 -DARDUINO_BOARD=\"PLATFORMIO_ESP12E\" -DFLASHMODE_DIO -DLWIP_OPEN_SRC -DNONOSDK22x_190703=1 -DTCP_MSS=536 -DLWIP_FEATURES=1 -DLWIP_IPV6=0 -DVTABLES_IN_FLASH -DMMU_IRAM_SIZE=0x8000 -I.pio/build/esp12e/core -I/home/mark/.platformio/packages/framework-arduinoespressif8266/tools/sdk/include -I/home/mark/.platformio/packages/framework-arduinoespressif8266/tools/sdk/libc/xtensa-lx106-elf/include -I/home/mark/.platformio/packages/framework-arduinoespressif8266/cores/esp8266 -I/home/mark/.platformio/packages/toolchain-xtensa/include -I/home/mark/.platformio/packages/framework-arduinoespressif8266/tools/sdk/lwip2/include -I/home/mark/.platformio/packages/framework-arduinoespressif8266/varia/home/mark/.platformio/packages/framework-arduinoespressif8266/cores/esp8266/time.cpp:27:14: fatal error: sys/_tz_structs.h: No such file or directory`

It's surprisingly easy to solve though. Just commenting out `#include <sys/_tz_structs.h>` from `framework-arduinoespressif8266/cores/esp8266/time.cpp` does the job. That would be this file+line in this repo: https://github.com/esp8266/Arduino/blob/master/cores/esp8266/time.cpp#L27

In case it's handy to know, my platformio.ini file has these lines:
```
platform_packages = 
    toolchain-xtensa@~2.100100.0
    framework-arduinoespressif8266@https://github.com/esp8266/Arduino.git#457692101a7221c857a40ab1ad55aa5200099fc6
```

Cheers,
Mark

Tech-TX, 2021-05-07:
In the template you ignored there's a check box for "I have searched the issue tracker for a similar issue"

You didn't look; TD-er posted [exactly the same issue](https://github.com/esp8266/Arduino/issues/7792) in December.

earlephilhower, 2021-05-10:
Dup of #7792