esp8266/Arduino #7999: Unable to flash Wemos D1 Mini v3 on reset rst cause:2, boot mode:(3,7)

rfsch, 2021-04-25:
Hello

I am unable to connect to flash. Waiting for MAC... I tried to connect D3 to GND (manually with cable).
Do I need to solder the cable ? How do I know the Wemos is dead `? It is blinking blue when the Nodemcu tries to flash. The port is connected

Here are the errors from Arduino IDE:

This is the log from the serial monitor

17:43:44.848 ->
17:43:44.848 -> ets Jan 8 2013,rst cause:2, boot mode:(3,7)
17:43:44.848 ->
17:43:44.848 -> ets_main.c

and this is when connecting

C:\Users\RS\AppData\Local\Arduino15\packages\esp8266\tools\python3\3.7.2-post1/python3 C:\Users\RS\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4/tools/upload.py --chip esp8266 --port COM7 --baud 921600 --before default_reset --after hard_reset write_flash 0x0 C:\Users\RS\AppData\Local\Temp\arduino_build_662067/sketch_apr17b.ino.bin
esptool.py v2.8
Serial port COM7
Connecting........_____....._____....._____....._____....._____....._____....._____
Traceback (most recent call last):
File "C:\Users\RS\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4/tools/upload.py", line 65, in <module>
esptool.main(cmdline)
File "C:/Users/RS/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/tools/esptool\esptool.py", line 2890, in main
esp.connect(args.before)
File "C:/Users/RS/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/tools/esptool\esptool.py", line 483, in connect
raise FatalError('Failed to connect to %s: %s' % (self.CHIP_NAME, last_error))
esptool.FatalError: Failed to connect to ESP8266: Timed out waiting for packet header
esptool.FatalError: Failed to connect to ESP8266: Timed out waiting for packet header

The interesting part is that I have a batch of 4 Wemos.

Two good ones: reply with boot mode:(3,6)
when resetting
This one: boot mode:(3,7)
Another one: has no power at all when connecting the mini usb

I have read that 7 means SDIO v2.0 Highspeed

Thanks for any help!

Bighoneypot, 2021-04-25:
Which board have you select in boards manager?

Tech-TX, 2021-04-26:
It's a hardware error -  the ets_main.c tells me that it didn't recognize the boot mode after reset. Sounds like a cheap clone with bad soldering or wonky components, especially since other boards work. 

This is a "how do I"  question better handled on esp8266.com,  as the issue template points out.  It's not a problem with the core or included libraries.

closed 2021-05-19 by earlephilhower