esp8266/Arduino #8000: ESP8266WiFi library is giving compilation error.

angsdey2, 2021-04-25:
Hi, I am using Arduino Uno with ESP8266 module. While compiling getting below compilation error

Arduino\libraries\ESP8266WiFi\src/ESP8266WiFi.h:28:10: fatal error: wl_definitions.h: No such file or directory
 #include <wl_definitions.h>
          ^~~~~~~~~~~~~~~~~~
compilation terminated.
exit status 1
Error compiling for board Arduino Uno.

d-a-v, 2021-04-25:
Congratulation for the issue number #8000.

You have an installation issue. Remove Arduino including the "arduino15" directory and reinstall it.
I can't tell you more because you deliberately deleted the issue template which would have told us about your operating system.

closed 2021-04-25 by d-a-v

angsdey2, 2021-04-25:
@d-a-v Below is the verbose log during compilation. Can you help me with the error?

```
Arduino: 1.8.13 (Windows Store 1.8.42.0) (Windows 10), Board: "Arduino Uno"

C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\hardware -hardware C:\Users\Angshuman\Documents\ArduinoData\packages -tools C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\tools-builder -tools C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\hardware\tools\avr -tools C:\Users\Angshuman\Documents\ArduinoData\packages -built-in-libraries C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\libraries -libraries C:\Users\Angshuman\Documents\Arduino\libraries -fqbn=arduino:avr:uno -ide-version=10813 -build-path C:\Users\ANGSHU~1\AppData\Local\Temp\arduino_build_111639 -warnings=none -build-cache C:\Users\ANGSHU~1\AppData\Local\Temp\arduino_cache_438378 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.arduinoOTA.path=C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\hardware\tools\avr -prefs=runtime.tools.arduinoOTA-1.3.0.path=C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\hardware\tools\avr -prefs=runtime.tools.avrdude.path=C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\hardware\tools\avr -prefs=runtime.tools.avrdude-6.3.0-arduino17.path=C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\hardware\tools\avr -prefs=runtime.tools.avr-gcc.path=C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\hardware\tools\avr -prefs=runtime.tools.avr-gcc-7.3.0-atmel3.6.1-arduino7.path=C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\hardware\tools\avr -verbose C:\Users\Angshuman\Desktop\UltraSonic-Test\sendEvent\sendEvent.ino

C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\arduino-builder -compile -logger=machine -hardware C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\hardware -hardware C:\Users\Angshuman\Documents\ArduinoData\packages -tools C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\tools-builder -tools C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\hardware\tools\avr -tools C:\Users\Angshuman\Documents\ArduinoData\packages -built-in-libraries C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\libraries -libraries C:\Users\Angshuman\Documents\Arduino\libraries -fqbn=arduino:avr:uno -ide-version=10813 -build-path C:\Users\ANGSHU~1\AppData\Local\Temp\arduino_build_111639 -warnings=none -build-cache C:\Users\ANGSHU~1\AppData\Local\Temp\arduino_cache_438378 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.arduinoOTA.path=C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\hardware\tools\avr -prefs=runtime.tools.arduinoOTA-1.3.0.path=C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\hardware\tools\avr -prefs=runtime.tools.avrdude.path=C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\hardware\tools\avr -prefs=runtime.tools.avrdude-6.3.0-arduino17.path=C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\hardware\tools\avr -prefs=runtime.tools.avr-gcc.path=C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\hardware\tools\avr -prefs=runtime.tools.avr-gcc-7.3.0-atmel3.6.1-arduino7.path=C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\hardware\tools\avr -verbose C:\Users\Angshuman\Desktop\UltraSonic-Test\sendEvent\sendEvent.ino

Using board 'uno' from platform in folder: C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\hardware\arduino\avr

Using core 'arduino' from platform in folder: C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\hardware\arduino\avr

Detecting libraries used...

"C:\\Program Files\\WindowsApps\\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\\hardware\\tools\\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10813 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR "-IC:\\Program Files\\WindowsApps\\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\\hardware\\arduino\\avr\\cores\\arduino" "-IC:\\Program Files\\WindowsApps\\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\\hardware\\arduino\\avr\\variants\\standard" "C:\\Users\\ANGSHU~1\\AppData\\Local\\Temp\\arduino_build_111639\\sketch\\sendEvent.ino.cpp" -o nul -DARDUINO_LIB_DISCOVERY_PHASE

Alternatives for ESP8266WiFi.h: [ESP8266WiFi@1.0]

ResolveLibrary(ESP8266WiFi.h)

  -> candidates: [ESP8266WiFi@1.0]

"C:\\Program Files\\WindowsApps\\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\\hardware\\tools\\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10813 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR "-IC:\\Program Files\\WindowsApps\\ArduinoLLC.Ar [... cut ...]
  -> candidates: []

In file included from C:\Users\Angshuman\Desktop\UltraSonic-Test\sendEvent\sendEvent.ino:1:0:

C:\Users\Angshuman\Documents\Arduino\libraries\ESP8266WiFi\src/ESP8266WiFi.h:28:10: fatal error: wl_definitions.h: No such file or directory

 #include <wl_definitions.h>

          ^~~~~~~~~~~~~~~~~~

compilation terminated.

Using library ESP8266WiFi at version 1.0 in folder: C:\Users\Angshuman\Documents\Arduino\libraries\ESP8266WiFi 

exit status 1

Error compiling for board Arduino Uno.

```

d-a-v, 2021-04-25:
Well that's indeed not an instalaltion error but a user error :

> `Using board 'uno' from platform in folder`

You did not select the right board.

There was also ` Error compiling for board Arduino Uno.` in your OP.

You need to select the right board, any of the esp8266 flavor.

jandrassy, 2021-04-25:
the ESP8266WiFi library is for this esp8266 Arduino boards support package, not for Uno with esp8266 connected.

tragac909, 2022-03-03:
I was finding tutorials on internet and I've found this link: http://arduino.esp8266.com/stable/package_esp8266com_index.json
added it to Additional Boards Manager URLs, downloaded ESP8266 by comunity in Boards Manager, moved originaly installed ESP8266WiFi library and it compiles now.

shaina-espulgar, 2023-06-22:
Can you elaborate how did you move the ESP8266WiFi library? Also, are you using Arduino with ESP8266 ESP01 wifi module?