esp8266/Arduino #8004: Used EEPROM addresses

HamzaHajeir, 2021-04-26:
### Basic Infos

- [x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [ ] I have tested that the issue is present in current master branch (aka latest git).
- [x] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [x] I have filled out all fields below.

#### Platform

- Hardware:					ESP-12F
- Core Version:			2.7.4
- Development Env:		Arduino IDE
- Operating System:		Windows 10

### Settings in IDE

- Module:  			Nodemcu 1.0
- Flash Mode:			[qio|dio|other]
- Flash Size: 		4MB
- lwip Variant:   v2 Lower Memory
- Reset Method:		nodemcu
- Flash Frequency:	[40Mhz]
- CPU Frequency:		80Mhz
- Upload Using:		SERIAL
- Upload Speed:   921600
[image]

### Problem Description

A different EEPROM start address to what's mentioned in 4m1m ld script
I'm using the flash tool to upload an integer number that's used in an authentication process beside a MAC address, so I've chosen to upload it using esptool.py, However at the moment I'm using ESP Flash Tool to validate it.

[image]

I'm uploading number 5 (in hexadecimal) to address [0x3FB000](https://github.com/esp8266/Arduino/blob/209e467120ad0cd84db6bdf8dadd93ea55924f8c/tools/sdk/ld/eagle.flash.2m.ld#L5-L5).
and number 10 (in hexadecimal) to addresss [0x5FB000](https://github.com/esp8266/Arduino/blob/master/tools/sdk/ld/eagle.flash.4m1m.ld#L5-L5)

The expected result is that EEPROM.read(0) would return 10, It's returning 5.

The two files are just 4 bytes, little endian, with hex view is: 
node_id5.bin: 05 00 00 00 
node_id10.bin: 0A 00 00 00 

The MCVE contains a  function that manually reads the target address's content, I can remove it if it's not helping at all.

What should be the issue? I'm not sure why it's reading another address other than the expected one.

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp
#include <EEPROM.h>

int get_id_in_flash() {
   int* id_read_addr = (int*) 0x405FB000;
   //we can read 4-byte aligned address directly without memcpy
   return *id_read_addr;
}

void initEEPROM()
{
  int value = get_id_in_flash();
  Serial.printf("get_id()=%u\n", value);
  //init, only called once.
  //our EEPROM area is only 4 bytes long.
  //triggers reading flash into internal (4-byte) buffer
  EEPROM.begin(4);

  //read value stored during production at EEPROM index 0
  uint32_t node_id = 0;
  EEPROM.get(0, node_id);
  //use node_id variable..
  Serial.printf("Node_ID=");
  Serial.println(node_id);
  // Serial.printf("Node_ID=%u\n",node_id);
  //optional: we can read the node ID into a RAM variable once and then
  //EEPROM.end();
  //to release the internal 4-byte buffer.
}
void setup()
{
  Serial.begin(115200);
  initEEPROM();
}

void loop() {
  // put your main code here, to run repeatedly:
}

```

### Debug Messages

```
get_id()=393244
Node_ID=5
```

d-a-v, 2021-04-27:
What about the result of
```cpp
extern "C" uint32_t _EEPROM_start;
...
    Serial.printf("eeprom start is %p\n", &_EEPROM_start);
```
The shown result and where EEPROM class is going to read its data depend on the flash configuration used at sketch build time. 
There is an ongoing PR (or is it testing, tested?) to make this automatic, depending only on the detected flash size and not on user (mis-)configuration: #6690

HamzaHajeir, 2021-04-28:
> What about the result of
> 
> ```c++
> extern "C" uint32_t _EEPROM_start;
> ...
>     Serial.printf("eeprom start is %p\n", &_EEPROM_start);
> ```
> 
> The shown result and where EEPROM class is going to read its data depend on the flash configuration used at sketch build time.
> There is an ongoing PR (or is it testing, tested?) to make this automatic, depending only on the detected flash size and not on user (mis-)configuration: #6690

Thanks @d-a-v, I've run the sketch and it gave me a pretty weird result:
> eeprom start is 0x405fb000

Where `EEPROM.get(0, node_id);` reads `5` instead of `10`.

d-a-v, 2021-04-28:
I'm not the expert. Docs say the flash chip is mapped to address `0x40200000`.
We have `0x405fb000 - 0x40200000 = 0x3fb000` which is where you stored `5` in your bin file.

HamzaHajeir, 2021-04-28:
I see, that's absolutely makes sense.

I thought it a different way.

HamzaHajeir, 2021-04-28:
Closed due to incorrect understanding of memory mapping.

closed 2021-04-28 by HamzaHajeir

d-a-v, 2021-04-29:
@HamzaHajeir the keywords are ["esp8266 memory map"](https://www.google.com/search?q=esp8266+memory+map)

HamzaHajeir, 2021-04-30:
I've found it, And here's it for reference:
https://github.com/esp8266/esp8266-wiki/wiki/Memory-Map