esp8266/Arduino #8007: ESP8266 gives random error when I use 120KB on PROGMEM (despite there're plenty of space, 700KB).

Secmess, 2021-04-29:
Microcontroller: NodeMCU 1.0 (ESP8266)

- Before error: Complete webserver with HTML saved in Index.h as a PROGMEM. Everything fully functional.
- Error after: I tried to implement CSS through PROGMEM, either separated in another file (Style.h) as a new PROGMEM variable, or combined with Index.h as single variable.
- The error: Totally random, unrelated messages like `error: 'findFeed' was not declared in this scope`, `error: 'handle_onConnect' was not declared in this scope`, or any other clearly defined variables/functions.

So, the core problem is that the error completely gone when I randomly cut away 75% of the content from the CSS file. So **it might means I have problem related to my flash being full**. But, here's the thing, when I cut the CSS and the IDE successfully compile, the flash used is:

`Sketch uses 345912 bytes (33%) of program storage space. Maximum is 1044464 bytes.`

Meanwhile, the full CSS file is only 161KB vs 45KB when it compiles. So 116KB of extra flash usage gives me a random error.

Files:

- [Index.h](https://github.com/esp8266/Arduino/files/6400619/Index.txt)
- [Style.h](https://github.com/esp8266/Arduino/files/6400589/Style.txt) (bootstrap CSS)
- .ino snippet below (I omitted loops etc which isn't part of initialization)

```
/*  References:
 *   - https://forum.arduino.cc/t/html-text-box-to-send-text-to-esp8266-web-server/553649/7
 *   - https://www.w3schools.com/tags/tryit.asp?filename=tryhtml5_input_type_time
 *   - https://github.com/PaulStoffregen/Time
 *   - https://circuits4you.com/2018/02/04/esp8266-ajax-update-part-of-web-page-without-refreshing/
 *   - https://tttapa.github.io/ESP8266/Chap10%20-%20Simple%20Web%20Server.html
 */

#include <ESP8266WiFi.h>
#include <ESP8266WebServer.h>
#include <Servo.h>
#include <TimeLib.h>
#include ""
#include "Style.h"

/* Put your SSID & Password */
const char* ssid = "Feeder";    // Enter SSID here
const char* password = "12345678";    //Enter Password here

/* Put IP Address details */
IPAddress local_ip(192,168,1,1);
IPAddress gateway(192,168,1,1);
IPAddress subnet(255,255,255,0);
ESP8266WebServer server(80);

// Definitions
uint8_t LEDpin = LED_BUILTIN;
uint8_t Serpin = D6; 
int Serdur = 5;    // In sec, 100% duration.
int Serrot = 0;     // Servo rotation, 180 = inverse, 90 = stop.

// Form data
bool ledStat = LOW;
int servForm = 90;
int feed[] = {2, 100};  // Daily, each (percentage).
long offTime = 5;       // Offset feed time (sec).

// Process
bool feedNow = false;
bool servNow = false;
int feedn = 1;
time_t feedTime;
time_t servStart;

time_t timeNow;
Servo servo;

void setup() {
    Serial.begin(115200);
    pinMode(LEDpin, OUTPUT);
    servo.attach(Serpin);
    findFeed();

    WiFi.softAP(ssid, password);
    WiFi.softAPConfig(local_ip, gateway, subnet);
    delay(100);
    
    server.on("/", handle_onConnect);
    server.on("/style.css", handle_css);
    server.on("/readTime", handle_getTime);
    server.on("/setTime",  handle_timeForm);
    server.on("/setLed",   handle_setLed);
    server.on("/readFeed", handle_getFeed);
    server.on("/feedForm", handle_feedForm);
    server.on("/servForm", handle_servForm);
    server.onNotFound(handle_notFound);

    server.begin();
    Serial.println("HTTP server started");
}

//
// Codes around here are all omitted such as
// individual button/form action, loop, etc.
//

void handle_onConnect() {
    server.send(200, "text/html", SendHTML()); 
}

void handle_css() {
    server.send(200, "text/css", SendCSS()); 
}

String SendCSS() {
    // Read static components from PROGMEM
    String ptr = FPSTR(STYLE_CSS);
    return ptr;
}

String SendHTML(){
    // Read static components from PROGMEM
    String ptr = FPSTR(INDEX_HTML);
    return ptr;
}
```

Secmess, 2021-04-29:
I managed to do this through uploading the data separately (index.html & style.css) into the SPIFFS/flash, and using the file system library to read those. Sure, ideally this is the most "elegant" solution, but it involves dealing with an entirely different methods.

But, what we're dealing in this thread is an entirely different problem. I have found zero documentation about this particular limitation yet, and I'd be happy to find a fix to this. This method is still nice and simple, managing everything in a single, same-y workflow like any generic Arduino program.

Extra: At the end of upload (full index.html, style.css, and program) , I got a notification of only 35% flash usage, so it's confirmed the problem isn't flash size.

d-a-v, 2021-04-29:
Did you try without `String` and just this:
```
void handle_onConnect() {
    server.send_P(200, "text/html", PSTR(INDEX_HTML)); 
}
```

What you should realize it that is is impossible to do that:
```
String ptr = FPSTR(STYLE_CSS);
```
because in this case the 161KB flash string is ***copied*** to DRAM.

Secmess, 2021-04-30:
Thanks a lot, was actually excited coz that does seem like my problem as I didn't realize of that.

But I removed any reassigment of the PROGMEM (and only use them like you instructed), still the same problem. I even tried to remove ALL usage of the INDEX_HTML & STYLE_CSS, still same problem -- it gives similar error (unrelated "was not declared") just by including them.

So I tried to trim the CSS down like what I did above. Sure, similarly, that particular problem is gone. But it gave me a new error of `initializer fails to determine size of'__c'`. Tho, even if that's fixed, we do still have core problem in loading large file.

Still tho, thanks for that coz I wasn't aware that I used DRAM all this time.

d-a-v, 2021-05-11:
> we do still have core problem in loading large file.

Is it still the case ?
Otherwise if problems are gone you can close this issue.

Secmess, 2021-05-11:
Oh, definitely still a problem, to the point I had to use another method entirely.

d-a-v, 2021-05-11:
I'm closing this issue.
If you find that your new problem is related with something wrong in this core, please open a new issue.

Secmess, 2021-05-11:
> If you find that your new problem is related with something wrong in this core,

There's no "new problem" here. The original problem still exists as I stated above. The functionality in question is still broken here. Or am I missing something here that you can close an issue just by using an entirely different method?

mobizt, 2021-05-14:
@Secmess

Your static documents (.css, .js, .html, .jpg) should be minified (in case of plain text) and compress with gzip compression to reduce the size.

You can use 7zip program to compress the files by selecting **archive format -> gzip** with **dictionary size -> 32 KB**, **word size -> 32**, **compression level -> normal**, and **compression method -> deflate**.

Convert the gzip file to byte array which you can use this [online tool](http://tomeko.net/online_tools/file_to_hex.php?lang=en).

To send the header and payload, just use this.

```cpp
//Define the array of compressed file and put the array data to it.
static const uint8_t your_js_gz[] PROGMEM = {....};

//Send the content encoding header as gzip
server.sendHeader("Content-Encoding", "gzip");

//Send PROGMEM data
server.send_P(200, "application/javascript", (const char *)your_js_gz, sizeof(your_js_gz)); 
```

This way you can send large data e.g. large html, css and javascript files efficiently.

You can load the dynamic data later with AJAX to update the static document.

I can send 60kB gzip file compressed from 520 kB minified javascript file without the problem.

Secmess, 2021-05-14:
@mobizt Yes, I know that, which is why I'm able to continue with the project in the first place.

So, do you really have no choice other than using a file lower than maybe 80KB using PROGMEM? Despite the fact that we have plenty of space in the flash (500KB++).

@d-a-v I will try to make a MCVE later on. But yes, 50-ish KB of PROGMEM works without this strange error.

mobizt, 2021-05-14:
๑Secmess I don't have problem to send large file and please consider about time out.

mobizt, 2021-05-14:
@Secmess 
Please check this gist how to send 500 KB jpeg image (no gzip compression).
https://gist.github.com/mobizt/2b515a1b2ee864018798d976a2d1261d

This proves that sending the large flash data works.

Secmess, 2021-05-14:
@mobizt Thank you so much for that. I tried it right away and as you said, it compiles and uploads seamlessly. With that, I can safely close this issue as your code debunks my problem:

> Using a relatively large data (100KB+) as PROGMEM gives a strange unrelated error on compile.

Thing is, before, I already tried to trim everything of my code down to just the PROGMEM part similar to yours (and including the  fix to use send_P command as pointed out by @d-a-v to avoid using DRAM) and my problem persisted: It worked when I reduce below 50KB, and error again when I restore to 100KB+.

One thing I haven't tried is to use bytes (I use strings). I'll try to put my findings here in the future, but the core problem is definitely fixed. Again, thanks for the time, both of you.

closed 2021-05-14 by Secmess