esp8266/Arduino #8029: esp8266 html gzip & client.println

GastonMelo, 2021-05-12:
Hi to all in this forum, Im using Arduino IDE 1.8.13 with the esp8266 from AI-Cloud (the black module esp8266). Im a modify version of the esp8266 example WiFi HTTPSServer and im sending the HTML web page using the client.println instruction.
this is my complete code:

`#include <FS.h>
#include <ESP8266WiFi.h>
#include <ESP8266WebServer.h>
#include <ESP8266mDNS.h>
#include <DNSServer.h>
#include <WiFiManager.h>
#include <DoubleResetDetector.h>
#include "ArduinoJson.h"
#include "index_html.h"
#define DRD_TIMEOUT 10

// RTC Memory Address for the DoubleResetDetector to use
#define DRD_ADDRESS 0

DoubleResetDetector drd(DRD_TIMEOUT, DRD_ADDRESS);
//#include <EEPROM.h>
char WiFiRelay[21] = "xxxxxxxxxxxxxxxxxxx";//max 32 characters
WiFiServer server(80);

String header = "HTTP/1.1 200 OK \r\n\r\n\r\n";
String html_1 = "<!DOCTYPE html><html><head><link href='data:image/x-icon;base64,AAABAAEAEBAQAAEABAAoAQAAFgAAACgAAAAQAAAAIAAAAAEABAAAAAAAgAAAAAAAAAAAAAAAEAAAAAAAAAAAAAAA/4QAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAEAERABAAEAEBAQABAQEBABAAEAEQARABAQABAQABAAAQARAAEQARAAAAAAAAAAAAAAAAAAAAAAEREAEQAQAAAQAAEAEBAAABAAAAAQEAAAERAAEQAREAAQAAEAABABABAAAQAQEAEAEREAEQAREAAAAAAAAAAAD//wAA2N0AAKuqAADdmQAArrsAANnMAAD//wAA//8AAIZvAAC9rwAAv68AAI5jAAC97QAAva0AAIZjAAD//wAA' rel='icon' type='image/x-icon' /><meta name='referer' http-equiv='refresh' content='20' content='origin' name='viewport' content='width=device-width, initial-scale=1.0, maximun-scale= 1.0, user-scalable=0'><meta charset='utf-8'><style>body {background-color: white;}.button {background-color: #555555; border: none;color: white;padding: 20px;text-align: center;text-decoration: none;display: inline-block;font-size: 16px;margin: 4px 2px;cursor: pointer;}.button1{background-color:#008CBA;}.button2{background-color:#008CBA;}.button3{background-color:#008CBA;}.button4{background-color:#008CBA;}.button5{background-color:#4CAF50;}h {display: inline-block;margin: 0;background-color: black;color: white;font-size: 25px;}.button1 {border-radius: 12px;}.button2 {border-radius: 12px;}.button3 {border-radius: 12px;}.button4 {border-radius: 12px;}</style></head><body><form method = 'get' action = 'evento'>";
String html_2 = "<button1 class='button button1' name = 'battery'>Battery</button1><h>0V</h><br>";
String html_3 = "<button2 class='button button2' name = 'current'>Current</button2><h>0A</h><br>";
String html_4 = "<button3 class='button button3' name = 'temp_bat'>Batery_Temp</button3><h>0°C</h><br>";
String html_5 = "<button4 class='button button4' name = 'temp_mosf'>Mosfet_Temp</button4><h>0°C</h><br>";
String html_6 = "<button class='button button' name = 'iniciar' type = 'submit' value='0'>Iniciar</button><br>";
String html_7 = "</form></body></html>";
String request = "";
String name_device = "";
bool shouldSaveConfig = false;
bool on_btn_flag = false;
bool fpga_conv = false;
//---------------------------variables de medicion-------------------------//
const int BUFFER_SIZE=10;
char from_fpga[BUFFER_SIZE];
char *puntero;
String bateria,corriente,temperatura_bateria,temperatura_mosfet,debug;
int ctrlostlink = 0;

void saveConfigCallback () {
  //Serial.println("Should save config");
  shouldSaveConfig = true;
}
void configModeCallback (WiFiManager *myWiFiManager) {
    //Serial.println("Entered config mode");
  //Serial.println(WiFi.softAPIP());
  drd.stop();
}
void setup() {
  Serial.begin(115200);
  //Serial.println("WifiRelay activado");
  //Serial.println();
  //EEPROM.begin(512);
  delay(10);
  puntero = from_fpga;
  //Serial.println("mounting FS...");
  //comienza lectura memoria
  if (SPIFFS.begin()) {
    if (SPIFFS.exists("/config.json")) {
      //file exists, reading and loading
      File configFile = SPIFFS.open("/config.json", "r");
      if (configFile) {
        size_t size = configFile.size();
        // Allocate a buffer to store contents of the file.
        std::unique_ptr<char[]> buf(new char[size]);
        configFile.readBytes(buf.get(), size);
        DynamicJsonBuffer jsonBuffer;
[... 178 lines cut ...]
    delay(100);
  }
  EEPROM.commit();
}
//
String EEPROM_ESP8266_LEER(int min, int max) {
EEPROM.begin(512); delay(100);String buffer;
  for (int L = min; L < max; ++L)
    if (isAlphaNumeric(EEPROM.read(L)))
      buffer += char(EEPROM.read(L));
  return buffer;
}
*/`

Is it possible to gzip the html page and send it using the client.println instruction?
Im trying to send it using this code:
`  client.println("HTTP/1.1 200 OK");
  //client.println("Content-Type: text/html");
  client.println("Content-Encoding: gzip; charset=UTF-8> \r\n");
  client.print("Content-Length:");
  client.println(sizeof(index_html_gz));
  client.print(index_html_gz[640]);`
And Chrome is showing a white window only.
Thanks
Gastón

themindfactory, 2021-05-12:
> String html_1 = "<style>body {background-color: white;}.button
> String html_2 = "Battery0V
> String html_3 = "Current0A
> String html_4 = "Batery_Temp0°C
> String html_5 = "Mosfet_Temp0°C
> String html_6 = "Iniciar
> html_2 = "Battery" + bateria + "
> html_3 = "Current" + corriente + "
> html_4 = "Battery_Temp" + temperatura_bateria + "
> html_5 = "Mosfet_Temp" + temperatura_mosfet + "
> html_6 = "Iniciar
> html_2 = "Battery0V
> html_3 = "Current0A
> html_4 = "Batery_Temp0°C
> html_5 = "Mosfet_Temp0°C

If the html is generated in by software then you need your own gzip
function to pass the html to.

RichardS

GastonMelo, 2021-05-12:
Hi themindfactory,
                               Im using cyberchef to do the gzip (into array of hex values) and store those values in array index_html_gz[].
If I use the above code to send the gzip data, chrome is showing a withe windows with nothing on it. I was thinking that sending that array of hex values in the index_html_gz will just work of out the box. 
I don't understant this part of your reply "your own gzip function to pass the html ". How would I do that? could you point me to some direction to look further? 
Thanks for your reply
Gastón

themindfactory, 2021-05-12:
Try client.write()

themindfactory, 2021-05-12:
> Gaston,
> let's assume the index_html_gz array contains the gzip data created by a
> function (not shown here).
> In your code, I see two problems, maybe bugs:
>
> client.println(sizeof(index_html_gz)); - it is ok to send *the same value*
> (length of the array) for all requests?
>
> client.print(index_html_gz[640]); - this command writes a string from
> 640th position of the input string that is obviously wrong, use
> client.write (as suggested @themindfactory
> <https://github.com/themindfactory>)

And remove the [640] in the call

themindfactory, 2021-05-12:
client.write(index_html_gz)

GastonMelo, 2021-05-12:
thanks for your help JiriBilek and the mindfactory and quick reply,
                                                                                                           the index_html_gz[] is :
`const uint8_t index_html_gz[] = {
 0x1f,0x8b,0x08,0x08,0xe7,0x8c,0x91,0x60,0x00,0xff,0x69,0x6e,0x64,0x65,0x78,0x5f,0x68,0x74,0x6d,0x6c,0x00,0x75,0x51,0xcd,0x6a,0xf3,0x30,0x10,0xbc,0xfb,0x29,
 0xb6,0xee,0x2d,0xe0,0x28,0x71,0xd2,0x1e,0x14,0xd7,0x10,0xd2,0x9f,0x63,0x7b,0x28,0x1f,0xf4,0x28,0x4b,0xfa,0x9c,0x25,0x8a,0x64,0x24,0x99,0x26,0x0d,0x7d,0xf7,
 0x48,0xb5,0x62,0x28,0x34,0xa7,0x85,0x99,0x9d,0xd9,0xd1,0x08,0xaa,0x9b,0xc7,0xd7,0xcd,0xfb,0xc7,0xdb,0x13,0x6c,0xfd,0x5e,0xd5,0x59,0x75,0x19,0x92,0x89,0x30,
 0x9c,0x3f,0x2a,0x59,0x67,0x8d,0x11,0x47,0x38,0x65,0x00,0x0d,0xe3,0xbb,0xd6,0x9a,0x5e,0x8b,0x82,0x1b,0x65,0x2c,0x85,0xcf,0x2d,0x7a,0xb9,0xca,0xbe,0xb3,0x69,
 0xd3,0x7b,0x6f,0xf4,0x95,0xb5,0xdb,0xe5,0x66,0xfd,0x7c,0x37,0x5b,0x01,0x99,0xc0,0x8b,0x95,0x52,0xc3,0x84,0xc4,0x45,0x63,0x85,0x0c,0xb4,0x36,0x3a,0x98,0x00,
 0xfc,0x36,0x05,0xe8,0x98,0x10,0xa8,0x5b,0x0a,0xe5,0xac,0x3b,0x44,0xc0,0xcb,0x83,0x2f,0x98,0xc2,0x56,0x53,0xe0,0x52,0x7b,0x69,0x47,0x54,0x48,0x6e,0x2c,0xf3,
 0x68,0x02,0x75,0xf1,0x13,0xe8,0x3a,0xc5,0x8e,0x14,0x50,0x2b,0xd4,0xb2,0x68,0x94,0xe1,0xbb,0x48,0xfc,0x37,0xda,0x17,0x0e,0xbf,0x24,0x85,0xf9,0xfd,0x60,0xbd,
 0x67,0xb6,0xc5,0xa0,0x5d,0x76,0x07,0x28,0x07,0x88,0xf7,0xd6,0xc5,0x3c,0x9d,0xc1,0xe1,0xd4,0xf8,0xcc,0x39,0x9c,0x86,0xec,0x85,0x65,0x02,0x7b,0x17,0x6c,0xa2,
 0x66,0xe4,0xcb,0x6b,0x7c,0x45,0x52,0xa7,0x15,0x49,0x1d,0xc7,0x72,0xe3,0x48,0xbe,0x5c,0x31,0xe7,0x1e,0xf2,0xd4,0x66,0x42,0xf3,0xba,0x61,0x21,0x00,0x32,0xba,
 0x98,0x2e,0xfe,0x55,0x24,0xc1,0x75,0xd5,0xd8,0x51,0x5a,0xfe,0x29,0x2d,0xf3,0x3a,0xd4,0x62,0x31,0x76,0x45,0xe7,0xeb,0x8b,0xb4,0x1c,0xa4,0x21,0x45,0x3a,0x4f,
 0x7e,0x3e,0xfe,0x0c,0x5f,0xc2,0x91,0x23,0x10,0x02,0x00,0x00
};`

and using the client.write(index_html_gz); throws this error:
`datalogger:205:29: error: call of overloaded 'write(const uint8_t [322])' is ambiguous
   client.write(index_html_gz);`
`call of overloaded 'write(const uint8_t [322])' is ambiguous`
Gastón

themindfactory, 2021-05-12:
Sorry might need to be client.write(index_html_gz, len(index_html_gz))

GastonMelo, 2021-05-12:
Hi to all,
                 using this code:
`  client.write("Content-Encoding: gzip; charset=UTF-8>");
  client.write(index_html_gz, index_html_gz_len);`
 I got this on chrome:
[image]

and i download windows is shown. 
Gastón

JiriBilek, 2021-05-12:
Fix the extra '>' in the first `client.write`, but I doubt it'd sort out the issue :(

themindfactory, 2021-05-12:
I think also you need one or two println after header if I remember right

GastonMelo, 2021-05-12:
I'm trying with this one now:
`  client.println("HTTP/1.1 200 OK");
  client.println("<meta http-equiv='content-type' content='gzip; charset=UTF-8'>");
  client.println("");
  client.println("");
  client.write(index_html_gz, index_html_gz_len);`

still the same result.

themindfactory, 2021-05-13:
guessing but I suspect its:

client.println("HTTP/1.1 200 OK");
client.println("Content-Encoding:gzip");
client.println("Content-Type:text/html");
client.println();
client.write(index_html_gz, index_html_gz_len);

RichardS

GastonMelo, 2021-05-13:
that's right!!
I don't understand why the: client.println("") its wrong. I see this line of code in other forums. And why after the content.encoding is sent the content-type:text-html? If I just sent the html coded in hex wouldn't just work only with Content-Encoding?
Thanks again for your help.
Gastón

themindfactory, 2021-05-13:
maybe not sure it may default to text-html.... better safe then sorry.

earlephilhower, 2021-05-19:
This is a "how do I" type of question which is better handled on https://esp8266.com or https://gitter.im/esp8266/Arduino, as mentioned in the template you erased.  Please move it there if you still need help.

closed 2021-05-19 by earlephilhower