esp8266/Arduino #8031: ESP8266 ESP-01 MONITOR DOESN´T APPEAR ANSWER INSIDE MINGW32

paulotirapelle2, 2021-05-13:
8. Use markup (buttons above) and the Preview tab to check what the issue will look like.
9. Delete these instructions from the above to the below marker lines before submitting this issue.

----------------------------- Delete above -----------------------------

### Basic Infos

- [ X] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [X ] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [X ] I have tested that the issue is present in current master branch (aka latest git).
- [ X] I have searched the issue tracker for a similar issue.
- [X ] If there is a stack dump, I have decoded it.
- [X ] I have filled out all fields below.

#### Platform

- Hardware:		                [ESP-01]
- Core Version:			[https://github.com/espressif/ESP8266_RTOS_SDK/tree/master/examples/get-started/hello_world]
- Development Env:		[mingw32]
- Operating System:		[Windows10]

### Settings in IDE

- Module:  			[ESP-IDF]
- Flash Mode:			[qio]
- Flash Size: 		[2MB]
- lwip Variant:   
- Reset Method:		[ck]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[160MHz]
- Upload Using:		[SERIAL]
- Upload Speed:   [115200]

### Problem Description

After program esp8266 with FTDI programmer, the command "make monitor" doesn´t answer. 

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```

```

### Debug Messages Error
```
Leaving...
Hard resetting via RTS pin...

Paulo@DESKTOP-LJ7FF2A MINGW32 ~/esp/ESP8266_RTOS_SDK/examples/get-started/hello_world
$ make monitor
Toolchain path: /opt/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc
Toolchain version: esp-2020r3-49-gd5524c1
Compiler version: 8.4.0
Python requirements from C:/msys32/home/Paulo/esp/ESP8266_RTOS_SDK/requirements.txt are satisfied.
MONITOR
--- idf_monitor on C:/msys32/dev/COM3 74880 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---

```

### Solution

 After program the chip, disconnect USB programmer, disconnect GPIO_0  from GND and plug ESB again. 
 Make a new command "make monitor".
 

### Debug Messages Working

`
Hello world!
This is ESP8266 chip with 1 CPU cores, WiFi, silicon revision 1, 2MB external fl
ash
Restarting in 10 seconds...
Restarting in 9 seconds...
Restarting in 8 seconds...
Restarting in 7 seconds...
Restarting in 6 seconds...
Restarting in 5 seconds...
Restarting in 4 seconds...
Restarting in 3 seconds...

`

earlephilhower, 2021-05-13:
This is not related to the ESP8266 Arduino core.  Please use the IDF repo for reporting/asking this.

closed 2021-05-13 by earlephilhower

paulotirapelle2, 2021-05-13:
sorry