esp8266/Arduino #8033: ESP8266 WebServer not working if gateway is same as the router

cg101, 2021-05-15:
Web Server doesn't work if my softAP gateway is the same as the router's gateway which my esp is connected to. 
My Dlink router's gateway is 192.168.100.1. When I manually configure my esp softAP gateway as the same, the webserver is not working. When I comment out the softAPConfig line, esp uses 192.168.4.1 as the gateway and the web server works fine.

I have the following code:

#include <ESP8266WiFi.h>
#include <ESP8266WebServer.h>
ESP8266WebServer server;

char* ssid = "D-Link";
char* pass = "MyPassword";
IPAddress apIP(192, 168, 100, 1);
IPAddress apGateway(192, 168, 100, 1);
IPAddress apSubmask(255, 255, 255, 0);

void setup() {
  Serial.begin(115200);
  WiFi.mode(WIFI_AP_STA);
  WiFi.softAPConfig(apIP, apGateway, apSubmask);
  WiFi.softAP("ESP8266","password");
  WiFi.begin(ssid,pass);
  
  while(WiFi.status()!=WL_CONNECTED){
    Serial.print(".");
    delay(500);
  }
  Serial.println();
  Serial.println("Connected.");
  Serial.println("Ip Address: ");
  Serial.println(WiFi.localIP()); 

  server.begin();
  server.on("/",[](){
    server.send(200,"text/plain","Hello");
  });
}

void loop() {
  server.handleClient();
}
Serial monitor output:
...
Connected.
Ip Address: 
192.168.100.3

jandrassy, 2021-05-15:
the gateway is not on the SoftAP network so don't configure it there.

earlephilhower, 2021-05-17:
As @jandrassy mentioned, this looks to be a coding/network config error.  Closing.

closed 2021-05-17 by earlephilhower