esp8266/Arduino #8045: OTA  not working on Robotdyn D1 R2 boards

paai, 2021-05-17:
### Basic Infos

- [ x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [ x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [ x] I have tested that the issue is present in current master branch (aka latest git).
- [ x] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [ ] I have filled out all fields below.

#### Platform

- Hardware:				Robotdyn D1 R2
- Core Version:			[latest git hash or date]
- Development Env:		Arduino 1.8.15 and older
- Operating System:		Kubuntu 18.04

### Settings in IDE

- Module:  			Wemos D1 mini r2
- Flash Mode:			[qio|dio|other]
- Flash Size: 		[4MB/1MB]
- lwip Variant:   [v1.4|v2 Lower Memory|Higher Bandwidth]
- Reset Method:		[ck|nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz|160MHz]
- Upload Using:		[OTA|SERIAL]
- Upload Speed:   [115200|other] (serial upload only)

### Problem Description
This is a problem related to the popular Robotdyn D1R2 boards.

Using the standard BasicOTA from the examples, the Robotdyn D1R2 stops OTA uploading after 25-50% when the AP is three meters away. Upload does not even start on APs that are farther away. All other wifi functions work just fine.Tested it on seven of these boards. 

The same code on  e.g., a Robotdyn D1 mini or LOLIN ESP32 (of course with different libraries) just works. 

I also installed the esp8266 library from the Board manager (after uninstalling the github tree) and exactly the same thing happens.

What surprises me is that I can not find references to this problem on the internet, although the Robotdyn D1R2 is a very popular board.

earlephilhower, 2021-05-17:
From your description of the same code working on other boards, this doesn't seem to be a core issue.  Your best bet would be to use esp8266.com or https://gitter.im/esp8266/Arduino, or the manufacturer.  Good luck!

closed 2021-05-17 by earlephilhower

d-a-v, 2021-05-17:
As said in https://github.com/esp8266/Arduino/issues/5990#issuecomment-842254878 and if the issue is easily repeatable, a tcpdump trace could be helpful for #4054 and #6369. It can be done with the integrated [netdump](../tree/master/libraries/Netdump) library.

paai, 2021-05-18:
Yesterday I tried to contact support@robotdyn.com, but the mailaddress does not exist (!).

I never used tcpdump and would not know what to look for.  I started "tcpdump host 192.168.178.50" on the commandline while uploading the sketch both for the failing D1R2 and the working D1-mini and captured the output, but I don't know what I am looking at. I am a bit reluctant to copy the output here, but here it is...

 

09:59:27.385609 IP 192.168.178.50.mdns > 224.0.0.251.mdns: 0*- [0q] 2/0/0 (Cache flush) A 192.168.178.50, (Cache flush) PTR myesp8266-bla.local. (88)
09:59:27.389386 IP 192.168.178.50.mdns > 224.0.0.251.mdns: 0*- [0q] 4/0/1 PTR _arduino._tcp.local., PTR myesp8266-bla._arduino._tcp.local., (Cache flush) SRV myesp8266-bla.local.:8266 0 0, (Cache flush) TXT "auth_upload=no" "board="ESP8266_WEMOS_D1MINI"" "ssh_upload=no" "tcp_check=no" (258)
09:59:28.394882 IP 192.168.178.50.mdns > 224.0.0.251.mdns: 0*- [0q] 2/0/0 (Cache flush) A 192.168.178.50, (Cache flush) PTR myesp8266-bla.local. (88)
09:59:28.401695 IP 192.168.178.50.mdns > 224.0.0.251.mdns: 0*- [0q] 4/0/1 PTR _arduino._tcp.local., PTR myesp8266-bla._arduino._tcp.local., (Cache flush) SRV myesp8266-bla.local.:8266 0 0, (Cache flush) TXT "auth_upload=no" "board="ESP8266_WEMOS_D1MINI"" "ssh_upload=no" "tcp_check=no" (258)
09:59:29.400917 IP 192.168.178.50.mdns > 224.0.0.251.mdns: 0*- [0q] 2/0/0 (Cache flush) A 192.168.178.50, (Cache flush) PTR myesp8266-bla.local. (88)
09:59:29.410441 IP 192.168.178.50.mdns > 224.0.0.251.mdns: 0*- [0q] 4/0/1 PTR _arduino._tcp.local., PTR myesp8266-bla._arduino._tcp.local., (Cache flush) SRV myesp8266-bla.local.:8266 0 0, (Cache flush) TXT "auth_upload=no" "board="ESP8266_WEMOS_D1MINI"" "ssh_upload=no" "tcp_check=no" (258)
09:59:41.271448 IP danzig.paijmans.local.43191 > 192.168.178.50.8266: UDP, length 48
09:59:42.403074 ARP, Request who-has danzig.paijmans.local tell 192.168.178.50, length 46
09:59:42.403102 ARP, Reply danzig.paijmans.local is-at f4:4d:30:49:3b:77 (oui Unknown), length 28
09:59:42.408801 IP 192.168.178.50.8266 > danzig.paijmans.local.43191: UDP, length 2
09:59:42.503109 IP 192.168.178.50.62757 > danzig.paijmans.local.51449: Flags [S], seq 6509, win 2144, options [mss 536,nop,nop,sackOK], length 0
09:59:42.503155 IP danzig.paijmans.local.51449 > 192.168.178.50.62757: Flags [S.], seq 4045840852, ack 6510, win 64240, options [mss 1460,nop,nop,sackOK], length 0
09:59:42.509244 IP 192.168.178.50.62757 > danzig.paijmans.local.51449: Flags [.], ack 1, win 2144, length 0
09:59:42.509539 IP danzig.paijmans.local.51449 > 192.168.178.50.62757: Flags [.], seq 1:537, ack 1, win 64240, length 536
09:59:42.509547 IP danzig.paijmans.local.51449 > 192.168.178.50.62757: Flags [.], seq 537:1073, ack 1, win 64240, length 536
09:59:42.509688 IP danzig.paijmans.local.51449 > 192.168.178.50.62757: Flags [P.], seq 1073:1461, ack 1, win 64240, length 388
09:59:42.519072 IP 192.168.178.50.62757 > danzig.paijmans.local.51449: Flags [.], ack 537, win 2144, length 0
09:59:42.524412 IP 192.168.178.50.62757 > danzig.paijmans.local.51449: Flags [P.], seq 1:4, ack 537, win 2144, length 3
09:59:42.524432 IP danzig.paijmans.local.51449 > 192.168.178.50.62757: Flags [.], ack 4, win 64237, length 0
09:59:42.524551 IP danzig.paijmans.local.51449 > 192.168.178.50.62757: Flags [.], seq 1461:1997, ack 4, win 64237, length 536
09:59:42.524562 IP danzig.paijmans.local.51449 > 192.168.178.50.62757: Flags [.], seq 1997:2533, ack 4, win 64237, length 536
09:59:42.549707 IP danzig.paijmans.local.51449 > 192.168.178.50.62757: Flags [.], seq 1997:2533, ack 4, win 64237, length 536
09:59:42.551319 IP 192.168.178.50.62757 > danzig.paijmans.local.51449: Flags [.], ack 1073, win 2144, length 0
09:59:42.554289 IP 192.168.178.50.62757 > danzig.paijmans.local.51449: Flags [P.], seq 4:7, ack 1073, win 2144, length 3
09:59:42.554625 IP danzig.paijmans.local.51449 > 192.168.178.50.62757: Flags [.], seq 2533:3069, ack 7, win 64234, length 536
09:59:42.554638 IP danzig.paijmans.local.51449 > 192.168.178.50.62757: Flags [.], seq 3069:3217, ack 7, win 64234, length 148
09:59:42.587319 IP 192.168.178.50.62757 > danzig.paijmans.local.51449: Flags [P.], seq 7:10, ack 1461, win 1756, length 3
09:59:42.613617 IP 192.168.178.50.62757 > danzig.paijmans.local.51449: Flags [.], ack 1997, win 2144, length 0
09:59:42.613659 IP danzig.paijmans.local.51449 > 192.168.178.50.62757: Flags [.], seq 3217:3753, ack 10, win 64231, length 536
09:59:42.622079 IP 192.168.178.50.62757 > danzig.paijmans.local.51449: Flags [P.], seq 10:13, ack 1997, win 2144, length 3
09:59:42.648661 IP 192.168.178.50.62757 > danzig.paijmans.local.51449: Flags [.], ack 2533, win 2144, length 0
09:59:42.648689 IP danzig.paijmans.local.51449 > 192.168.178.50.62757: Flags [P.], seq 3753:4289, ack 13, win 64228, length 536
09:59:42.653467 IP 192.168.178.50.62757 > danzig.paijmans.local.51449: Flags [P.], seq 13:16, ack 2533, win 2144, length 3
09:59:42.675927 IP 192.168.178.50.62757 > danzig.paijmans.local.51449: Flags [.], ack 2533, win 2144, length 0
09:59:42.688534 IP 192.168.178.50.62757 > danzig.paijmans.local.51449: Flags [.], ack 3069, win 2144, length 0
09:59:42.688572 IP danzig.paijmans.local.51449 > 192.168.178.50.62757: Flags [.], seq 4289:4825, ack 16, win 64225, length 536
09:59:42.691031 IP 192.168.178.50.62757 > danzig.paijmans.local.51449: Flags [P.], seq 16:19, ack 3069, win 2144, length 3
[... cut ...]
09:59:52.677677 IP danzig.paijmans.local.51449 > 192.168.178.50.62757: Flags [.], seq 30529:31041, ack 163, win 64078, length 512
09:59:55.734608 IP danzig.paijmans.local.51449 > 192.168.178.50.62757: Flags [R.], seq 32625, ack 163, win 64078, length 0
09:59:55.741603 IP 192.168.178.50.62757 > danzig.paijmans.local.51449: Flags [.], ack 30529, win 2144, length 0
09:59:55.741662 IP danzig.paijmans.local.51449 > 192.168.178.50.62757: Flags [R], seq 4045871381, win 0, length 0
10:05:49.283593 ARP, Request who-has 192.168.178.50 tell paai_grashoek.paijmans.local, length 46

d-a-v, 2021-05-18:
Thanks for this dump @paai.
What we can see is that at the end ESP is asking for retransmission from 30529, that is honoured by PC but never acknowledged by esp then PC shuts down the connection. It'll hopefully help to understand while reading between the lines of espota.py and the ArduinoOTA library.
```
09:59:45.609463 IP pc.51449 > esp.62757: Flags [.], seq 29993:30529, ack 150, win 64091, length 536
09:59:45.666551 IP pc.51449 > esp.62757: Flags [.], seq 30529:31041, ack 153, win 64088, length 512
09:59:45.735191 IP esp.62757 > pc.51449: Flags [.], ack 30529, win 2144, length 0
09:59:45.762370 IP esp.62757 > pc.51449: Flags [P.], seq 160:163, ack 30529, win 2144, length 3
09:59:46.501716 IP pc.51449 > esp.62757: Flags [.], seq 30529:31041, ack 163, win 64078, length 512
09:59:47.366922 IP pc.51449 > esp.62757: Flags [.], seq 30529:31041, ack 163, win 64078, length 512
09:59:49.093774 IP pc.51449 > esp.62757: Flags [.], seq 30529:31041, ack 163, win 64078, length 512
09:59:52.677677 IP pc.51449 > esp.62757: Flags [.], seq 30529:31041, ack 163, win 64078, length 512
09:59:55.741603 IP esp.62757 > pc.51449: Flags [.], ack 30529, win 2144, length 0
```

paai, 2021-05-18:
Thank you. But I am rapidly getting out of my depth here and I don't speak python (the curly braces on the keyboard are there for a good reason!).  
The bottom line seems to be that something is broken in Robotdyn D1R2 boards that cannot be easily remedied. But it is a popular board and has been around or at least four years, so it is peculiar that nobody stumbled over it before. Ah well... Thank you all for helping me.