esp8266/Arduino #8046: #6583 is breaking change - developers should know.

mathertel, 2021-05-17:
### Basic Infos

- [X] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [X] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [X] I have **checked** that the issue is present in current master branch (aka latest git).
- [X] I have searched the issue tracker for a similar issue.
- [X] If there is a stack dump, I have decoded it.
- [ ] I have filled out all fields below.

### The topic

There was a change should be listed under **breaking changes** as you have to re-write the parameter on any handler overriding the existing functions. If you have no override annotation it even does compile - but do not work.

Was introduced in https://github.com/esp8266/Arduino/pull/6583

As there is no example where a class is deriving from RequestHandler this is not detected in builds.

Hieromon, 2021-05-17:
@earlephilhower, Is it possible to notify other library developers of such significant changes in advance?
You say RequestHandler is basically private, but I understood it was the public way to make the RequestHandler implementation. (That's why I used that utility)

earlephilhower, 2021-05-17:
@Hieromon while it does look like this escaped notice as a breaking change, it was in the development repo for almost a full year before  the 3.0.0 release, which is really the only way we have of letting the world know where the repo is going. 

@mathertel we can add a note on the next bugfix about this, sure.  Because this is a 2.x -> 3.x transition, breaking changes are expected, but looks like we missed a note about this.

mathertel, 2021-05-21:
Here is a code snipped to declare the function so they can be implemented using the same code:

`#if defined(ESP8266)
  bool handle(WebServer &server, HTTPMethod requestMethod, const String &requestUri) override
#elif defined(ESP32)
  bool handle(WebServer &server, HTTPMethod requestMethod, String requestUri) override
#endif
`

Same for `upload`, `canHandle` and `canUpload`

Hieromon, 2021-05-22:
I reluctantly adopted the following implementation to ensure backward compatibility with my library. It's an incomplete method that relies on fixed member function signatures and is verbose if I try to write it completely.
Is there a better alternative to this way?

```cpp
template<typename T>
struct TypeOfArgument;
template<typename T, typename U, typename... V>
struct TypeOfArgument<U(T::*)(V...)> {
  template<size_t i>
  struct arg {
    typedef typename std::tuple_element<i, std::tuple<V...>>::type  type;
  };
};

// Determines the type of the uri argument contained in the member function
// signature of the RequestHandler class. This redefinition procedure ensures
// backward compatibility with ESP8266 arduino core 3.0.0 and later.
// However, it relies solely on the canHandle member function as a criterion
// and lacks completeness.
using URI_TYPE_SIGNATURE = std::conditional<
  std::is_lvalue_reference<TypeOfArgument<decltype(&RequestHandler::canHandle)>::arg<1>::type>::value,
  const String&,
  String
>::type;

Class Foo : RequestHandler {
 public :
  Foo(){}
  ~Foo(){}
	
  bool canHandle(HTTPMethod requestMethod, URI_TYPE_SIGNATURE requestUri) override;
  bool canUpload(URI_TYPE_SIGNATURE uri) override;
  bool handle(WebServerClass& server, HTTPMethod requestMethod, URI_TYPE_SIGNATURE requestUri) override;
  void upload(WebServerClass& server, URI_TYPE_SIGNATURE requestUri, HTTPUpload& upload) override;
};
```

Initially, I tried to use **esp8266::coreVersionNumeric**, which is the recommended method for absorbing version, but [`constexpr parseNthInteger`](https://github.com/esp8266/Arduino/blob/8dc80b5ef6739b28917a12da9a1c10ce3741971a/cores/esp8266/core_esp8266_version.h#L104) can not be used in the class member function declaration in order to be evaluated at the same timing as the instantiation of the template.

d-a-v, 2021-05-22:
Would a global define restoring the old-and-esp32-compatible API help ?
(something like `#ifdef COMPAT_ESP32` or `#if COMPAT_ESP32 >= 10001 // >= v1.00.01`, you would have to use a global `#define COMPAT_ESP32 10001` in your project)

d-a-v, 2021-06-24:
Closing this issue per following actions
- Breaking change list summary for 3.0.0 has been updated to highlight #6583
- Release described by \#defines have been added and will be available [for release 3.0.1](https://github.com/esp8266/Arduino/issues/8049#issuecomment-862798730)

closed 2021-06-24 by d-a-v

Hieromon, 2021-06-25:
@d-a-v In conclusion, I find it difficult to completely eliminate the legacy method of relying on #define to satisfy static type determination in C++11 as a way to keep backward compatibility. It detains us.
These problems are common in infrastructure components that have been in use for a long time, but at the same time prove that they are valuable.
Anyway, thank you for your support.