esp8266/Arduino #8047: Problems with creating webserver ( javascript part).

Xuifyy, 2021-05-17:
Hey there! I am very new to programing and have a lot to learn. I have my own ledstrip project and I use an esp8266 for this project. I wanted to control the ledstrip using a webserver so I beginning to learn html and javascript (basics). It was all working fine till I tried to transver my code from visual studio code to arduino (control c -> control v). The html part works perfectly fine but the javascript part doesnt seem to work and I cant figure out why.

### Basic Infos

- [ ] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [ ] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [ ] I have tested that the issue is present in current master branch (aka latest git).
- [ ] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [ ] I have filled out all fields below.

#### Platform

- Hardware:					[ESP-12|ESP-01|ESP-07|ESP8285 device|other]
- Core Version:			[latest git hash or date]
- Development Env:		[Arduino IDE|Platformio|Make|other]
- Operating System:		[Windows|Ubuntu|MacOS]

### Settings in IDE

- Module:  			[Generic ESP8266 Module|Wemos D1 mini r2|Nodemcu|other]
- Flash Mode:			[qio|dio|other]
- Flash Size: 		[4MB/1MB]
- lwip Variant:   [v1.4|v2 Lower Memory|Higher Bandwidth]
- Reset Method:		[ck|nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz|160MHz]
- Upload Using:		[OTA|SERIAL]
- Upload Speed:   [115200|other] (serial upload only)

### Problem Description

Detailed problem description goes here.
I made a html string with all the code in it and the html& css part of the code works but the javascript part doesnt. When I inspect it in google it says the following : "(index):1 Uncaught ReferenceError: Sweeping is not defined
    at HTMLDivElement.onmouseover ((index):1)"
Sweeping is one of my animations functions.

_**This is just the part of the code that matters for my problem. If you need more info because I missed something then let met know. Thank you so much for ur time**_

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp

#include <Arduino.h>

void Kamiels_world() {

  // Build a String with all the HTML in it.
  String HTML = "";
  HTML += "<html>";
  HTML += "  <title>Kamiels world</title>";
  HTML += " <link rel='shortcut icon' type='image/jpg' src='.\TropicLovepicca.jpg'";
  HTML += " <head>";
  HTML += "   ";
  HTML += "";
  HTML += "   <center>";
  HTML += "     <h2> Test bestand project Kamiels world</h2>";
  HTML += "   </center> <br>";
  HTML += "   <center>";
  HTML += "     ";
  HTML += "     <p1> Mogelijk wordt deze code gebruikt voor het toekomstige prototype </p1>";
  HTML += "   </center>";
  HTML += "   <br>";
  HTML += "   <br>";
  HTML += "   <br>";
  HTML += "   <style>";
  HTML += "     ";
  HTML += "     .CardHolder{";
  HTML += "       display: flex;";
  HTML += "       flex-direction: row;";
[... 280 lines cut ...]
  HTML += "       if(speed == 0){";
  HTML += "         url+=`&speed=70` ";
  HTML += "       }";
  HTML += "       fetch(url,";
  HTML += "     {";
  HTML += "       method: 'POST'";
  HTML += "     }).then(result => {";
  HTML += "     })";
  HTML += "     .catch(error => {";
  HTML += "     })";
  HTML += "   }";
  HTML += " </script>";
  HTML += "</html>";

  // Send the string to the browser as HTML text
  server.send(200, "text/html", HTML);
}

```

### Debug Messages

```
Debug messages go here
```

Xuifyy, 2021-05-17:
Sorry if something is not clear or I left a part of my code out that matters. I am very new to this and am interested to learn more about programming. Thank you so much for ur time!

earlephilhower, 2021-05-17:
This is not the appropriate spot for general "how do I" questions, as the template says.  Please use esp8266.com or https://gitter.im/esp8266/Arduino for general programming assistance.

closed 2021-05-17 by earlephilhower

Xuifyy, 2021-05-17:
> This is not the appropriate spot for general "how do I" questions, as the template says. Please use esp8266.com or https://gitter.im/esp8266/Arduino for general programming assistance.

Ow sorry my bad, thank you for the reply