esp8266/Arduino #8050: Function with return type without 'return' crashes the ESP (is it a bug?)

atrent, 2021-05-18:
### Basic Infos

- [x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x] I have tested that the issue is present in current master branch (aka latest git).
- [x] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [ ] I have filled out all fields below.

#### Platform

- Hardware:					Wemos D2 mini
- Core Version:			3.0.0
- Development Env:		Arduino IDE
- Operating System:		Debian (bullseye)

### Settings in IDE

- Module:  			Wemos D1 mini (clone)
- Flash Mode:			dout
- Flash Size: 		4MB (FS 2MB)
- lwip Variant:   v2 Lower Memory
- Reset Method:		
- Flash Frequency:	40Mhz
- CPU Frequency:		80Mhz
- Upload Using:		SERIAL
- Upload Speed:   921600

### Problem Description

Discovered by accident:

- define a function that declares a return type
- forget the return clause
- call the function
- CRASH

see example below, if you run it on a simple Arduino UNO it works seamlessly (even if I'd expect some warning from the compiler), on my ESP it crashes (see also the stacktrace)

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp
/**
 * on Arduino OK
 * 
 * CRASHES on ESP8266!!!
 */

int crash(){
  Serial.println("crash");
  delay(500);

  // NO RETURN!!!

  //return 1; // decomment this line and it works
}

void setup() {
	delay(2000);
  Serial.begin(115200);
  Serial.println("setup");
  //crash();
}

void loop() {
  Serial.println("loop");
  crash();
}
```

### Debug Messages

```
--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Exception (0):
epc1=0x40201038 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

>>>stack>>>

ctx: cont
sp: 3ffffdf0 end: 3fffffc0 offset: 0190
3fffff80:  0001c200 00000000 3ffee53c 40201038  
3fffff90:  3fffdad0 00000000 3ffee53c 40201089  
3fffffa0:  feefeffe feefeffe 3ffee53c 402018d4  
3fffffb0:  feefeffe feefeffe 3ffe85d8 40100b55  
<<<stack<<<

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 3460, room 16 
tail 4
chksum 0xcc
load 0x3fff20b8, len 40, room 4 
tail 4
chksum 0xc9
csum 0xc9
v00041c10
~ld

```

dok-net, 2021-05-18:
It's annoying and unexpected, may have changed since the compiler was recently updated from ancient, but:
- There is a warning during compilation (`warning: no return statement in function returning non-void [-Wreturn-type]`)
- The behavior is defined in the C++ standard as undefined behavior (pun intended) according to multiple Q&As to this effect over the past years in various online forums

atrent, 2021-05-18:
> It's annoying and unexpected, may have changed since the compiler was recently updated from ancient, but:
> * There is a warning during compilation (`warning: no return statement in function returning non-void [-Wreturn-type]`)

You're right, but I had to activate warnings (strange that it's on "none" by default) explicitly in the IDE.

> * The behavior is defined in the C++ standard as undefined behavior (pun intended) according to multiple Q&As to this effect over the past years in various online forums

I did not know... thank you

Anyway I find a bit odd that on the the two platforms (i.e., Arduino and ESP) the behaviour is (un)defined in different ways.

May this difference be considered an issue?

earlephilhower, 2021-05-18:
No, undefined means "don't do this, please."  :)  

This is caused by the low-level GCC machine definition, and there is not much we can do about it at this level.

closed 2021-05-18 by earlephilhower