esp8266/Arduino #8051: Conflict with WifiMulti and ESP8266WebServerSecure libraries

cociweb, 2021-05-18:
### Basic Infos

- [X] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [X] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [X] I have tested that the issue is present in current master branch (aka latest git).
- [X] I have searched the issue tracker for a similar issue.
- [X] If there is a stack dump, I have decoded it.
- [X] I have filled out all fields below.

#### Platform

- Hardware:					[ESP-12]
- Core Version:			[Espressif 8266 (2.6.3) > Espressif ESP8266 ESP-12E]
- Development Env:		[Platformio]
- Operating System:		[Ubuntu]

### Settings in IDE

- Module:  			[Generic ESP8266 Module]
- Flash Mode:			[other]
- Flash Size: 		[4MB]
- lwip Variant:   [?]
- Reset Method:		[ck]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz]
- Upload Using:		[SERIAL]
- Upload Speed:   [115200)

### Problem Description

I try to get working the wifimulti and ESP8266WebServerSecure example codes together without success on platformio.

I would like to ask one(or more) of my ESP8266-12F (ESP8266EX) mates to reproduce the issue, to make me sure is it a bug, or I'm so noob.

I have the following symptomes:
-I have no error or warning during compile time. the start is ok, DHCP assign me valid and correct ip address.
HTTPS timeouts, but If i implement a parallel HTTP server it is ok for http only (Eg secure redirect purposes)
When I roll back to only 1 single AP, the HTTPS server works flawlessly. So the two library works great if they are divided, but the combination of them give me headaches.

Can somebody confirm me this behaviour?

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

combination of [https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266WiFi/examples/WiFiMulti/WiFiMulti.ino](WifiMulti)  and [https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266WebServer/examples/HelloServerBearSSL/HelloServerBearSSL.ino](ESP8266WebServerSecure) libraries.

### Debug Messages

There is no relevant

earlephilhower, 2021-05-19:
You've not given a real MCVE, please do so.  Also, enable full debugging.  I would bet that you are simply running out of memory because HTTPS servers are very memory intensive, and when combined with other advanced modes there just isn't space in RAM for both to work.

cociweb, 2021-05-27:
Hello,
Sorry for the delay in the response. I've tested it with the thousand of variations.
I've tried to reuce my code to create an MCVE, but I've decided to start it from scretch. That gave me soluion: the really shortened code is working but nothing more. Adding a few more lines to the mix of the samples cause OoM. 
So sorry for the inconvenience, there is nothing to do with the samples/libraries.

closed 2021-05-27 by cociweb