esp8266/Arduino #8052: v3.0.0 compile error with ESPhttpupdate

JimDrewGH, 2021-05-19:
- [X] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [X] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [X] I have tested that the issue is present in current master branch (aka latest git).
- [X] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [X] I have filled out all fields below.

- Hardware: ESP-12F
- Core Version: v3.0.0 (05/18/2021) - release
- Development Env: Arduino IDE
- Operating System:	Windows|

- Module:  Nodemcu 1.0 (and others)
- Flash Mode: qio
- Flash Size: 4MB
- lwip Variant: v2 Lower Memory (ALL other options also tested)
- Reset Method: nodemcu
- Flash Frequency: 40Mhz
- CPU Frequency: 80Mhz
- Upload Using: SERIAL
- Upload Speed: 115200

**PROBLEM**

Any code using any part of the ESP8266httpupdate class no longer compiles successfully as of v3.0.0.  It works fine with all previous versions.  Below is the error, followed by a sketch which is the example given on the current documentation page, found here:

https://arduino-esp8266.readthedocs.io/en/3.0.0/ota_updates/readme.html#http-server

**ERROR:**

[update_error.txt](https://github.com/esp8266/Arduino/files/6505361/update_error.txt)

**SKETCH:**
```

#include <ESP8266httpUpdate.h>

void setup() {
  // put your setup code here, to run once:

  DoUpdate();
}

void loop() {
  // put your main code here, to run repeatedly:

}

void DoUpdate() {
  t_httpUpdate_return ret = ESPhttpUpdate.update("192.168.0.2", 80, "/esp/update/arduino.php", "optional current version string here");

  switch (ret) {
    case HTTP_UPDATE_FAILED:
      Serial.println("[update] Update failed.");
      break;
    case HTTP_UPDATE_NO_UPDATES:
      Serial.println("[update] Update no Update.");
      break;
    case HTTP_UPDATE_OK:
      Serial.println("[update] Update ok."); // may not be called since we reboot the ESP
      break;
  }

```

earlephilhower, 2021-05-19:
What exact version of the core does your example compile on?  I see nothing like the prototype you're using in the change logs from 2.7.4 to now.

It's quite possible the docs got rusty over time, true, but we do have examples in the core that build and run w/the update and supported `update` prototypes as part of CI.

earlephilhower, 2021-05-19:
https://github.com/esp8266/Arduino/commits/master/libraries/ESP8266httpUpdate/src/ESP8266httpUpdate.h

The 2.7.4 release came out on August 2, 2020, meaning the only change to the interface file was in https://github.com/esp8266/Arduino/commit/59873908c44f0ba0bc6e8bb644ab9fcecc9814b1#diff-9827d55cbab21f79faea204a0d6956a66b5646ab091be0d4677d2c1f640e60b0

JimDrewGH, 2021-05-19:
The example compiles on EVERY version of the core from 2.3.0 to 2.7.4.  I have literally used every single version as they were released over the last 5 or so years with zero issues.  Today I updated to 3.0.0 (using the Arduino IDE's library updater function) and get the error you see.  Going back to 2.7.4 fixed it.  I then went to 2.3.0 and that worked, back 3.0.0 and it failed - back to 2.4.0 and that worked - back to 3.0.0 and it failed.   So, I am back on 2.7.4 until this issue is resolved.  I have to have the OTA function.

JimDrewGH, 2021-05-19:
Just copy and paste the above example into a new project and run the compiler.  Try 3.0.0 with the core and you will see the error, then go install the v2.7.4 core and you will see that it compiles just fine.

earlephilhower, 2021-05-19:
Per 2.7.4's compile warnings, the example used a deprecated in 2.7.4, removed in 3.0 way of accessing httpUpdate.

This is as-intended, because there are issues when the connection is not owned by the app itself.

````
/home/earle/Arduino/sketch_may18a/sketch_may18a.ino: In function 'void DoUpdate()':
/home/earle/Arduino/sketch_may18a/sketch_may18a.ino:16:133: warning: 't_httpUpdate_return ESP8266HTTPUpdate::update(const String&, uint16_t, const String&, const String&)' is deprecated (declared at /home/earle/.arduino15/packages/esp8266/hardware/esp8266/2.7.4/libraries/ESP8266httpUpdate/src/ESP8266httpUpdate.h:131) [-Wdeprecated-declarations]
  t_httpUpdate_return ret = ESPhttpUpdate.update("192.168.0.2", 80, "/esp/update/arduino.php", "optional current version string here");
````

You're going to have to update the application, or stick to the obsolete core versions.

closed 2021-05-19 by earlephilhower

JimDrewGH, 2021-05-21:
FWIW, even in verbose mode under the Arduino IDE I do not get this deprecated warning during the compile.

JimDrewGH, 2021-06-05:
BTW, you DO get a deprecated warning when using Platform I/O (which I do not use), but there is NO such warning under the Arduino IDE, even in verbose mode.

ramanraja, 2024-03-16:
Was this issue settled? I stumbled upon the same error with the same code. Then I realized that the signature of `ESPhttpUpdate.update()` has changed. Now it takes an additional parameter of a client object (like WiFiClient) before the URL.
```
WiFiClient client;
ESPhttpUpdate.update(client, url) ;
```