esp8266/Arduino #8054: Problem with FastLed since 3.0.0

Guenni75, 2021-05-19:
Since I update the ESP-8266 Board to 3.0.0 i can't compile FastLed with my AZ-Delivery D1 Mini NodeMcu with ESP8266-12F.
Every FastLed-Line gives the error:
..........libraries\FastLED\chipsets.h:###:##: warning: ISO C++17 does not allow 'register' storage class specifier [-Wregister]

ESP-8266 Board 2.7.0 works fine.

Why??

RandomSpaceship, 2021-05-21:
Have you checked that all your libraries and dependencies are up-to-date, and updated them if they weren't?

d-a-v, 2021-05-21:
The keyword `register` used by Fastled is now forbidden by c++17. 
Gcc devs think that the compiler is more clever than us about when to decide whether a variable should be mapped to a register.
This has to be mentioned to the Fastled authors.
I would advise to tell them to try this in their `.h`:
```cpp
#if __cplusplus >= 201703L
#define register // keyword 'register' is banned with c++17
#endif
```

Raul-7-7, 2021-05-22:
I had similar issue with FASTLED and ESP core 3.0.0 here is my story incase anyone in the same boat: 

Hi guys,

I just wanted to share my experience incase someone comes across the same issue. For the past few days, I had a strange issue with my ESP8266 FASTLED and WS2812B led. It has always worked without issue with Wifi and Over the Air update, until 3-4 days ago after about a year I updated my sketch and when uploaded, the "FIRST" LED was blinking (flickering) all the time!

I tried everything from, Allow_Intrupt to 0 or 1 etc, adding resistor, capacitor you name it! nothing worked!

Then I noticed the ESP BOARD was recently updated to version 3.0.0 ! in Arduino IDE I downgraded to version 2.7.4 and uploaded the exact same sketch, no issue! all working again.

Just thought I share my experience in case someone out there is searching for this !

Cheers,

d-a-v, 2021-05-22:
That's interesting @Raul-7-7, [this is similar](https://github.com/FastLED/FastLED/issues/1080) and of course I was running the git version at that time, between 2.7.4 and 3.0.0.
Given the comment there, it seems to be related with g++-10 that optimizes code differently.

Raul-7-7, 2021-05-22:
@d-a-v interesting, the post you have linked is EXACTLY what I was experiencing! any sketch, if I had WiFi on it, LED 0 flickered , no WiFi, all worked! when on Core 3.0.0 
same sketch with Wifi and OTA started to work as expected as soon as I downgraded to 2.7.4 ! been solid ever since!

Here is something I also tried, it was flickering when OFF as well (LED 0) so when you sent 
fastled.clear()
fastled.show()
all off except LED 0 flickering, I then added pinMode(PIN, INPUT); 
and it stopped the flickering when ALL LED off, however, when I had to change the pinmode to output , first LED flickering back! 

Anyway, not sure what is causing it exactly, but for now. 2.7.4 is working great without the need to change pinmode or any other modification to the code.

cptskippy, 2021-05-26:
> I had similar issue with FASTLED and ESP core 3.0.0 here is my story incase anyone in the same boat:
>
> Hi guys,
>
> I just wanted to share my experience incase someone comes across the same issue. For [...]

Same.

I have two different projects based on the ESP-01 that have WS2812B LED strips connected to GPIO2.  One is driving the LEDs at 3.3v and the other is using a level shifter to drive them at 5v.  I upgraded both to v3 of the ESP8266 Arduino core library and LED 0 will show a random color and flicker. When I downgrade back to v2.7.4 the problem goes away.

Both projects are using ESP8266WiFi, ESP8266mDNS, WiFiClient, ArduinoOTA, and Espalexa.

Tech-TX, 2021-05-26:
@d-a-v David, I picked up a short 1m strip of WS2812B LEDs a couple of months ago if you'd like scope / logic analyzer captures of what's going on there.  I can rig a photodiode or phototransistor over LED0 to use the 'flicker' as a trigger.  Scope and logic analyzer are separate units, so I can't synchronize the two at the moment.  I'm still waiting on the logic analyzer pod for my scope to arrive.  

I've seen that LED0 flicker mentioned elsewhere around the 'net before 3.0.0; one person used a 'sacrificial' WS2812B chip as a level-shifter to drive the rest of the strip to get around that.

d-a-v, 2021-05-26:
I'm not sure the issue is because of level shifter. Other ws281x libraries, maybe not as fast, work without flickering when WiFi is enabled.
It would not be surprising that it had to do with fine tuning as seen in files from `FastLED/src/platforms/esp/8266/` due to the gcc update.
One has to play with these files and gcc directives / pragmas, debugging with a scope or a logic analyzer would help someone who knows their code well.
@CRCinAU [seems to have spent some time on this](https://github.com/FastLED/FastLED/issues/1080#issuecomment-705610290) may have more insights. FastLed folks seems to be not interested in this issue yet, though.

Tech-TX, 2021-05-26:
I'll take a look this weekend,  purely comparing the signals into the first LED vs. what it feeds to the next in the string.  I have some old tricks if I see fast glitches hitting the first LED.

sblantipodi, 2021-05-30:
I can compile my code with lots of warning but core 3.0 destroyed my FastLED procjets, it flickers like a mess, pretty unusable.

sascha432, 2021-06-20:
https://github.com/FastLED/FastLED/issues/1237

It seems that the show() method has a timing issue, replacing it with my own code makes it work again.
Unfortunately my show method does not support all FastLED feastures, so I am not going to make it public and the issue with the show() seems pretty akward, at least nothing I could find there to fix it.

Since the last comment at the bug report is just 4 days old we might see a fix soon.