esp8266/Arduino #8055: HTTPClient writeToStream times out with 3.0.0

TheNitek, 2021-05-19:
----------------------------- Delete below -----------------------------

If your issue is a general question, starts similar to "How do I..", is related to 3rd party libs, or is related to hardware, please discuss at a community forum like esp8266.com.

INSTRUCTIONS
============
If you do not follow these instructions, your issue may be dismissed.

1. Follow the checklist under Basic Infos and fill in the [ ] spaces with an X.
2. Fill in all the fields under Platform and Settings in IDE marked with [ ] (pick the correct option for you in each case, delete the others).
3. If you haven't already done so, test your issue against current master branch (aka latest git), because it may have been already fixed.
4. Describe your problem.
5. If you have a STACK DUMP decode it:

https://arduino-esp8266.readthedocs.io/en/latest/Troubleshooting/stack_dump.html

6. Include a Minimal Complete Reproducible Example sketch that shows your issue. Do not include your entire project, or a huge piece of code.
7. Include debug messages:

https://arduino-esp8266.readthedocs.io/en/latest/Troubleshooting/debugging.html 

8. Use markup (buttons above) and the Preview tab to check what the issue will look like.
9. Delete these instructions from the above to the below marker lines before submitting this issue.

----------------------------- Delete above -----------------------------

### Basic Infos

- [ ] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [ ] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [ ] I have tested that the issue is present in current master branch (aka latest git).
- [ ] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [ ] I have filled out all fields below.

#### Platform

- Hardware:					[ESP-12|ESP-01|ESP-07|ESP8285 device|other]
- Core Version:			[latest git hash or date]
- Development Env:		[Arduino IDE|Platformio|Make|other]
- Operating System:		[Windows|Ubuntu|MacOS]

### Settings in IDE

- Module:  			[Generic ESP8266 Module|Wemos D1 mini r2|Nodemcu|other]
- Flash Mode:			[qio|dio|other]
- Flash Size: 		[4MB/1MB]
- lwip Variant:   [v1.4|v2 Lower Memory|Higher Bandwidth]
- Reset Method:		[ck|nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz|160MHz]
- Upload Using:		[OTA|SERIAL]
- Upload Speed:   [115200|other] (serial upload only)

### Problem Description

Detailed problem description goes here.

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp

#include <Arduino.h>

void setup() {
  HTTPClient httpClient;
  std::unique_ptr<BearSSL::WiFiClientSecure>client(new BearSSL::WiFiClientSecure);
  client->setInsecure();

  httpClient.begin(*client, "https://mediandr-a.akamaihd.net/download/podcasts/podcast4223/AU-20210510-1944-4900.mp3");
  int httpCode = httpClient.GET();
  if (httpCode != HTTP_CODE_OK) {
    Serial.printf("invalid response code: %d\n", httpCode);
    httpClient.end();
    return;
  }

  sdfat::File32 episodeFile;
  if(!episodeFile.open("/download.tmp", sdfat::O_WRITE | sdfat::O_CREAT | sdfat::O_TRUNC)) {
    Serial.println(F("Failed to open target file"));
    httpClient.end();
    return;
  }

  int byteCount = httpClient.writeToStream(&episodeFile);
  Serial.printf("Downloaded %d bytes\n", byteCount);

}

void loop() {

}

```

### Debug Messages

```
Debug messages go here
```

TheNitek, 2021-05-22:
After further debugging:
StreamSend.SendGenericPeekBuffer times out because SDFat doesn't seem to override availableForWrite(), so it will return 0 everytime and no write will ever be attempted.

d-a-v, 2021-05-22:
I think @earlephilhower 's [fork made for this core](https://github.com/earlephilhower/ESP8266SdFat/commit/0a46e4ebb2de585c5a64f981dbc2b2223a438984) already helps.
It seems little is needed for `::availableForWrite()` to be implemented

TheNitek, 2021-05-22:
I guess this would degrade performance by quite a lot

Wouldn't it be an option to replace
https://github.com/esp8266/Arduino/blob/8dc80b5ef6739b28917a12da9a1c10ce3741971a/cores/esp8266/StreamSend.cpp#L85
with
`w = w ? std::min(w, avpk) : avpk;`

Comment for `::availableForWrite()` is
// default to zero, meaning "a single write may block"
// should be overriden by subclasses with buffering
so always returning 0 seems to be valid case which is not handled correctly by `::SendGenericPeekBuffer`

d-a-v, 2021-05-22:
`0` is a valid and possible value.
For example network can be blocking for immediate send, and FS can tell this way that SDcard is full.
Maybe @earlephilhower has some ideas about SdFat and `availableforWrite()` performance ?

TheNitek, 2021-05-22:
But that means that if Print* is like "I can never guarantee a non-blocking write" (i.e. always return `0`) it can never be used as the target of a stream. That does not sound correct to me.

d-a-v, 2021-05-22:
That's true. So `availableForWrite()` has to be always implemented. It should be pure virtual.
Arduino API is a very sensible beast to touch though.

The stream::send API is currently unusable when `::availableForWrite()` is not implemented.
It could be updated for such cases. Let's first try to solve the SdFat issue.

TheNitek, 2021-05-22:
> I think @earlephilhower 's [fork made for this core](https://github.com/earlephilhower/ESP8266SdFat/commit/0a46e4ebb2de585c5a64f981dbc2b2223a438984) already helps.
> It seems little is needed for `::availableForWrite()` to be implemented

I just took a closer look at this and it looks like performance impact might be nearly not existing. So I'll give this a try.

TheNitek, 2021-05-22:
`::availableSpaceForWrite()` also returns `0` all the time, returning at this line:
https://github.com/earlephilhower/ESP8266SdFat/blob/9f94df41ccdb064ff4cb98d7b2737a369f6be1a0/src/FatLib/FatFile.cpp#L1514

d-a-v, 2021-05-23:
`::availableSpaceForWrite()` is in fact already [used by `::availableForWrite()`](https://github.com/esp8266/Arduino/blob/60fe7b4ca8cdca25366af8a7c0a7b70d32c797f8/libraries/SDFS/src/SDFS.h#L280-L283).

I made a try and I get `0` too.

TheNitek, 2021-06-18:
In Case someone needs a workaround, I just use my own StreamFile" now:
```
    class StreamFile: public File32 {
      int availableForWrite() override {
        return 512;
      }
    };
```

d-a-v, 2021-06-23:
@TheNitek Can you try with #8167 ?

PR #8167 (2021-06-25): SDFS: fix AvailableForWrite: do not always return 0 when space is available
fixes #8055
closed 2021-06-25 by d-a-v