esp8266/Arduino #8058: ESP8266httpUpdate fails to compile with v3.0.0

JimDrewGH, 2021-05-20:
- [X] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [X] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [X] I have tested that the issue is present in current master branch (aka latest git).
- [X] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [X] I have filled out all fields below.

Hardware: ESP-12E, ESP-12F, ESP-07
Core Version: v3.0.0 - (tested as of 05/20/2021)
Development Env: Arduino IDE (1.8.13)
Operating System: Windows, Mac, Linux
### Problem Description

I have used every revision of the ESP8266 core (starting with v2.3.0) with the same project code in numerous commercial applications.  All of these applications use the over-air update capability via ESPhttpUpdate.  When updating to v3.0.0 of the core (through the Arduino IDE's board manager), I find that NONE of my applications will compile due to an error with anything that involves ESPhttpUpdate.  I used the example code found in the core's OTA documentation to base my code on, and those examples also do not compile under v3.0.0 (but do compile with *any* of the previous core versions - I tested them all!)   See the simple and advanced updater section here: **https://arduino-esp8266.readthedocs.io/en/latest/ota_updates/readme.html#http-server**

So, again anything that attempts to use any part of the ESPhttpUpdate class generates an error under the v3.0.0 core.

### Error

_no matching function for call to 'ESP8266HTTPUpdate::update(const char [46])'_

To easily replicate this error, drop this super simple sketch into your Arduino IDE and try to compile it:

```
#include <ESP8266WiFi.h>
#include <ESP8266httpUpdate.h>

void setup() {

  t_httpUpdate_return ret = ESPhttpUpdate.update("http://www.arduino.cc/3p_updates/IoT1172.bin");
}

void loop() {

}
```

If you use the board manager and switch back and forth between core v3.0.0 and any other version you will see that all earlier versions compile just fine, and v3.0.0 generates the same error shown above.  I have tested this on multiple Arduino IDE installations on Windows, Mac, and even Linux (RPi4 mini computer), all with the exact same results.  I did not try different Arduino IDE versions.  I am using v1.8.13 on all of my development setups.

I also tried to use the syntax shown in the documentation example - using a URL, port, and file.  I have been using the single URL method for years now with zero issues.  But for fun, I tried that as well with a similar compiler fail.  Something has apparently changed that has broken the ESPhttpUpdate class.

d-a-v, 2021-05-20:
This change was indeed not marked as "breaking change".
I just updated the release comments so removals such as this one are more visible.
(https://github.com/esp8266/Arduino/releases/tag/3.0.0)

closed 2021-05-20 by d-a-v

JimDrewGH, 2021-05-21:
Thanks... for some reason, I don't see a "closed" section unless I click a link that brings me here (like from your message in the community).  Weird.... bookmarked now though!

JimDrewGH, 2021-05-21:
Scratch that... it's just not very intuitive that you need to click on text that is greyed out to switch to the closed area.