esp8266/Arduino #8065: Making a Http post request using Esp8266 ( aws )

Friday8229, 2021-05-21:
### Basic Infos

- [ X] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [X ] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [ X] I have tested that the issue is present in current master branch (aka latest git).
- [X ] I have searched the issue tracker for a similar issue.
- [ X] If there is a stack dump, I have decoded it.
- [X ] I have filled out all fields below.

#### Platform

- Hardware:					[ESP-12E|ESP8266 device]
- Core Version:			[2.3.0]
- Development Env:		[Arduino IDE]
- Operating System:		[Ubuntu]

### Settings in IDE

- Module:  			[Nodemcu]
- Flash Mode:			[dio]
- Flash Size: 		[4MB]
- lwip Variant:   [v1.4|v2 Lower Memory|Higher Bandwidth]
- Reset Method:		[nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz]
- Upload Using:		[SERIAL]
- Upload Speed:   [115200]

### Problem Description

Hi i am trying to make a http post request from an esp8266. The attached image shows how i do this using postmaster, 
[image]
what i want to do is send two keys namely serialnumber and poweroutput to the aws server and get the response. When i use postmaster to do the same everything works fine but when i use my nodemcu it Responds with a httpcode: -1, It should respond with the current status and timestamp like in the  postman 
 screenshot. Am i using the library in a wrong way? The code that i am using is attached below. Any help is appreciated.

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp

#include <ESP8266HTTPClient.h>
#include <ESP8266WiFi.h>
 
void setup() {
 
  Serial.begin(115200);                 //Serial connection
  WiFi.begin("Jinhit_Entertainment", "10813ad0908");   //WiFi connection
 
  while (WiFi.status() != WL_CONNECTED) {  //Wait for the WiFI connection completion
 
    delay(500);
    Serial.println("Waiting for connection");
 
  }
 
}
 
void loop() {
 
  if (WiFi.status() == WL_CONNECTED) { //Check WiFi connection status
 
    HTTPClient http;    //Declare object of class HTTPClient
 
    http.begin("https://uczj81sy9d.execute-api.us-east-1.amazonaws.com/prod/inverterdata");      //Specify request destination
    http.addHeader("Content-Type", "text/plain");  //Specify content-type header

    int httpCode = http.POST("serialnumber: 4001, poweroutput: 4.1");   //Send the request
    String payload = http.getString();                  //Get the response payload
 
    Serial.println(httpCode);   //Print HTTP return code
    Serial.println(payload);    //Print request response payload
 
    http.end();  //Close connection
 
  } else {
 
    Serial.println("Error in WiFi connection");
 
  }
 
  delay(30000);  //Send a request every 30 seconds
 
}

```

d-a-v, 2021-05-23:
Please fill-in the issue template.
Please try with the latest version of this core, which you're not using, and enable HTTPClient debug option and post logs.
Please use the same parameters in your postman request and your aduino sketch: You used `text/plain` here and `application/json` there.

earlephilhower, 2021-06-18:
No feedback in one month.  Closing.

closed 2021-06-18 by earlephilhower