esp8266/Arduino #8066: Ticker.h - big float and unt32_t numbers

enjoyneering, 2021-05-21:
#### Platform

- Hardware:					[ESP-12]
- Core Version:			[2.7.4] & [3.0.0]
- Development Env:		[Arduino IDE]
- Operating System:		[Windows]

### Settings in IDE

- Module:  			[Generic ESP8266]
- Flash Mode:			[qio|dio|other]
- Flash Size: 		[4MB]
- lwip Variant:   [v2 Higher Bandwidth]
- Reset Method:		[nodemcu]
- Flash Frequency:	[80Mhz]
- CPU Frequency:		[80Mhz]
- Upload Using:		[OTA|SERIAL]
- Upload Speed:   [115200] 

### Problem Description

Any Ticker tasks fire OK if time is short (tested up to 60 seconds (60 000ms)). However, when I try to control the brightness after sunset, the time gets big (for example 44944 sec (44 944 000ms)) & the same task no longer start. The code below is not exactly what I have in my project, but close.

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp

#include <Arduino.h>

Ticker ticker_autoBrightness;

uint32_t timeSec = 44079;

void setup() {
pinMode(2, OUTPUT);

task_clockDayBrightness();
}

void loop() {
//empty
}

void task_clockDayBrightness()
{
  digitalWrite(2, LOW) //turn the LED on

  ticker_autoBrightness.once_scheduled(timeSec, task_clockNightBrightness);
}

void task_clockNightBrightness()
{
  digitalWrite(2, HIGH) //turn the LED off

  ticker_autoBrightness.once_scheduled(timeSec, task_clockDayBrightness);
}

mobizt, 2021-05-22:
The allowed ranges of software timer (used in Ticker library) are 5 milliseconds to 1.9 hours (timing is in millisecond).
https://github.com/esp8266/Arduino/blob/da6ec83b5fdbd5b02f04cf143dcf8e158a8cfd36/libraries/Ticker/src/Ticker.cpp#L48

See [ESP8266 Non-OS SDK API Ref](https://www.espressif.com/sites/default/files/documentation/2c-esp8266_non_os_sdk_api_reference_en.pdf) at section 3.1.1. os_timer_arm

enjoyneering, 2021-05-22:
Thank you mobizt. Didn't know about it.  Float and uint32_t confused me. This info should be in the doc.

`once_scheduled(float seconds, callback_function_t callback)`
`void _attach_ms(uint32_t milliseconds, bool repeat)`

earlephilhower, 2021-06-03:
Updated the title a bit.  I haven't dug into it, but if Ticker is passed a value it can't actually support it should generate an error somehow, not just silently do nothing so it's somewhere between a bug and enhancement need IMHO.

dok-net, 2021-06-03:
@earlephilhower Ticker has all "void" functions. Short of changing all of them, there are two solutions that come to mind:
- `assert()` in `Ticker::_attach_ms()`
- In `Ticker::_attach_ms()`, if the duration is not supported, set `_timer = nullptr;`, which in turn makes `bool Ticker::active() const` `false`, something a user can check for after calling any of the `attach...` functions.

closed 2022-11-01 by mcspr