esp8266/Arduino #8068: AutoConnect: error: no matching function for call to 'EEPROMClass::EEPROMClass()'

pete68, 2021-05-22:
Hello

I have this problem: compiling any program that uses Hieromon's Autoconnect library gives me the error: "AutoConnect \ src \ AutoConnectCredential.cpp: 55: 21: error: no matching function for call to 'EEPROMClass :: EEPROMClass () ' 55 | _eeprom.reset (new EEPROMClass); "

any suggestions?
thank you

Hieromon, 2021-05-22:
AutoConnect does not yet support core 3.0.0. I'm currently working on that task. I will support the core 3.0.0  in the next turn. (AutoConnect v1.3.0 is planned)
https://github.com/Hieromon/AutoConnect/issues/361

pete68, 2021-05-22:
3.0.0 of what? of the esp8266 board? because my arduino ide does not show versions lower than 2.0.0

Hieromon, 2021-05-22:
It the version of the ESP8266 arduino core you are using. You can check it in the Arduino IDE Board Manager.
In any case, please post to the discussion or issue in the AutoConnect repository first. In that case, please refer to the [AutoConnect documentation](https://hieromon.github.io/AutoConnect/faq.html#4-reports-the-issue-to-autoconnect-github-repository) and enter the necessary information.

earlephilhower, 2021-06-03:
No MCVE was given, the template was ignored, etc.  Closing.

FWIW, `EEPROM::reset()` is NOT in the Arduino library documentation and not supported (3.0.0 or prior) in the ESP8266.  https://www.arduino.cc/en/Reference/EEPROM

closed 2021-06-03 by earlephilhower