esp8266/Arduino #8070: Simple add to String variable results in saving garbage to file

jmmprodukts, 2021-05-22:
### Basic Infos

- [X] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [X] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [X] I have tested that the issue is present in current master branch (aka latest git).
- [X] I have searched the issue tracker for a similar issue.
- [X] If there is a stack dump, I have decoded it.
- [X] I have filled out all fields below.

#### Platform

- Hardware:			ESP8266
- Core Version:			3.0.0
- Development Env:		Arduino IDE 1.8.13
- Operating System:		Windows

### Settings in IDE

- Module:  		Wemos D1 mini
- Flash Mode:		[qio|dio|other]
- Flash Size: 		16MB
- lwip Variant:   	[v1.4|v2 Lower Memory|Higher Bandwidth]
- Reset Method:		[ck|nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:	[80Mhz]
- Upload Using:		OTA
- Upload Speed:   [115200|other] (serial upload only)

### Problem Description

Simple add to String variable results in saving garbage to file

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

Code below adds "1\r\n" or "0\r\n" to String_to_Save String.

  if (Irms > Log_threshold){
    String_to_Save += "1\r\n";
  }else{
    String_to_Save += "0\r\n";
  }

Saved String however sometimes looks like this:

	2021-05-22 14:26:56;0,531;6,056;157,469;ÿÿÿ2021-05-22 14:27:07;0,523;6,255;162,621;1
	2021-05-22 14:27:17;0,521;5,253;136,569;1
	2021-05-22 14:27:27;0,528;6,275;163,163;1

Where:

	2021-05-22 14:26:56;0,531;6,056;157,469;1
	2021-05-22 14:27:07;0,523;6,255;162,621;1
	2021-05-22 14:27:17;0,521;5,253;136,569;1
	2021-05-22 14:27:27;0,528;6,275;163,163;1

is expected. for some unknown reason the "1\r\n" part is lost and "ÿÿÿ" (3 times value 255) is saved instead

below complete sample:

	2021-05-22 14:26:56;0,531;6,056;157,469;ÿÿÿ2021-05-22 14:27:07;0,523;6,255;162,621;1
	2021-05-22 14:27:17;0,521;5,253;136,569;1
	2021-05-22 14:27:27;0,528;6,275;163,163;1
	2021-05-22 14:27:37;0,530;5,609;145,831;ÿÿÿ2021-05-22 14:27:47;0,522;5,409;140,635;ÿÿÿ2021-05-22 14:27:58;0,520;5,811;151,093;1
	2021-05-22 14:28:08;0,527;5,556;144,462;1
	2021-05-22 14:28:18;0,524;5,609;145,839;1
	2021-05-22 14:28:29;0,525;5,983;155,570;ÿÿÿ2021-05-22 14:28:40;0,527;5,306;137,958;ÿÿÿ2021-05-22 14:28:50;0,521;5,374;139,732;1
	2021-05-22 14:29:00;0,523;5,505;143,120;1
	2021-05-22 14:29:10;0,527;5,883;152,959;1
	2021-05-22 14:29:20;0,528;5,498;142,954;ÿÿÿ2021-05-22 14:29:30;0,529;5,228;135,916;ÿÿÿ2021-05-22 14:29:40;0,529;5,367;139,553;1
	2021-05-22 14:29:51;0,526;5,252;136,542;1
	2021-05-22 14:30:02;0,520;5,877;152,799;1
	2021-05-22 14:30:12;0,515;5,623;146,189;ÿÿÿ2021-05-22 14:30:22;0,521;5,908;153,600;ÿÿÿ2021-05-22 14:30:32;0,515;5,949;154,677;1
	2021-05-22 14:30:43;0,519;5,362;139,410;1
	2021-05-22 14:30:53;0,520;5,559;144,539;ÿÿÿ2021-05-22 14:31:03;0,526;5,484;142,578;1
	2021-05-22 14:31:14;0,528;5,125;133,246;ÿÿÿ2021-05-22 14:31:24;0,527;5,030;130,775;1
	2021-05-22 14:31:34;0,526;5,520;143,510;1
	2021-05-22 14:31:44;0,525;5,772;150,063;ÿÿÿ2021-05-22 14:31:54;0,522;5,230;135,983;ÿÿÿ2021-05-22 14:32:05;0,523;5,624;146,234;1
	2021-05-22 14:32:15;0,521;5,785;150,417;1
	2021-05-22 14:32:25;0,520;5,463;142,045;1
	2021-05-22 14:32:35;0,522;6,057;157,479;ÿÿÿ2021-05-22 14:32:45;0,529;5,620;146,113;ÿÿÿ2021-05-22 14:32:55;0,524;5,288;137,479;1
	2021-05-22 14:33:05;0,526;5,539;144,007;1
	2021-05-22 14:33:15;0,530;5,234;136,091;1
	2021-05-22 14:33:25;0,532;5,862;152,412;ÿÿÿ2021-05-22 14:33:35;0,528;5,049;131,269;ÿÿÿ2021-05-22 14:33:46;0,530;5,695;148,068;1
	2021-05-22 14:33:56;0,530;5,637;146,552;1
	2021-05-22 14:34:06;0,530;5,255;136,624;1
	2021-05-22 14:34:16;0,530;5,046;131,199;ÿÿÿ2021-05-22 14:34:26;0,529;5,355;139,228;ÿÿÿ2021-05-22 14:34:36;0,530;5,543;144,114;1
	2021-05-22 14:34:46;0,531;5,053;131,369;1
	2021-05-22 14:34:56;0,530;5,439;141,424;ÿÿÿ2021-05-22 14:35:06;0,525;5,241;136,257;1
	2021-05-22 14:35:17;0,525;5,348;139,048;ÿÿÿ2021-05-22 14:35:27;0,528;5,259;136,723;1
	2021-05-22 14:35:37;0,531;5,540;144,028;1

### Debug Messages

none

earlephilhower, 2021-05-22:
Please post a complete MCVE, the small snippet is not enough for anyone else to run anything.  Also, try running with full debugging options and catch the logs.  Sometimes, especially with Strings, there are out of memory issues which don't cause the app to crash, but which to end up corrupting Strings.

jmmprodukts, 2021-05-23:
After some debugging found that just before writing to file (SPIFFS here), the contents of the String is as expected (ending with 1\r\n). however after reading the file, for some unknown reason, the 1\r\n is manifested as ÿÿÿ (3 times 0xFF).

After replacing the (deprecated) SPIFFS with LIttleFS the phenomenon has disappeared

Code snippet:

      DEBUG_PRINTERHans.print("String_to_Save = #" + String_to_Save + "#\r\n");
      f.print(String_to_Save); 
      f.close();

End conclusion:

There seems to be some conflicting interaction between the 3.0.0 core and the deprecated SPIFFS

closed 2021-05-23 by jmmprodukts