esp8266/Arduino #8072: getCoreVersion results in empty String

jmmprodukts, 2021-05-23:
### Basic Infos

- [ ] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [ ] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x] I have tested that the issue is present in current master branch (aka latest git).
- [x] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [x] I have filled out all fields below.

#### Platform

- Hardware:					[ESP-12|ESP-01|ESP-07|ESP8285 device|other]
- Core Version:			[3.0.0]
- Development Env:		[Arduino IDE]
- Operating System:		[Windows]

### Settings in IDE

- Module:  			[Wemos D1 mini]
- Flash Mode:			[qio|dio|other]
- Flash Size: 		[4MB/1MB]
- lwip Variant:   [v1.4|v2 Lower Memory|Higher Bandwidth]
- Reset Method:		[ck|nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz]
- Upload Using:		[SERIAL]
- Upload Speed:   [115200)

### Problem Description

getCoreVersion results in empty String

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

void setup() {
Serial.begin(115200);
  Serial.println("");  Serial.println("");  Serial.println("");
  Serial.println("Booting : ");
  Serial.println("");
  display_Running_Sketch_Serial();
}

void display_Running_Sketch_Serial(void){   // displays at startup the Sketch running in the Arduino
  //String  Compiled = String(F(__DATE__))+ ", "+String(F(__TIME__))+", "+String(F(__VERSION__));
  //int  Arduino_IDE_version=ARDUINO;
  Serial.print("Sketch Compiled : #" + String(F(__DATE__))+ ", "+String(F(__TIME__))+", "+String(F(__VERSION__)));                  Serial.print("#\r\n");
  Serial.print("Arduino version : #" + String (ARDUINO));                   Serial.print("#\r\n");
  Serial.print("ESP SdkVersion  : #");  Serial.print(ESP.getSdkVersion());  Serial.print("#\r\n");
  Serial.print("ESP CoreVersion : #");  Serial.print(ESP.getCoreVersion()); Serial.print("#\r\n");
  Serial.print("ESP FullVersion : #");  Serial.print(ESP.getFullVersion()); Serial.print("#\r\n");
}

void loop() {
  // put your main code here, to run repeatedly:
}

Serial output:

Booting : 

Sketch Compiled : #May 23 2021, 10:01:19, 10.2.0#
Arduino version : #10813#
ESP SdkVersion  : #2.2.2-dev(38a443e)#
ESP CoreVersion : ##     <== NOTHING (EMPTY STRING) HERE
ESP FullVersion : #SDK:2.2.2-dev(38a443e)/Core:3.0.0=30000000/lwIP:STABLE-2_1_2_RELEASE/glue:1.2-48-g7421258/BearSSL:c0b69df#

```
Debug messages go here
```

dok-net, 2021-05-23:
### Done with a Git worktree, checkout is master, revision at tagged 3.0.0

At first, the MCVE sketch doesn't compile - removing
```
Serial.print("Sketch Compiled : #" + String(F(DATE))+ ", "+String(F(TIME))+", "+String(F(VERSION))); Serial.print("#\r\n");
```
helps with that.
Running results in this on the serial terminal:
```
Booting : 

Arduino version : #108010#
ESP SdkVersion : #2.2.2-dev(38a443e)#
ESP CoreVersion : #efb0341a#
ESP FullVersion : #SDK:2.2.2-dev(38a443e)/Core:3.0.0=30000000/lwIP:STABLE-2_1_2_RELEASE/glue:1.2-48-g7421258/BearSSL:c0b69df#
```

### Done with Arduino board manager BSP version 3.0.0
Just as invalid as with Git checkout:
```
Serial.print("Sketch Compiled : #" + String(F(DATE))+ ", "+String(F(TIME))+", "+String(F(VERSION))); Serial.print("#\r\n");
```
Removed that line, again.
Running:
```
Booting : 

Arduino version : #108010#
ESP SdkVersion : #2.2.2-dev(38a443e)#
ESP CoreVersion : ##
ESP FullVersion : #SDK:2.2.2-dev(38a443e)/Core:3.0.0=30000000/lwIP:STABLE-2_1_2_RELEASE/glue:1.2-48-g7421258/BearSSL:c0b69df#
```

dok-net, 2021-05-23:
@d-a-v I don't know about the packaging process, do you have an idea what's wrong?

jmmprodukts, 2021-05-23:
`void setup() {`
`  Serial.begin(115200);`
`  Serial.println("");  Serial.println("");  Serial.println("");`
`  Serial.println("Booting : ");`
`  Serial.println("");`
`  display_Running_Sketch_Serial();`
`}`
``
`void display_Running_Sketch_Serial(void){   // displays at startup the Sketch running in the Arduino`
 ` //String  Compiled = String(F(__DATE__))+ ", "+String(F(__TIME__))+", "+String(F(__VERSION__));`
`  //int  Arduino_IDE_version=ARDUINO;`
 ` Serial.print("Sketch Compiled : #" + String(F(__DATE__))+ ", "+String(F(__TIME__))+", "+String(F(__VERSION__)));`
`Serial.print("#\r\n");`
`  Serial.print("Arduino version : #" + String (ARDUINO));                   Serial.print("#\r\n");`
`  Serial.print("ESP SdkVersion  : #");  Serial.print(ESP.getSdkVersion());  Serial.print("#\r\n");`
`  Serial.print("ESP CoreVersion : #");  Serial.print(ESP.getCoreVersion()); Serial.print("#\r\n");`
`  Serial.print("ESP FullVersion : #");  Serial.print(ESP.getFullVersion()); Serial.print("#\r\n");`
`}`
``
`void loop() {`
`  // put your main code here, to run repeatedly:`
`}`

jmmprodukts, 2021-05-23:
Plain pasting the code seems to remove underscores. so again the code here enclosed within Back Quotes:

`void setup() {`
`  Serial.begin(115200);`
`  Serial.println("");  Serial.println("");  Serial.println("");`
`  Serial.println("Booting : ");`
`  Serial.println("");`
`  display_Running_Sketch_Serial();`
`}`
``
`void display_Running_Sketch_Serial(void){   // displays at startup the Sketch running in the Arduino`
 ` //String  Compiled = String(F(__DATE__))+ ", "+String(F(__TIME__))+", "+String(F(__VERSION__));`
`  //int  Arduino_IDE_version=ARDUINO;`
 ` Serial.print("Sketch Compiled : #" + String(F(__DATE__))+ ", "+String(F(__TIME__))+", "+String(F(__VERSION__)));`
`Serial.print("#\r\n");`
`  Serial.print("Arduino version : #" + String (ARDUINO));                   Serial.print("#\r\n");`
`  Serial.print("ESP SdkVersion  : #");  Serial.print(ESP.getSdkVersion());  Serial.print("#\r\n");`
`  Serial.print("ESP CoreVersion : #");  Serial.print(ESP.getCoreVersion()); Serial.print("#\r\n");`
`  Serial.print("ESP FullVersion : #");  Serial.print(ESP.getFullVersion()); Serial.print("#\r\n");`
`}`
``
`void loop() {`
`  // put your main code here, to run repeatedly:`
`}`

jandrassy, 2021-05-23:
it is this https://github.com/esp8266/Arduino/issues/8049

jmmprodukts, 2021-05-23:
Very likely #8049

d-a-v, 2021-05-23:
closing as duplicate, please follow #8049
We're working on it, it will hopefully be fixed with release 3.0.1.

closed 2021-05-23 by d-a-v