esp8266/Arduino #8084: Using EC Client Certificates with WifiClientSecure

matthiasmoke, 2021-05-29:
Hello,
I got a question about the Client Certificates section in the documentation, specifically about `setClientECCert()` method.
(https://arduino-esp8266.readthedocs.io/en/latest/esp8266wifi/bearssl-client-secure-class.html#setclientrsacert-setclienteccert)

There, it is stated, that for using EC certs, additional information is required as shown in the prototype. That leads me to my question:
Which prototype? If it already exists, it would be a great help if you could link the prototype there.
If not, could you provide an usage example for setting EC client certs?

d-a-v, 2021-05-29:
Is that what you're looking for ?
https://github.com/esp8266/Arduino/search?q=setClientECCert&type=code

matthiasmoke, 2021-05-29:
Kind of, I already found that code parts. But what I'm curious about is the actual usage of the additional parameters `allowed_usages` and `cert_issuer_key_type`. In other words, in which format those params need to be provided.
I thought there my maybe is an example in the docs that is not linked.

earlephilhower, 2021-05-29:
This really isn't the best spot to ask general questions, as the template said.  https://esp8266.com or https://gitter.im/esp8266/Arduino are more appropriate.

That said, EC certs are kind of a pain in the butt.  They are slower than RSA ones on the 8266 by a lot, for starters.  And they require add'l info about them not put in the key.  There is a working EC cert server in the examples, FWIW.

````
void setClientECCert(const X509List *cert, const PrivateKey *sk, unsigned allowed_usages, unsigned cert_issuer_key_type)
````

The two extra fields are either `BR_KEYTYPE_KEYX`` and/or `BR_KEYTYPE_SIGN`.  They should be known upon creation of the EC key.

For more detailed info you'll need to dig into bearssl.org or other public key references.  Good luck./

closed 2021-05-29 by earlephilhower

matthiasmoke, 2021-07-04:
For anyone who stumbles upon the same problems, I found out how the additional parameters need to be provided:
- `allowed_usages`: a combination of usages, namely BR_KEYTYPE_KEYX and/or BR_KEYTYPE_SIGN
- `cert_issuer_key_type`: the algorithm type of the key used by the issuing CA to sign the server's certificate. The value is either BR_KEYTYPE_RSA or BR_KEYTYPE_EC.

@earlephilhower We are currently doing measurements for encrypted MQTT communication between an ESP8266 and some broker (in our case a Raspberry Pi running mosquitto). As I got ECC running, I compared TLS initialization time for both, RSA and EC certs. As theory suggests, ECC (~ 3.6s) was a lot faster compared to RSA (~ 5.8s).
Do you still know the details of the circumstances in which RSA was faster than ECC?