esp8266/Arduino #8085: Proxy support for ESP8266HTTPClient

ChrisSmith, 2021-05-30:
### Problem Description

I would be nice if the http client supported http proxies. Adding proxy support is relatively straightforward as its done using headers. 

Regular HTTP request (connection to target 1.1.1.1)
```
GET / HTTP/1.1
Host: 1.1.1.1:8080
User-Agent: curl/7.68.0
```

Request to a proxy (connection to proxy 2.2.2.2)
```
GET http://1.1.1.1:8000/ HTTP/1.1
Host: 1.1.1.1:8000
User-Agent: curl/7.68.0
```

The primary difference is that the tcp connection doesn't match the host header and the full uri is used in the request line. Authorization could be manually performed using the current `addHeader()` method to add the `Proxy-Authorization` header.

I have working changes locally that I'd be happy to contribute if the maintainers think this is a worthwhile feature.

d-a-v, 2021-05-30:
Your Pull-Request is welcome !

FAAAAT, 2021-06-03:
Hello ChrisSmith,

I have a proxy request similar as yours, but it's for https requests through http proxy. Is your PR support https request?

Thanks.

ChrisSmith, 2021-06-03:
@FAAAAT are you referring to 8086? My implementation doesn't support the TCP tunneling required for CONNECT

I don't think thats required for a https proxy though, an alternative implementation would perform tls termination at the proxy which would be possible to support in my implementation but I have not tried it yet.