esp8266/Arduino #8086: Proxy support request

FAAAAT, 2021-05-30:
Hello guys,

I found that there is no http proxy support when using WiFiClient and WiFiClientSecure. 

So, is there any alternative way to use http proxy? Or must implement one by my self?

Thanks.

d-a-v, 2021-05-30:
#8085 ?

FAAAAT, 2021-05-31:
> #8085 ?

Thanks d-a-v. 

I'm using websockets on my project, and no proxy support stopping me. I'm making a local change to let WiFiClient and WiFiClientSecure support http proxy. 

So, are there any guide on how to build and test ArduinoCore?

jandrassy, 2021-05-31:
WiFiClient and WiFiClientSecure are a TCP socket. they have nothing to do with HTTP.

d-a-v, 2021-05-31:
`HTTPClient::begin()` client now needs a `WifiClient` or `WiFiClientSecure` as first parameter.
So does arduinoWebsockets.

> So, are there any guide on how to build and test ArduinoCore?

@FAAAAT, maybe @ChrisSmith can share his changes as he proposed in #8085:

> I have working changes locally that I'd be happy to contribute if the maintainers think this is a worthwhile feature.

FAAAAT, 2021-05-31:
> WiFiClient and WiFiClientSecure are a TCP socket. they have nothing to do with HTTP.

Hello jandrassy,
I think when using a http proxy with CONNECT header support, protocols based on TCP can work fine with it.
https://developer.mozilla.org/en-US/docs/Web/HTTP/Methods/CONNECT

> @FAAAAT, maybe @ChrisSmith can share his changes as he proposed in #8085:

I have read this post, but I think may be if Chris does his change in HttpClient, it can't help me with websockets and wss.

I have a rough idea. I think we can define a class support proxy (like http proxy, or socks proxy) inherit from WiFiClient. Then any program that using WiFiClient can use proxy-supported WiFiClient class.

What do you guys think?

Thanks.

jandrassy, 2021-06-01:
> I think when using a http proxy with CONNECT header support, protocols based on TCP can work fine with it.
> https://developer.mozilla.org/en-US/docs/Web/HTTP/Methods/CONNECT

the link describes HTTP connect. HTTP happens over TCP. HTTP is an application layer protocol. the proxy support must of course use TCP socket but not inside socket connect implementation.

d-a-v, 2021-06-01:
Websocket is usually associated with a webserver. Are you using a websocket client ?

WiFiClient can surely be extended to support SOCKS protocol, but HTTP proxy support has to be added in the HTTP client class. From what I understand, the update to be done in the HTTP client class is to make the TCP connection to the proxy instead of to the URL server.

FAAAAT, 2021-06-01:
> Websocket is usually associated with a webserver. Are you using a websocket client ?

Yes, I'm using a websocket client to receive data from a site. And this site is blocked in our country for some reason. So I need a proxy solution.

>WiFiClient can surely be extended to support SOCKS protocol, but HTTP proxy support has to be added in the HTTP client class. From what I understand, the update to be done in the HTTP client class is to make the TCP connection to the proxy instead of to the URL server (edit: it is well described in #8085)

About #8085 ,  there are two methods to access a website through HTTP proxy. 
One is described in #8085, 
[image]
Another is using CONNECT header.
https://developer.mozilla.org/en-US/docs/Web/HTTP/Methods/CONNECT
https://en.wikipedia.org/wiki/HTTP_tunnel#HTTP_CONNECT_method

The differences between these two are:
#8085 : just can be used to proxy HTTP packet.
CONNECT header: can be used by any protocols based on TCP (like https, websockets etc.)

I think implement #8085 on HttpClient is a correct choosen. But for other proxy mechanism I think extending WiFiClient is better.

I understand extending WiFiClient can be done in an additional lib, add it to ArduinoFramework is unnecessary. But for WiFiClientSecure, I think it must be refactored to support proxy mechanism.

Thanks for you guys time.