esp8266/Arduino #8089: weird error in arduino.h on makeWord () with core 3.0.0

stef-ladefense, 2021-05-31:
hi,
i have a weird error in arduino.h on makeWord ().
I use word (h, l) in my code and with core 3.0.0 it gives a fatal error during compilation.

`In file included from C:\Users\stephane\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.0\libraries\ESP8266WiFi\src/WiFiClient.h:25,
                 from C:\Users\stephane\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.0\libraries\ESP8266WiFi\src/ESP8266WiFi.h:39,

C:\Users\stephane\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.0\cores\esp8266/Arduino.h:247:19: error: 'uint16_t makeWord' redeclared as different kind of entity
  247 | uint16_t makeWord(byte h, byte l);
      |                   ^~~~
C:\Users\stephane\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.0\cores\esp8266/Arduino.h:246:10: note: previous declaration 'uint16_t makeWord(uint16_t)'
  246 | uint16_t makeWord(uint16_t w);
      |          ^~~~~~~~
C:\Users\stephane\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.0\cores\esp8266/Arduino.h:247:19: error: reference to 'byte' is ambiguous
  247 | uint16_t makeWord(byte h, byte l);
      |                   ^~~~
`

i got around the problem by changing in arduino.h
uint16_t makeWord (byte h, byte l); -> uint16_t makeWord (uint8_t h, uint8_t l);

What do you think ?

d-a-v, 2021-05-31:
Are you using a git installation, and in that case did you upgrade the toolchain with the script `tools/get.py` ?

stef-ladefense, 2021-05-31:
hi,
no i use classic install with boards manager
(oh escuse me)

d-a-v, 2021-05-31:
We need a way to reproduce, with a MCVE.

side note to maintainers:
the core need this change th match Arduino.h (to avoid `undefined reference to '_Z8makeWordt'`)
(but that won't solve your issue @stef-ladefense)
```diff
--- a/cores/esp8266/WMath.cpp
+++ b/cores/esp8266/WMath.cpp
@@ -77,7 +77,7 @@ long map(long x, long in_min, long in_max, long out_min, long out_max) {
     return (delta * dividend + (divisor / 2)) / divisor + out_min;
 }
 
-unsigned int makeWord(unsigned int w) {
+unsigned int makeWord(uint16_t w) {
     return w;
 }
```

dok-net, 2021-05-31:
I've a few observations. The use of the identifier "byte" is a bit onerous. In `Arduino.h`, there is a `typedef uint8_t byte;`, whereas `core_esp8266_si2c.cpp` and `Esp.cpp` use `byte` as object identifier. On top of that, the signature in `Arduino.h` is not just different for `makeWord(unsigned int)`, but also literally different for `uint16_t makeWord(byte h, byte l);` (Arduino.h) vs. `unsigned int makeWord(unsigned char h, unsigned char l)` (WMath.cpp). 
According to https://en.cppreference.com/w/cpp/types/byte , this also was introduced as an official type in the std namespace with C++17.
As a typename, `byte` is hardly ever used in ESP8266 Core for Arduino (trivial to fix one-time slip being in `core_esp8266_main.cpp`). In `libraries/ESP8266WiFiMesh/src/TypeConversionFunctions.cpp`, it also is used in error, as `libraries/ESP8266WiFiMesh/src/TypeConversionFunctions.h` specifies the affected 2 functions using `uint8_t`.

I'm providing PR #8090 as a suggested fix to this, including @d-a-v's change.

stef-ladefense, 2021-05-31:
first I apologize for not following the issue template, but I did not imagine the benefit.
I'm happy with the outcome, it's going in the right direction
thank you all

dok-net, 2021-05-31:
With the PR, this compiles fine now:
```
void setup()
{
	auto w1 = word(1025);
	auto w2 = word(4, 1);
}

void loop()
{
}
```

dok-net, 2021-06-03:
@d-a-v #8097 builds and works with `byte`, `uint8_t`, and `unsigned char` for the `makeWord(h, l)` arguments. I'm keeping #8090 for the time being, in case someone considers removing `byte` has any value of itself after all.

earlephilhower, 2021-06-13:
Closing as no feedback and a merged PR which we believe fixes things.

Sai-Kiran-31, 2021-06-26:
> thank you all

Sir, How did you resolve the problem I too facing the same

earlephilhower, 2021-06-26:
@Sai-Kiran-31 you need to either use the `git` master branch, or wait for 3.0.1 to be released to get the change incorporated.

bill-orange, 2021-06-28:
I re-tested with 3.0.1 and the noted error still exists.  Thoughts?  2.7.4 works fine.

> Arduino: 1.8.15 (Windows 10), Board: "NodeMCU 1.0 (ESP-12E Module), 80 MHz, Flash, Disabled (new aborts on oom), Disabled, All SSL ciphers (most compatible), 32KB cache + 32KB IRAM (balanced), Use pgm_read macros for IRAM/PROGMEM, 4MB (FS:2MB OTA:~1019KB), 2, v2 Lower Memory, Disabled, None, Only Sketch, 115200"
> 
> In file included from C:\Users\William\Documents\Arduino\libraries\NtpClientLib\src\NtpClientLib.h:59,
> 
>                  from C:\Users\William\Documents\Arduino\libraries\NtpClientLib\src\NTPClientLib.cpp:32:
> 
> C:\Users\William\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.1\cores\esp8266/Arduino.h:247:19: error: 'uint16_t makeWord' redeclared as different kind of entity
> 
>   247 | uint16_t makeWord(byte h, byte l);
> 
>       |                   ^~~~
> 
> C:\Users\William\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.1\cores\esp8266/Arduino.h:246:10: note: previous declaration 'uint16_t makeWord(uint16_t)'
> 
>   246 | uint16_t makeWord(uint16_t w);
> 
>       |          ^~~~~~~~
> 
> C:\Users\William\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.1\cores\esp8266/Arduino.h:247:19: error: reference to 'byte' is ambiguous
> 
>   247 | uint16_t makeWord(byte h, byte l);
> 
>       |                   ^~~~
> 
> In file included from c:\users\william\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\3.0.3-gcc10.3-9bcba0b\xtensa-lx106-elf\include\c++\10.3.0\bits\stl_algobase.h:61,
> 
>                  from c:\users\william\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\3.0.3-gcc10.3-9bcba0b\xtensa-lx106-elf\include\c++\10.3.0\array:40,
> 
>                  from c:\users\william\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\3.0.3-gcc10.3-9bcba0b\xtensa-lx106-elf\include\c++\10.3.0\tuple:39,
> 
>                  from c:\users\william\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\3.0.3-gcc10.3-9bcba0b\xtensa-lx106-elf\include\c++\10.3.0\functional:54,
> 
>                  from C:\Users\William\Documents\Arduino\libraries\NtpClientLib\src\NtpClientLib.h:44,
> 
>                  from C:\Users\William\Documents\Arduino\libraries\NtpClientLib\src\NTPClientLib.cpp:32:
> 
> c:\users\william\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\3.0.3-gcc10.3-9bcba0b\xtensa-lx106-elf\include\c++\10.3.0\bits\cpp_type_traits.h:404:30: note: candidates are: 'enum class std::byte'
> 
>   404 |   enum class byte : unsigned char;
> 
>       |                              ^~~~
> 
> In file included from C:\Users\William\Documents\Arduino\libraries\NtpClientLib\src\NtpClientLib.h:59,
> 
>                  from C:\Users\William\Documents\Arduino\libraries\NtpClientLib\src\NTPClientLib.cpp:32:
> 
> C:\Users\William\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.1\cores\esp8266/Arduino.h:160:17: note:                 'typedef uint8_t byte'
> 
>   160 | typedef uint8_t byte;
> 
>       |                 ^~~~
> 
> In file included from C:\Users\William\Documents\Arduino\libraries\NtpClientLib\src\NtpClientLib.h:59,
> 
>                  from C:\Users\William\Documents\Arduino\libraries\NtpClientLib\src\NTPClientLib.cpp:32:
> 
> C:\Users\William\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.1\cores\esp8266/Arduino.h:247:27: error: reference to 'byte' is ambiguous
> 
>   247 | uint16_t makeWord(byte h, byte l);
> 
>       |                           ^~~~
> 
> In file included from c:\users\william\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\3.0.3-gcc10.3-9bcba0b\xtensa-lx106-elf\include\c++\10.3.0\bits\stl_algobase.h:61,
> 
>                  from c:\users\william\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\3.0.3-gcc10.3-9bcba0b\xtensa-lx106-elf\include\c++\10.3.0\array:40,
> 
>                  from c:\users\william\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\3.0.3-gcc10.3-9bcba0b\xtensa-lx106-elf\include\c++\10.3.0\tuple:39,
> 
[... 27 lines cut ...]
> 
> Multiple libraries were found for "ESP8266WebServer.h"
> 
>  Used: C:\Users\William\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.1\libraries\ESP8266WebServer
> 
>  Not used: C:\Users\William\Documents\Arduino\libraries\ESPWebServer-master
> 
>  Not used: C:\Users\William\Documents\Arduino\libraries\WebServer_tng-master
> 
> Multiple libraries were found for "Ticker.h"
> 
>  Used: C:\Users\William\Documents\Arduino\libraries\Ticker
> 
>  Not used: C:\Users\William\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.1\libraries\Ticker
> 
> Multiple libraries were found for "aREST.h"
> 
>  Used: C:\Users\William\Documents\Arduino\libraries\aREST
> 
>  Not used: C:\Users\William\Documents\Arduino\libraries\arduino_732104
> 
> exit status 1
> 
> Error compiling for board NodeMCU 1.0 (ESP-12E Module).
>

stef-ladefense, 2021-06-28:
tested 3.00 > 3.01 and bug 
`

In file included from C:\Users\stephane\Documents\Arduino\libraries\NtpClient-3.0.2-beta\src\NtpClientLib.h:59,
                 from C:\Users\stephane\Documents\Arduino\libraries\NtpClient-3.0.2-beta\src\NTPClientLib.cpp:32:
C:\Users\stephane\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.1\cores\esp8266/Arduino.h:247:19: error: 'uint16_t makeWord' redeclared as different kind of entity
  247 | uint16_t makeWord(byte h, byte l);
      |                   ^~~~
C:\Users\stephane\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.1\cores\esp8266/Arduino.h:246:10: note: previous declaration 'uint16_t makeWord(uint16_t)'
  246 | uint16_t makeWord(uint16_t w);
      |          ^~~~~~~~
C:\Users\stephane\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.1\cores\esp8266/Arduino.h:247:19: error: reference to 'byte' is ambiguous
  247 | uint16_t makeWord(byte h, byte l);
      |                   ^~~~
In file included from c:\users\stephane\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\3.0.3-gcc10.3-9bcba0b\xtensa-lx106-elf\include\c++\10.3.0\bits\stl_algobase.h:61,
                 from c:\users\stephane\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\3.0.3-gcc10.3-9bcba0b\xtensa-lx106-elf\include\c++\10.3.0\array:40,
                 from c:\users\stephane\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\3.0.3-gcc10.3-9bcba0b\xtensa-lx106-elf\include\c++\10.3.0\tuple:39,
                 from c:\users\stephane\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\3.0.3-gcc10.3-9bcba0b\xtensa-lx106-elf\include\c++\10.3.0\functional:54,
                 from C:\Users\stephane\Documents\Arduino\libraries\NtpClient-3.0.2-beta\src\NtpClientLib.h:44,
                 from C:\Users\stephane\Documents\Arduino\libraries\NtpClient-3.0.2-beta\src\NTPClientLib.cpp:32:
c:\users\stephane\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\3.0.3-gcc10.3-9bcba0b\xtensa-lx106-elf\include\c++\10.3.0\bits\cpp_type_traits.h:404:30: note: candidates are: 'enum class std::byte'
  404 |   enum class byte : unsigned char;
      |                              ^~~~
In file included from C:\Users\stephane\Documents\Arduino\libraries\NtpClient-3.0.2-beta\src\NtpClientLib.h:59,
                 from C:\Users\stephane\Documents\Arduino\libraries\NtpClient-3.0.2-beta\src\NTPClientLib.cpp:32:
C:\Users\stephane\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.1\cores\esp8266/Arduino.h:160:17: note:                 'typedef uint8_t byte'
  160 | typedef uint8_t byte;
      |                 ^~~~
In file included from C:\Users\stephane\Documents\Arduino\libraries\NtpClient-3.0.2-beta\src\NtpClientLib.h:59,
                 from C:\Users\stephane\Documents\Arduino\libraries\NtpClient-3.0.2-beta\src\NTPClientLib.cpp:32:
C:\Users\stephane\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.1\cores\esp8266/Arduino.h:247:27: error: reference to 'byte' is ambiguous
  247 | uint16_t makeWord(byte h, byte l);
      |                           ^~~~
In file included from c:\users\stephane\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\3.0.3-gcc10.3-9bcba0b\xtensa-lx106-elf\include\c++\10.3.0\bits\stl_algobase.h:61,
                 from c:\users\stephane\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\3.0.3-gcc10.3-9bcba0b\xtensa-lx106-elf\include\c++\10.3.0\array:40,
                 from c:\users\stephane\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\3.0.3-gcc10.3-9bcba0b\xtensa-lx106-elf\include\c++\10.3.0\tuple:39,
                 from c:\users\stephane\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\3.0.3-gcc10.3-9bcba0b\xtensa-lx106-elf\include\c++\10.3.0\functional:54,
                 from C:\Users\stephane\Documents\Arduino\libraries\NtpClient-3.0.2-beta\src\NtpClientLib.h:44,
                 from C:\Users\stephane\Documents\Arduino\libraries\NtpClient-3.0.2-beta\src\NTPClientLib.cpp:32:
c:\users\stephane\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\3.0.3-gcc10.3-9bcba0b\xtensa-lx106-elf\include\c++\10.3.0\bits\cpp_type_traits.h:404:30: note: candidates are: 'enum class std::byte'
  404 |   enum class byte : unsigned char;
      |                              ^~~~
In file included from C:\Users\stephane\Documents\Arduino\libraries\NtpClient-3.0.2-beta\src\NtpClientLib.h:59,
                 from C:\Users\stephane\Documents\Arduino\libraries\NtpClient-3.0.2-beta\src\NTPClientLib.cpp:32:
C:\Users\stephane\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.1\cores\esp8266/Arduino.h:160:17: note:                 'typedef uint8_t byte'
  160 | typedef uint8_t byte;
      |                 ^~~~
exit status 1
Erreur de compilation pour la carte Generic ESP8266 Module
`

dok-net, 2021-06-28:
Food for thought, from NtpClientLib.h:
```
#if defined ESP8266 || defined ESP32
#include <functional>
using namespace std;

[...]
```

d-a-v, 2021-07-10:
It appears that there are a number of libraries with `using namespace std;` *and* using `byte`.
Arduino's `byte` and `using namespace std;` are incompatible.
The fix should be to remove all `using namespace std;` and use `std::stuff` instead of `stuff` where relevant but is is not possible to go and fix all external project or libraries.
Removing Arduino's `byte` in favor of `using std::byte;` is not an option because `std::byte test = 0;` is an error (`int` incompatible with `std::byte`…)
I wonder whether there's another path than to patch the toolchain and make `std::byte` definition optional for this core.

mcspr, 2021-07-10:
`std::byte test{0}` is allowed (but might be looking foreign to anyone coming from C), and so is `std::byte test = (std::byte)test` btw. 

With the linked issue, the general problem is with `using namespace std;` in the global namespace, which should probably be discouraged instead of making it silently working with a ton of workarounds?
Also see https://www.cryptopp.com/wiki/Std::byte

d-a-v, 2021-07-10:
I am now closing again this issue that I re-opened.
esp8266/Arduino maintainers won't be doing anything at all to try to keep backward compatibility with already existing codes which are broken when using this core-v3 with the introduction of c++17's `std::byte` when bad c++ practices are involved (such as `using namespace std;`).

closed 2021-07-10 by d-a-v