esp8266/Arduino #8093: LittleFS panic in 30.0.  OK in 2.7.4

fanfanlatulipe26, 2021-06-01:
I  have program running fine in release 2.7.4 but giving a panic dump with 3.0.0
I reproduced the problem with the following small testcase
(I admit that it may seem a bit strange, but it is out of the context of the full program...)
Working with IDE Windows 1.8.13
ESP8266 module  ESP01
Compiled with "Generic ESP8266 Module"

`#include <LittleFS.h>
char traceFileName[] = "File1";
char ligne [] = "abcdefghijklmnopqrstuvwxyz";
File traceFile;
void setup() {
  Serial.begin(115200);
  Serial.println("\n\n\nTest ...");
  Serial.printf("begin LittleFS: %s\n", LittleFS.begin() ? F("OK") : F("FAIL!"));
  traceFile = LittleFS.open(traceFileName, "a");
  Serial.printf("Open append file: %s\n", traceFile ? F("OK") : F("FAIL!"));
  Serial.printf("Format %s\n", LittleFS.format() ? F("OK") : F("FAIL!"));
  LittleFS.end();
  Serial.printf("begin LittleFS: %s\n", LittleFS.begin() ? F("OK") : F("FAIL!"));
  Serial.print("Open append file:" );
  traceFile = LittleFS.open(traceFileName, "a");
  Serial.println(traceFile ? "OK" : "FAIL");
  int  longueurEcriture = traceFile.write((uint8_t *)&ligne, sizeof(ligne));
  Serial.print("write:" );
  Serial.println(longueurEcriture == sizeof(ligne) ? "OK" : "FAIL");
  traceFile.close();
  Serial.println("The end ");
}
void loop() {
  // put your main code here, to run repeatedly:

}
`
`

Decoding stack results
0x401008cc: malloc(size_t) at C:\Users\Francis\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.0\cores\esp8266\umm_malloc\umm_malloc.cpp line 821
0x40204cbe: lfs_file_close at c:\users\francis\appdata\local\arduino15\packages\esp8266\hardware\esp8266\3.0.0\libraries\littlefs\lib\littlefs/lfs.c line 5089
0x40205c98: littlefs_impl::LittleFSFileImpl::close() at C:\Users\Francis\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.0\libraries\LittleFS\src/LittleFS.h line 458
0x40206312: littlefs_impl::LittleFSFileImpl::~LittleFSFileImpl() at c:\users\francis\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\3.0.0-newlib4.0.0-gnu23-48f7b08\xtensa-lx106-elf\include\c++\10.2.0\bits/shared_ptr_base.h line 1183
0x4020a4cc: std::_Sp_counted_ptr_inplace  , (__gnu_cxx::_Lock_policy)0>::_M_dispose() at c:\users\francis\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\3.0.0-newlib4.0.0-gnu23-48f7b08\xtensa-lx106-elf\include\c++\10.2.0\ext/new_allocator.h line 154
0x4020a3a4: std::__shared_count(__gnu_cxx::_Lock_policy)0>::~__shared_count() at c:\users\francis\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\3.0.0-newlib4.0.0-gnu23-48f7b08\xtensa-lx106-elf\include\c++\10.2.0\bits/shared_ptr_base.h line 316
0x40205404: fs::File::operator=(fs::File&&) at c:\users\francis\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\3.0.0-newlib4.0.0-gnu23-48f7b08\xtensa-lx106-elf\include\c++\10.2.0\bits/shared_ptr_base.h line 1183
0x40206fcc: Print::print(char const*) at C:\Users\Francis\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.0\cores\esp8266\Print.cpp line 121
0x40205580: setup() at C:\Users\Francis\Documents\Arduino\mes exemples\WIFI\FIle system LittleFS\Crash ESP8266 3.0.0\small_test/small_test.ino line 16
0x40206d34: HardwareSerial::write(unsigned char const*, unsigned int) at C:\Users\Francis\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.0\cores\esp8266/HardwareSerial.h line 193
0x40205f0c: fs::FS::_defaultTimeCB() at C:\Users\Francis\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.0\cores\esp8266/FS.h line 249
0x402076a8: loop_wrapper() at C:\Users\Francis\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.0\cores\esp8266\core_esp8266_main.cpp line 198

`

earlephilhower, 2021-06-03:
This is actually a LFS panic as reported below.  Can repro, need to verify the test is solid and see what's going awry in the wrapper code...

````
08:15:32.793 -> Test ...
08:15:32.793 -> begin LittleFS: OK
08:15:32.793 -> Open append file: OK
08:15:32.893 -> Format OK
08:15:32.893 -> begin LittleFS: OK
08:15:32.893 -> Open append file:
08:15:32.893 -> User exception (panic/abort/assert)
08:15:32.893 -> --------------- CUT HERE FOR EXCEPTION DECODER ---------------
08:15:32.893 -> 
08:15:32.893 -> Panic lfs.c:5087 lfs_file_close: Assertion 'lfs_mlist_isopen(lfs->mlist, (struct lfs_mlist*)file)' failed.
08:15:32.926 -> 

Decoding stack results
0x401008cc: malloc(size_t) at /home/earle/Arduino/hardware/esp8266com/esp8266/cores/esp8266/umm_malloc/umm_malloc.cpp line 821
0x40204cbe: lfs_file_close at /home/earle/Arduino/hardware/esp8266com/esp8266/libraries/LittleFS/src/../lib/littlefs/lfs.c line 5089
0x40205c98: littlefs_impl::LittleFSFileImpl::close() at /home/earle/Arduino/hardware/esp8266com/esp8266/libraries/LittleFS/src/LittleFS.h line 458
0x40206312: littlefs_impl::LittleFSFileImpl::~LittleFSFileImpl() at /home/earle/Arduino/hardware/esp8266com/esp8266/tools/xtensa-lx106-elf/xtensa-lx106-elf/include/c++/10.2.0/bits/shared_ptr_base.h line 1183
0x4020a4cc: std::_Sp_counted_ptr_inplace  , (__gnu_cxx::_Lock_policy)0>::_M_dispose() at /home/earle/Arduino/hardware/esp8266com/esp8266/tools/xtensa-lx106-elf/xtensa-lx106-elf/include/c++/10.2.0/ext/new_allocator.h line 154
0x4020a3a4: std::__shared_count(__gnu_cxx::_Lock_policy)0>::~__shared_count() at /home/earle/Arduino/hardware/esp8266com/esp8266/tools/xtensa-lx106-elf/xtensa-lx106-elf/include/c++/10.2.0/bits/shared_ptr_base.h line 316
0x40205404: fs::File::operator=(fs::File&&) at /home/earle/Arduino/hardware/esp8266com/esp8266/tools/xtensa-lx106-elf/xtensa-lx106-elf/include/c++/10.2.0/bits/shared_ptr_base.h line 1183
0x40206fcc: Print::print(char const*) at /home/earle/Arduino/hardware/esp8266com/esp8266/cores/esp8266/Print.cpp line 121
0x40205580: setup() at /home/earle/Arduino/sketch_jun03a/sketch_jun03a.ino line 15
0x40206d34: HardwareSerial::write(unsigned char const*, unsigned int) at /home/earle/Arduino/hardware/esp8266com/esp8266/cores/esp8266/HardwareSerial.h line 193
0x40205f0c: fs::FS::_defaultTimeCB() at /home/earle/Arduino/hardware/esp8266com/esp8266/cores/esp8266/FS.h line 249
0x402076a8: loop_wrapper() at /home/earle/Arduino/hardware/esp8266com/esp8266/cores/esp8266/core_esp8266_main.cpp line 198
````

earlephilhower, 2021-06-03:
This is a coding error, not a LittleFS or core issue.

The code  `format()`s the filesystem with an open `File` on it.  That's going to blow up on the ESP8266 and most everywhere else.  You need to `close()` all open files before a) `format()` or b)`end()`ing the FS (or they'll point to stale memory locations from the prior begin).  If it worked before with this, it was sheer luck that all the memory allocations/etc. came out at the exact same spot and didn't end up blowing up the code.

````
  traceFile = LittleFS.open(traceFileName, "a");
  Serial.printf("Open append file: %s\n", traceFile ? F("OK") : F("FAIL!"));
// ****  need to traceFile.close() here
  Serial.printf("Format %s\n", LittleFS.format() ? F("OK") : F("FAIL!"));
  LittleFS.end();
  Serial.printf("begin LittleFS: %s\n", LittleFS.begin() ? F("OK") : F("FAIL!"));
  Serial.print("Open append file:" );
  traceFile = LittleFS.open(traceFileName, "a");
````

Adding the close before the format and all is well
````
08:29:53.323 -> begin LittleFS: OK
08:29:53.323 -> Open append file:OK
08:29:53.345 -> write:OK
08:29:53.345 -> The end 
````

closed 2021-06-03 by earlephilhower

fanfanlatulipe26, 2021-06-03:
Thanks for your help. Sorry ...