esp8266/Arduino #8096: Fatal Python Error when using any ESP8266 board with Arduino IDE.

saintofinternet, 2021-06-02:
Hi,

my arduino ide installation is throwing up a strange problem when I change the board to any ESP8266... following is the error... unable to understand what the issue is.

```
Fatal Python error: initfsencoding: unable to load the file system codec
ModuleNotFoundError: No module named 'encodings'

Current thread 0x00000e14 (most recent call first):
exit status 3
C:\Program Files (x86)\Arduino\arduino-builder returned 3
Error compiling for board NodeMCU 1.0 (ESP-12E Module).

```

earlephilhower, 2021-06-03:
Can you please describe your system in more detail, @saintofinternet ?  We need to be able to reproduce the failure to do anything here, and this is not an issue we've seen on the Windows CI VMs or most other users' Windows installs.

What Windows version?  Which IDE version and how installed (MS Store, download, standalone)?  Running as an admin or unpriv. user?  Is Python installed already?  How installed (Store, Python.org download, etc.)?  Is that Python in your default path?

Also, to capture all your environment variables, can you run a `cmd.exe` prompt and output the results of `set` (with no parameters).  That should dump all environment vars.

saintofinternet, 2021-06-11:
tried all options but Arduino IDE is still failing.

it keeps going to \AppData\\Local\\Arduino15\\packages\\esp8266\\tools\\python3\\3.7.2-post1/python3

and gives the following error

Fatal Python error: initfsencoding: unable to load the file system codec
ModuleNotFoundError: No module named 'encodings'

Current thread 0x00000fec (most recent call first):
exit status 3
C:\Program Files (x86)\Arduino\arduino-builder returned 3
Error compiling for board LOLIN(WEMOS) D1 R2 & mini.

I have Python 3.8 installed on my Windows 7 machine and the path has been added to the environment variables.

Python can be executed from command prompt through any folder or directly without issues.

Only Arduino IDE is unable to use Python for itself. I also can't understand how to change the path of Python which is used by Arduino IDE when compiling.

I think I will have to stop using Arduino IDE permanently. 

I have now shifted to Visual Studio Code and it's working fine on VS Code

earlephilhower, 2021-06-23:
No MCVE/script and nonreproducible given the limited info present.  If someone has the same problem and can give the add'l requested info, we can look at it.  Until then, closing.

closed 2021-06-23 by earlephilhower

hase-berlin, 2022-10-29:
Arduino 1.8.19 on Windows10.
Downloaded from Arduino.cc Website. running as unpriv, as far as I can tell. I am more of a Linux guy...
RP2040 Core installed.
Pico selected as board.

Compiling throws
...
Sketch wird kompiliert...
"C:\\Users\\hase\\AppData\\Local\\Arduino15\\packages\\rp2040\\tools\\pqt-python3\\1.0.1-base-3a57aed/python3" -I "C:\\Users\\hase\\AppData\\Local\\Arduino15\\packages\\rp2040\\hardware\\rp2040\\2.6.2/tools/signing.py" --mode header --publickey "C:\\Users\\hase\\Arduino_WS\\Etherweb/public.key" --out "C:\\Users\\hase\\AppData\\Local\\Temp\\arduino_build_625167/core/Updater_Signing.h"
Fatal Python error: initfsencoding: unable to load the file system codec
ModuleNotFoundError: No module named 'encodings'

So I went to C:\\Users\\hase\\AppData\\Local\\Arduino15\\packages\\rp2040\\tools\\pqt-python3\\1.0.1-base-3a57aed/ in a Powershell and started the python3.exe in that directory manualls.
Same eror :-)
PS C:\Users\hase\AppData\Local\Arduino15\packages\rp2040\tools\pqt-python3\1.0.1-base-3a57aed> .\python3.exe
Fatal Python error: initfsencoding: unable to load the file system codec
ModuleNotFoundError: No module named 'encodings'

Current thread 0x00003108 (most recent call first):
The messages show the python from the Arduino install? 
I have an install of Python 3.9.7 from the python.org website on the same machine, but Arduino seems to call absolute not via the Path.

Arduino-IDE 2.0.1 calls the same binary and throws the same error.

For me this is new, compiling for the Pico on this machine did work before.
The only changes I can see is a recent update of the Arduino-Pico core.

PR #8226 (2021-07-26): Use only installed Python on Windows
When Python is installed on Windows separately (i.e. Windows Store,
Python.org MSI, etc.) it set the PYTHONHOME environment variable to
the installed path.

When we call our own portable Python, it tries to use the support
PYC files in PYTHONHOME.  If they're from a different version of
Python, however, they won't work.  Even though we're running our
own distributed Python.exe, we are crashing on the system-wide
Python installation.

Add -I to all python3 calls, ignore PYTHONHOME/etc. environment vars.

Tested under Ubuntu 18.04 Linux w/o any ill effects (w [... cut]
commit a0d2a7a8 (2021-07-26): Use only installed Python on Windows (#8226)

When Python is installed on Windows separately (i.e. Windows Store,
Python.org MSI, etc.) it set the PYTHONHOME environment variable to
the installed path.

When we call our own portable Python, it tries to use the support
PYC files in PYTHONHOME.  If they're from a different version of
Python, however, they won't work.  Even though we're running our
own distributed Python.exe, we are crashing on the system-wide
Python installation.

Add -I to all python3 calls, ignore PYTHONHOME/etc. environment vars.

Tested under Ubuntu 18.04 Linux w/o any ill e [... cut]