esp8266/Arduino #8105: ESP8266 V3.0.0 broke all my projects, yea, really all... Where to start?

sblantipodi, 2021-06-06:
As title,
core 3.0.0 completely destroyed all my projects.

Where can I start to debug? 
I think that I'm not the only one with this problem, Is there any suggestion?

d-a-v, 2021-06-06:
- If you think this is a bug in the core, take a simple example, make it a MCVE, open a new issue and fill its template, let us reproduce.

- Ask your question in a dedicated forum ([esp8266.com](https://www.esp8266.com/), stackoverflow arduino), or [gitter](https://gitter.im/esp8266/arduino).

closed 2021-06-06 by d-a-v

sblantipodi, 2021-06-06:
@d-a-v If you want to hear a feedback this is the worst way possible to upgrade a platform.
features must be deprecated before breaking them.

you simply destroyed mosts of the ESP8266 ecosystem in one udpate, what is the purpose of a similar update?

d-a-v, 2021-06-06:
> what is the purpose of an update like this?

Fixing issue, improving stability, improving performances, bringing new features.
Sometimes [breaking changes](https://github.com/esp8266/Arduino/releases/tag/3.0.0) are needed.