esp8266/Arduino #8112: Migrate to ArduinoCore-API

dok-net, 2021-06-10:
Migrating / porting Arduino Core for ESP8266 to use https://github.com/arduino/ArduinoCore-API should bring closure to debates about Arduino compatibility. 

According to the README there, in excerpts:
```
All Arduino official cores are being ported to the new structure so they take advantage of this single repo.

[...]
Some cores have been ported to the new structure, for example:

megaAVR (https://github.com/arduino/ArduinoCore-megaAVR)
nRF52-mbedos (https://github.com/arduino/ArduinoCore-nRF528x-mbedos)
classic AVR (https://github.com/arduino/ArduinoCore-avr/tree/api)
SAMD (https://github.com/arduino/ArduinoCore-samd/tree/api)
```

earlephilhower, 2021-06-12:
It might make sense to move to this, but to me it's not at all clear how to get there.

`String`, for example, has been significantly improved (IMHO) here and the API folder includes its own generic implementation which would overwrite it.  (Not to mention the fact that there is no `WString.h` anymore, only `String.h`....which conflicts on Windows w/the standard library `<string.h>`!)  There are also issues w/the way it's actually handled (Windows `git` doesn't work w/ symlinks so if you want Win users to be able use `master` you need to actually just check in copies of the files in your repo).

As I'm using it on the [Raspberry Pi Pico RP2040 core](https://github.com/earlephilhower/arduino-pico) I'm also finding incompatibilities (at least w/the latest API revisions) with the API and some of the IDE-shipped examples(!).  So it's not 100% clean going that way.

Manually picking a choosing header files to pull in from there might be the best way forward, but it would definitely be work intensive and break lots of little things...

dok-net, 2021-06-13:
It's a bit weird reading what is going on at Arduino-core-api's issue tracking, a lot of discussions involving hardware specific matters. Makes it look more and more like it's best to let matters settle before adopting that. Arduino-compatibility is becoming a moving target, isn't it?

earlephilhower, 2021-06-14:
> Arduino-compatibility is becoming a moving target, isn't it?

I think it's always been a moving target.  That's just the way things happen when you have so many boards to support and tons of users like the Arduino.cc folks.

In the interim, seems like a couple options are available:
* Pick and choose individual APIs to move to latest API compat (i.e. instead of `uint8_t` everything is now `PinType_t` for `digitalWrite` et. al.
* It may make sense to at least include the Arduino IDE shipped examples as part of CI.  That would increase runtime for each PR but I'm not sure it would help finding anything given the age of the 8266 core.

In any case, moving to the ArduinoCore-API directly isn't indicated at this point in time for all the above reasons..