esp8266/Arduino #8113: ESP8266 Board version 3.0 is throwing error with Firebase Arduino Master

TrickSumo, 2021-06-10:
Steps to reprodure error:-

1. Update ESP8266 to version 3.0
2. Install Arduino Firebase master library
3. Write simple program to push data to firebase real time database.
4. Compile errror

Error: https://user-images.githubusercontent.com/85671823/121488688-4f29f700-c9a1-11eb-8409-a81669a503ac.png

d-a-v, 2021-06-10:
A copy paste of the error instead of a screen capture can help us answer.
Best would be to try one of their bundled examples and create an issue in their issue tracker with a reference here, in this issue.
This is anyway not a real issue with this core. v3.0.0 is a major release with [breaking changes](https://github.com/esp8266/Arduino/releases/tag/3.0.0). The http client interface used by Firebase was deprecated for a long time.
I'm keeping this opened because we lack a proper way to determine the major version compatible with `#if` / `#ifdef`.

TrickSumo, 2021-06-10:
Hi

Complete Error is:-

> C:\Users\TrickSumo\Documents\Arduino\libraries\firebase-arduino-master\src\FirebaseHttpClient_Esp8266.cpp: In member function 'virtual void FirebaseHttpClientEsp8266::begin(const string&)':
C:\Users\TrickSumo\Documents\Arduino\libraries\firebase-arduino-master\src\FirebaseHttpClient_Esp8266.cpp:47:50: error: no matching function for call to 'begin(const char*, const char [60])'
   47 |     http_.begin(url.c_str(), kFirebaseFingerprint);
      |                                                  ^
In file included from C:\Users\TrickSumo\Documents\Arduino\libraries\firebase-arduino-master\src\FirebaseHttpClient_Esp8266.cpp:9:
C:\Users\TrickSumo\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.0\libraries\ESP8266HTTPClient\src/ESP8266HTTPClient.h:166:10: note: candidate: 'bool HTTPClient::begin(String, uint16_t, String)' (near match)
  166 |     bool begin(String host, uint16_t port, String uri = "/")  __attribute__ ((error("obsolete API, use ::begin(WiFiClient, host, port, uri)")));
      |          ^~~~~
C:\Users\TrickSumo\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.0\libraries\ESP8266HTTPClient\src/ESP8266HTTPClient.h:166:10: note:   conversion of argument 2 would be ill-formed:
C:\Users\TrickSumo\Documents\Arduino\libraries\firebase-arduino-master\src\FirebaseHttpClient_Esp8266.cpp:47:30: error: invalid conversion from 'const char*' to 'uint16_t' {aka 'short unsigned int'} [-fpermissive]
   47 |     http_.begin(url.c_str(), kFirebaseFingerprint);
      |                              ^~~~~~~~~~~~~~~~~~~~
      |                              |
      |                              const char*
In file included from C:\Users\TrickSumo\Documents\Arduino\libraries\firebase-arduino-master\src\FirebaseHttpClient_Esp8266.cpp:9:
C:\Users\TrickSumo\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.0\libraries\ESP8266HTTPClient\src/ESP8266HTTPClient.h:167:10: note: candidate: 'bool HTTPClient::begin(String, const uint8_t*)' (near match)
  167 |     bool begin(String url, const uint8_t httpsFingerprint[20])  __attribute__ ((error("obsolete API, use ::begin(WiFiClientSecure, ...)")));
      |          ^~~~~
C:\Users\TrickSumo\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.0\libraries\ESP8266HTTPClient\src/ESP8266HTTPClient.h:167:10: note:   conversion of argument 2 would be ill-formed:
C:\Users\TrickSumo\Documents\Arduino\libraries\firebase-arduino-master\src\FirebaseHttpClient_Esp8266.cpp:47:30: error: invalid conversion from 'const char*' to 'const uint8_t*' {aka 'const unsigned char*'} [-fpermissive]
   47 |     http_.begin(url.c_str(), kFirebaseFingerprint);
      |                              ^~~~~~~~~~~~~~~~~~~~
      |                              |
      |                              const char*
C:\Users\TrickSumo\Documents\Arduino\libraries\firebase-arduino-master\src\FirebaseHttpClient_Esp8266.cpp: In member function 'virtual void FirebaseHttpClientEsp8266::begin(const string&, const string&)':
C:\Users\TrickSumo\Documents\Arduino\libraries\firebase-arduino-master\src\FirebaseHttpClient_Esp8266.cpp:51:60: error: invalid conversion from 'const char*' to 'const uint8_t*' {aka 'const unsigned char*'} [-fpermissive]
   51 |     http_.begin(host.c_str(), kFirebasePort, path.c_str(), kFirebaseFingerprint);
      |                                                            ^~~~~~~~~~~~~~~~~~~~
      |                                                            |
      |                                                            const char*
In file included from C:\Users\TrickSumo\Documents\Arduino\libraries\firebase-arduino-master\src\FirebaseHttpClient_Esp8266.cpp:9:
C:\Users\TrickSumo\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.0\libraries\ESP8266HTTPClient\src/ESP8266HTTPClient.h:168:70: note:   initializing argument 4 of 'bool HTTPClient::begin(String, uint16_t, String, const uint8_t*)'
  168 |     bool begin(String host, uint16_t port, String uri, const uint8_t httpsFingerprint[20])  __attribute__ ((error("obsolete API, use ::begin(WiFiClientSecure, ...)")));
      |                                                        ~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~
exit status 1
Error compiling for board NodeMCU 1.0 (ESP-12E Module).

Code is:-
```

#include <ESP8266WiFi.h>
#include <FirebaseArduino.h>
                          
#define FIREBASE_HOST ""                     //Your Firebase Project URL goes here without "http:" , "\" and "/"
#define FIREBASE_AUTH "" //Your Firebase Database Secret goes here

#define WIFI_SSID "Wifiname-here"                                               //WiFi SSID to which you want NodeMCU to connect
#define WIFI_PASSWORD "Wifipassword-here"                                      //Password of your wifi network 

int val=0, val3=1000;

void setup() {

  Serial.begin(115200);

  Serial.println("Serial communication started\n\n");  
           
  WiFi.begin(WIFI_SSID, WIFI_PASSWORD);                                     //try to connect with wifi
  Serial.print("Connecting to ");
  Serial.print(WIFI_SSID);

  
  while (WiFi.status() != WL_CONNECTED) {
    Serial.print(".");
    delay(500);
  }
[... 20 lines cut ...]
  }
  
 else { 
    Serial.println("Everything is ready!");
    delay(300); Serial.println("Everything is ready!");
    delay(300); Serial.println("Everything is ready! \n \n \n");
    delay(300);

    Firebase.setInt("/data",val);
   Serial.println(val);
    delay(300); Serial.println("uploaded val to firebase \n \n \n");

      Firebase.setInt("/test/val3",val3);
   Serial.println(val3);
    delay(300); Serial.println("uploaded val3 to firebase \n \n \n");
  

  val++; val3++;
 }

  
}
```

Thanks

earlephilhower, 2021-06-10:
This is a Firebase bug/incompatibility.

Firebase is using a call that was deprecated ~1 year ago and removed in 3.0 completely, as mentioned in the error message:
>  168 | bool begin(String host, uint16_t port, String uri, const uint8_t httpsFingerprint[20]) attribute ((error("obsolete API, use ::begin(WiFiClientSecure, ...)")));

An update to Firebase to use the supported version of the call which gets a `WiFiClientSecure` passed in is the proper resolution.

closed 2021-06-10 by earlephilhower