esp8266/Arduino #8119: Interrupts httpcode returns -1

Bartras, 2021-06-12:
Hello,

I try to update a domoticz device in function of my waterflow sensor (YF-B1).
I use a Wemos D1 mini.

Here below the simple code to read the flow value in serial monitor:

`int NbTopsFan;
int Calc;
int hallsensor = D6;

void ICACHE_RAM_ATTR rpm()
{
  NbTopsFan++;
}

void setup() {
pinMode (hallsensor, INPUT_PULLUP);
Serial.begin(9600);
attachInterrupt(digitalPinToInterrupt(hallsensor), rpm, FALLING);
 
}

void loop() {
  NbTopsFan = 0;
  sei();
  delay(1000);
  cli();
  Calc = (NbTopsFan * 2.2);
  Serial.print (Calc, DEC);
  Serial.println("L/h");
}`

Here below the test code to update my domoticz device (update contineously with 300L/h):

`#include <ESP8266WiFi.h>
#include <ESP8266HTTPClient.h>

const char* ssid     = "xxx";
const char* password = "xxx";
const char* host = "192.168.1.100";
const int   port = 8080;
const char*   id = "x";
const char*   mdp = "x";
const int   idx = 15804;
const int   watchdog = 5000; // Fréquence d'envoi (ms) des données à Domoticz - Frequency of sending data to Domoticz
unsigned long previousMillis = millis();

HTTPClient http;

void setup() {
  
  Serial.begin(115200);
  delay(10);

  Serial.setDebugOutput(true);  
  Serial.println("Connecting Wifi...");

  WiFi.begin(ssid, password);
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
   
  Serial.println("");
  Serial.println("WiFi connected");
  Serial.println("IP address: ");
  Serial.print(WiFi.localIP()); 

}

void loop() {
  unsigned long currentMillis = millis();

  if ( currentMillis - previousMillis > watchdog ) {
    previousMillis = currentMillis;

    if(WiFi.status() != WL_CONNECTED) {
      Serial.println("WiFi not connected !");
    } else {  
      Serial.println("Send data to Domoticz");
      
      String url = "/json.htm?username=";
      url += String(id);
      url += String("&password=");
      url += String(mdp);
      url += String("&type=command&param=udevice&idx=");
      url += String(idx);
      url += String("&nvalue=0&svalue=");
      url += String(300);
       
      sendDomoticz(url);
      }
   }
 }

void sendDomoticz(String url){
  Serial.print("connecting to ");
  Serial.println(host);
  Serial.print("Requesting URL: ");
  Serial.print(host);
  Serial.print(":");
  Serial.print(port);
  Serial.println(url);
  http.begin(host,port,url);
  int httpCode = http.GET();
  Serial.println(httpCode);
    if (httpCode) {
      if (httpCode == 200) {
        String payload = http.getString();
        Serial.println("Domoticz response "); 
        Serial.println(payload);
      }
    }
  Serial.println("closing connection");
  http.end();
}`

Notice that the two previous codes work perfectly.

So I merge the both and here below the final code:

`#include <ESP8266WiFi.h>
#include <ESP8266HTTPClient.h>

const char* ssid     = "xxx";
const char* password = "xxx";
const char* host = "192.168.1.100";
const int   port = 8080;
const char*   id = "x";
const char*   mdp = "x";
const int   idx = 15804;
const int   watchdog = 5000; // Fréquence d'envoi (ms) des données à Domoticz - Frequency of sending data to Domoticz
unsigned long previousMillis = millis();

int NbTopsFan;
int Calc;
int hallsensor = D6;

int value = 0;

void ICACHE_RAM_ATTR rpm()
{
  NbTopsFan++;
}

HTTPClient http;

void setup() {

  pinMode (hallsensor, INPUT_PULLUP);
  
  Serial.begin(115200);
  delay(10);

  attachInterrupt(digitalPinToInterrupt(hallsensor), rpm, FALLING);

  Serial.setDebugOutput(true);  
  Serial.println("Connecting Wifi...");

  WiFi.begin(ssid, password);
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
   
  Serial.println("");
  Serial.println("WiFi connected");
  Serial.println("IP address: ");
  Serial.print(WiFi.localIP()); 
}

void loop() {
  
      NbTopsFan = 0;
      sei();
      delay(5000);
      cli();
      Calc = (NbTopsFan * 2.2);
      Serial.print (Calc, DEC);
      Serial.println("L/h");
      
  unsigned long currentMillis = millis();

  if ( currentMillis - previousMillis > watchdog ) {
    previousMillis = currentMillis;

    if(WiFi.status() != WL_CONNECTED) {
      Serial.println("WiFi not connected !");
    } else {  
      Serial.println("Send data to Domoticz");
   
      String url = "/json.htm?username=";
      url += String(id);
      url += String("&password=");
      url += String(mdp);
      url += String("&type=command&param=udevice&idx=");
      url += String(idx);
      url += String("&nvalue=0&svalue=");
      url += String(Calc);
         
      sendDomoticz(url);
      }
   }
 }

void sendDomoticz(String url){
  Serial.print("connecting to ");
  Serial.println(host);
  Serial.print("Requesting URL: ");
  Serial.print(host);
  Serial.print(":");
  Serial.print(port);
  Serial.println(url);
  http.begin(host,port,url);
  int httpCode = http.GET();
  Serial.println(httpCode);
    if (httpCode) {
      if (httpCode == 200) {
        String payload = http.getString();
        Serial.println("Domoticz response "); 
        Serial.println(payload);
      }
    }
  Serial.println("closing connection");
  http.end();
}`

And with that code, the httpcode returns -1.

I don't know why it doesn't work...

Have you any solution ?
Thank a lot.

enjoyneering, 2021-07-29:
try to add  noInterrupts(); after serial & interrupts(); at the end in sendDomoticz(String url).

```c++
void sendDomoticz(String url){
  Serial.print("connecting to ");
  Serial.println(host);
  Serial.print("Requesting URL: ");
  Serial.print(host);
  Serial.print(":");
  Serial.print(port);
  Serial.println(url);

  noInterrupts();
  http.begin(host,port,url);
  int httpCode = http.GET();
  Serial.println(httpCode);
    if (httpCode) {
      if (httpCode == 200) {
        String payload = http.getString();
        Serial.println("Domoticz response "); 
        Serial.println(payload);
      }
    }
  Serial.println("closing connection");
  http.end();

 interrupts(); 
}
```

Bartras, 2021-07-29:
Thank you for helping.

I tried, and same result.
The http code returns -1