esp8266/Arduino #8121: Power off at the ending of ESPhttpUpdate.update, exception occurs for later power on

zhfei1979, 2021-06-13:
### Basic Infos

- [ ] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [ ] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [ ] I have tested that the issue is present in current master branch (aka latest git).
- [ ] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [ ] I have filled out all fields below.

#### Platform

- Hardware:					[ESP8266]
- Core Version:			[SDK:2.2.2-dev(38a443e)/Core:3.0.0=30000000/lwIP:STABLE-2_1_2_RELEASE/glue:1.2-48-g7421258/BearSSL:c0b69df]
- Development Env:		[Arduino IDE 1.8.13]
- Operating System:		[Windows]

### Settings in IDE

- Module:  			[Generic ESP8266 Module]
- Flash Mode:			[DOUT]
- Flash Size: 		[2MB]
- lwip Variant:   [v2 Lower Memory]
- Reset Method:		[dtr]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz]
- Upload Using:		[OTA]
- Upload Speed:   [115200|other] (serial upload only)

### Problem Description

Power off at the ending of ESPhttpUpdate.update, exception occurs for later power on.
Note that works fine if no power off during ESPhttpUpdate.update.

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

    ESPhttpUpdate.setLedPin(LED_PIN);
    ESPhttpUpdate.rebootOnUpdate(false);
    WiFiClient client;
    S_PL(urlOTA);
    t_httpUpdate_return OTA = ESPhttpUpdate.update(client, urlOTA);
    if(OTA != HTTP_UPDATE_OK) {
      S_P("UpdateByOTA: ");S_P(ESPhttpUpdate.getLastErrorString().c_str());S_PL("!");
    }
    ESP.restart();

### Debug Messages

```
[httpUpdate] Header read fin.
[httpUpdate] Server header:
[httpUpdate]  - code: 200
[httpUpdate]  - len: 463312
[httpUpdate] ESP8266 info:
[httpUpdate]  - free Space: 581632
[httpUpdate]  - current Sketch Size: 463312
[httpUpdate] runUpdate flash...
[httpUpdate] peekBytes x：4
sleep disable
pm open,type:0 0
[httpUpdate] **Update ok**
state: 5 -> 0 (0)
rm 0
pm close 7
del if0
usl
⸮⸮*⸮P*V⸮⸮ET⸮⸮⸮⸮T⸮⸮+⸮⸮u⸮U+UZ⸮Z⸮T⸮⸮⸮TQT⸮UQ⸮⸮jP⸮⸮⸮u⸮⸮⸮⸮⸮ZT⸮(UE⸮⸮⸮⸮QR⸮Eu⸮⸮⸮⸮*⸮⸮⸮jPE*⸮u⸮⸮E+⸮⸮⸮T)UZV⸮R⸮E⸮j*QR⸮E⸮_⸮*⸮ER⸮⸮_
 ets Jan  8 2013,rst cause:1, boot mode:(3,7)

load 0x4010f000, len 3460, room 16 
tail 4
chksum 0xcc
load 0x3fff20b8, len 40, room 4 
tail 4
chksum 0xc9
csum 0xc9
v000711d0
~ld
Fatal exception 0(IllegalInstructionCause):
epc1=0x4024dd50, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x00000000

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Exception (0):
epc1=0x4024dd50 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

>>>stack>>>

ctx: sys
sp: 3fffe8c0 end: 3fffffb0 offset: 0190
3fffea50:  00000000 00000000 00000000 00000000  
3fffea60:  4024e721 00000000 00000000 00000000  
3fffea70:  4025c8c7 40105bb8 00000000 00000000  
3fffea80:  00000000 400042db 40105e6d 000001fd  
3fffea90:  00000082 00000020 00000000 00000001  
3fffeaa0:  40105e67 40105cd7 00000003 00000000  
3fffeab0:  ffffe7f4 ffffffff ffff0001 00000007  
3fffeac0:  482d4c47 00322d36 3ffffeec 40211c90  
3fffead0:  40211c84 3ffef260 3ffffeec 40212019  
3fffeae0:  00030503 47000001 3236486c 00313034  
3fffeaf0:  6d3ffffe da402122 003ffe8d 0d000000  
3fffeb00:  fd6f000a da402156 f03ffe8d ef4bc6a7  
3fffeb10:  380bc6a7 00353634 00000000 f0000000  
3fffeb20:  7c4bc6a7 003fffff e28ad284 d296942e  
3fffeb30:  1dba09f8 1c1c0229 911b3f99 a815e2ca  
3fffeb40:  c74311de 1d7214fb ffffffff ffffffff  
3fffeb50:  ffffffff ffffffff 1f0000ff a1f6347a  
3fffeb60:  00000005 38383838 00000038 40250442  
```

=========== DECODED ===============
Exception 0: Illegal instruction
PC: 0x4024dd50
EXCVADDR: 0x00000000

Decoding stack results
0x40211c90: UpdaterClass::printError(Print&) at C:\Users\Administrator\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.0\cores\esp8266\Updater.cpp line 561
0x40211c84: UpdaterClass::printError(Print&) at C:\Users\Administrator\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.0\cores\esp8266\Updater.cpp line 559
0x40212019: String::changeBuffer(unsigned int) at C:\Users\Administrator\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.0\cores\esp8266\WString.cpp line 207
0x40100d08: umm_free_core(umm_heap_context_t*, void*) at C:\Users\Administrator\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.0\cores\esp8266\umm_malloc\umm_malloc.cpp line 565
0x40100cd3: umm_free_core(umm_heap_context_t*, void*) at C:\Users\Administrator\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.0\cores\esp8266\umm_malloc\umm_malloc.cpp line 553
0x40100f6e: umm_init_common(size_t, void*, size_t, bool) at C:\Users\Administrator\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.0\cores\esp8266\umm_malloc\umm_malloc.cpp line 438
0x40100f60: umm_init_common(size_t, void*, size_t, bool) at C:\Users\Administrator\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.0\cores\esp8266\umm_malloc\umm_malloc.cpp line 428
0x40211c90: UpdaterClass::printError(Print&) at C:\Users\Administrator\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.0\cores\esp8266\Updater.cpp line 561
0x40211c84: UpdaterClass::printError(Print&) at C:\Users\Administrator\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.0\cores\esp8266\Updater.cpp line 559
0x40212019: String::changeBuffer(unsigned int) at C:\Users\Administrator\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.0\cores\esp8266\WString.cpp line 207
0x4021226d: String::String(unsigned int, unsigned char) at C:\Users\Administrator\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.0\cores\esp8266\WString.cpp line 78
0x402156fd: DhcpServer::parse_options(unsigned char*, short) at C:\Users\Administrator\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.0\cores\esp8266\LwipDhcpServer.cpp line 790
[ESPhttpUpdate.update升级快完成时掉电，系统异常.pdf](https://github.com/esp8266/Arduino/files/6643683/ESPhttpUpdate.update.pdf)

zhfei1979, 2021-06-13:
'at the ending of ESPhttpUpdate.update' means the update isn't done yet and will be done in around 5 seconds.

zhfei1979, 2021-06-16:
@earlephilhower I'm sorry to @you, wish you can give some advice as I'm confused & upset & totally lost for this issue, thx.

I hardly get any useful help from docs why it happens, even the source codes such as Updater.cpp, ESP.cpp.
I get nothing of what happens for the last 3-5 seconds' random prints during ESPhttpUpdate.update.
I suppose the flash has current & new/download sketch sections as said in the end of https://arduino-esp8266.readthedocs.io/en/3.0.1/ota_updates/readme.html, which makes the sketches should never get broken.

attached serial prints of ESPhttpUpdate.update twice, the 1st works very well without power off; and power off in the 2nd; the chip crashes when power on again.
[httpUpdate.log](https://github.com/esp8266/Arduino/files/6663737/httpUpdate.log)

zhfei1979, 2021-06-17:
Finally I guess I find the root cause, because power off happens at ACTION_COPY_RAW in Updater.cpp.
In this case, the copy failed & the target image broken; my question is why dont' have master/slave sketches support?

  if (_command == U_FLASH) {
    eboot_command ebcmd;
    ebcmd.action = ACTION_COPY_RAW;
    ebcmd.args[0] = _startAddress;
    ebcmd.args[1] = 0x00000;
    ebcmd.args[2] = _size;
    eboot_command_write(&ebcmd);

d-a-v, 2021-06-17:
> my question is why dont' have master/slave sketches support?

That's #6538

zhfei1979, 2021-06-17:
> 
> 
> > my question is why dont' have master/slave sketches support?
> 
> That's #6538

Thanks for the hints.
The merge of davisonja:ota_commands_in_flash to master will be later milestones like 3.0.1 or 3.1 if I understand correctly?
That'll be great if so!

earlephilhower, 2021-06-20:
Closing as semi-dup of of #6538.  This is a known thing with the current core and that referenced PR is intended to change it.

closed 2021-06-20 by earlephilhower