esp8266/Arduino #8124: Decreased free heap in v3.0.0

maakbaas, 2021-06-15:
### Basic Infos

- [x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [ ] I have tested that the issue is present in current master branch (aka latest git). <== I don't know how with PlatformIO?
- [x] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [x] I have filled out all fields below.

#### Platform

- Hardware:				[ESP-12]
- Core Version:			3.0.0
- Development Env:		[Platformio]
- Operating System:		[Windows]

### Settings in IDE

- Module:  			[Nodemcu]
- Flash Size: 		[4MB]
- Reset Method:		[nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz]
- Upload Using:		[SERIAL]
- Upload Speed:   [115200] (serial upload only)

### Problem Description

I suddenly found my free heap reduced significantly.

I did some testing for this and I got the following numbers. Test A was a minimal sketch without the ESPAsyncWebServer, and test B was using the full code below.

Test A - Arduino 2.7.4: 51752 kB free heap
Test A - Arduino 3.0.0: 50784 kB free heap

Test B - Arduino 2.7.4:  43496 kB free heap
Test B - Arduino 3.0.0:  38480 kB free heap

This means that on the sketch with ESP Async WebServer I have lost 5Kb of heap without any changes to the code, except for the update to 3.0.0. The Async WebServer PlatformIO package has not been updated in over a year, so the codebase for that is the same.  This is also why I opened the issue here, and not there.

Which changes in 3.0.0 could have caused this increased memory usage, and would there be ways to mitigate this?

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp

#include <Arduino.h>
#include <ESPAsyncWebServer.h> //this line was removed for test A

void setup() 
{
    Serial.begin(9600);
    AsyncWebServer server = AsyncWebServer(80); //this line was removed for test A
    AsyncWebSocket ws = AsyncWebSocket("/ws"); //this line was removed for test A
}

void loop()
{
    Serial.println(ESP.getFreeHeap());
    delay(5000);
}

```

mobizt, 2021-06-15:
Read this https://arduino-esp8266.readthedocs.io/en/latest/mmu.html

sh-user, 2021-06-16:
In my project, I noticed a heap loss of about 2KB. Оption `16KB cache + 48KB IRAM and 2nd heap (shared)` is even worse.

maakbaas, 2021-06-16:
For me the option does also not make a difference in the reported free heap by the `ESP.getFreeHeap()` command. In any case, allthough it might be a mitigation, I am more curious where this heap actually went in 3.0.0.

mobizt, 2021-06-16:
Example sketch to check free IRAM and DRAM memory
```cpp

#include <Arduino.h>
#include <umm_malloc/umm_heap_select.h>

void setup()
{
    Serial.begin(115200);

    {
        HeapSelectIram ephemeral;
        Serial.printf("IRAM free: %6d bytes\r\n", ESP.getFreeHeap());
    }
    
    {
        HeapSelectDram ephemeral;
        Serial.printf("DRAM free: %6d bytes\r\n", ESP.getFreeHeap());
    }
}

void loop()
{
    // put your main code here, to run repeatedly:
}
```

sh-user, 2021-06-16:
Really `#include <umm_malloc/umm_heap_select.h>` and `HeapSelectIram ephemeral;` returned a bunch as in 2.7.4
I wonder if it is possible to make an `MMU: 48KB cache + 16KB IRAM`

earlephilhower, 2021-06-20:
@maakbaas using the `ELF` intermediate file would let us see if this is a compile-time thing or a runtime thing, and point out what the difference is.  Can you please build the same test w/2.7.4 and 3.0.0 and attach each one's `sketch.ino.elf` (not BIN)?

TD-er, 2021-06-24:
I've been looking into this option of using a 2nd heap as I always run out of resources like RAM.
I noticed there have been reports of a free heap of order of 50'000 bytes.
How can that be achieved? I never got it above 30-ish kB.
Using the 2nd heap results in a number of issues by the way, will open new issues for them.

Jason2866, 2021-06-26:
Theo adopted Tasmota to use 2nd heap. Tasmota log buffer and settings are in iram heap when using core 3.0.0. Still some code clean up in Tasmota needed.
[image]
 Before we had about 28k free memory

TD-er, 2021-06-26:
you sum it I assume?
Some examples online show the free heap as a single number > 32k

dalbert2, 2021-07-09:
Every 15 minutes, my application turns on wifi (station mode), scans for wifi APs, connects if a configured AP is found, exchanges some data with a server, and then turns off wifi.  Since upgrading to 3.0.x, I frequently see 2K disappear each pass through that loop (which wasn't a problem in 2.7.x)   Moreover, once it starts happening, a single call to WiFi.scanNetworks() will leak the same amount of memory (~2K).  I'm starting to suspect changes in the LWIP stack.

Note: I am using IRAM48; I have the same results with and without SECHEAP_SHARED

d-a-v, 2021-07-09:
> I'm starting to suspect changes in the LWIP stack.

To my knowledge, nearly nothing has changed with defaults on this side between 2.7.4 and 3.0.0.
scanNetwork is not working on the lwIP level.
[From what you wrote](https://gitter.im/esp8266/Arduino?at=60e7b282f2b4077d6bb7fea7), maybe you could try to instrument [the function itself](https://github.com/esp8266/Arduino/blob/2946ce055c142319dddba7b20d8839146817d6ee/libraries/ESP8266WiFi/src/ESP8266WiFiScan.cpp#L71-L107) to narrow down to where memory is allocated ?

dalbert2, 2021-07-10:
I rolled back to PlatformIO 2.6.3 (Core 2.7.4, SDK 2.2.2, LWIP:2.1.2) and using the same code, the behavior is different:

The problem seems to manifest when the STA is having trouble connecting to the AP.  Every 15 minutes, my code scans for the configured APs and if one is found, attempts to connect, exchange data, and then turns off WiFi.  

The problem I am having is when the connection does not succeed.  My code then retries in 30 seconds.  With 3.0.x, the heap drops ~2K each time this happens; with 2.7.4, there is no memory leak (although the connection is still repeatedly failing).

I'm going to try to turn on more WiFi stack debugging to better understand why the connection is failing which may point to the memory leak.

maakbaas, 2021-07-12:
> @maakbaas using the `ELF` intermediate file would let us see if this is a compile-time thing or a runtime thing, and point out what the difference is. Can you please build the same test w/2.7.4 and 3.0.0 and attach each one's `sketch.ino.elf` (not BIN)?

@earlephilhower: The ELF file for 2.7.4:
http://gofile.me/5GnMu/w123yRD1O

The ELF file for 3.0.0:
http://gofile.me/5GnMu/5iSt6FGEY

Links are valid for 2 weeks. Thanks for looking into this.

earlephilhower, 2021-07-12:
Only ~350 bytes difference in .data:
````
earle@server:/tmp$ objdump -t ~/Desktop/firmware_3.0.0.elf  | grep \\.data | awk '{print $5 " " $6 }' | sort -r
0000016c __global_locale
000000f0 impure_data
00000020 rx_gain_swp
00000004 _ZN10__cxxabiv119__terminate_handlerE
00000004 _ZL8_pwmFreq
00000004 tcb
00000004 soft_wdt_interval
00000004 iss$0
00000004 _impure_ptr
00000004 chip_version
00000004 bin_sum_len
00000004 AWSC_PING_PAYLOAD
00000003 test_rffreq_txcap
00000002 udp_port
00000002 tx_rf_ana_gain
00000002 sleep_start_wait_time
00000002 numempty
00000002 NoiseTimerInterval
00000001 wifi_station_default_hostname
00000001 TmpSTAAPCloseAP
00000001 timer2_ms_flag
00000001 TestStaFreqCalValOK
00000001 test_freq_val_first
00000001 rfcal_bb_atten_init
00000001 reconnect_internal
00000001 dhcps_flag
00000001 dhcpc_flag
00000001 chan14_mic_most_power
00000001 CanDoFreqCal
00000000 .data
earle@server:/tmp$ objdump -t ~/Desktop/firmware_2.7.4.elf  | grep \\.data | awk '{print $5 " " $6 }' | sort -r
000000f0 impure_data
00000038 lconv
00000020 rx_gain_swp
00000020 lc_message_charset
00000020 lc_ctype_charset
00000004 _ZN10__cxxabiv119__terminate_handlerE
00000004 __wctomb
00000004 tcb
00000004 soft_wdt_interval
00000004 __mb_cur_max
00000004 iss$5349
00000004 _impure_ptr
00000004 dhcps_lease_time
00000004 __ctype_ptr__
00000004 chip_version
00000004 bin_sum_len
00000004 AWSC_PING_PAYLOAD
00000003 test_rffreq_txcap
00000002 udp_port
00000002 tx_rf_ana_gain
00000002 tcp_port
00000002 sleep_start_wait_time
00000002 NoiseTimerInterval
00000001 wifi_station_default_hostname
00000001 TmpSTAAPCloseAP
00000001 timer2_ms_flag
00000001 TestStaFreqCalValOK
00000001 test_freq_val_first
00000001 rfcal_bb_atten_init
00000001 reconnect_internal
00000001 offer
00000001 dhcps_flag
00000001 dhcpc_flag
00000001 chan14_mic_most_power
00000001 CanDoFreqCal
00000000 .data
````

Basically 0 difference in .bss
````
earle@server:/tmp$ objdump -t ~/Desktop/firmware_2.7.4.elf  | grep \\.bss | awk '{print $5 " " $6 }' | sort -r | head -20
000006a8 g_ic
000001b0 dns_table
00000100 pmc
00000100 event_TaskQueue
000000f0 arp_table
000000b4 Update
0000009c gScanStruct
00000098 netif_git
00000094 chip6_sleep_params
00000074 chip6_phy_init_ctrl
00000040 wDevCtrl
0000002c dhcp_rx_options_val
00000028 _ZZL15do_global_ctorsvE2ob
00000028 buffed_eb_arr
00000024 sntp_servers
00000024 info
00000020 lmacConfMib
00000020 if_ctrl
0000001c Serial1
0000001c Serial
earle@server:/tmp$ objdump -t ~/Desktop/firmware_3.0.0.elf  | grep \\.bss | awk '{print $5 " " $6 }' | sort -r | head -20
000006a8 g_ic
00000200 _ZL8wvfState
000001b0 dns_table
00000100 pmc
00000100 event_TaskQueue
000000f0 arp_table
000000b4 Update
0000009c gScanStruct
00000098 netif_git
00000094 chip6_sleep_params
00000074 chip6_phy_init_ctrl
00000040 wDevCtrl
00000030 dhcpSoftAP
0000002c dhcp_rx_options_val
00000028 _ZZL15do_global_ctorsvE2ob
00000028 buffed_eb_arr
00000024 sntp_servers
00000024 info
00000020 Serial1
00000020 Serial
````

And about 0 difference in .rodata:
````
earle@server:/tmp$ objdump -t ~/Desktop/firmware_2.7.4.elf  | grep \\.rodata | awk '{print $5 " " $6 }' | sort -r | head -20
0000002c memp_pools
00000013 SDK_VERSION
00000011 CSWTCH$17
00000011 CSWTCH$14
00000011 CSWTCH$11
00000010 tcp_pcb_lists
00000010 CSWTCH$9
00000010 CSWTCH$19
0000000d tcp_backoff
00000007 tcp_persist_backoff
00000006 ethzero
00000006 ethbroadcast_LWIP2
00000006 ethbroadcast
00000005 dhcp_discover_request_options
00000004 ip_addr_broadcast
00000004 ip_addr_any
00000004 _global_impure_ptr
00000004 dns_mquery_v4group
00000002 memp_UDP_PCB
00000002 memp_TCP_SEG
earle@server:/tmp$ objdump -t ~/Desktop/firmware_3.0.0.elf  | grep \\.rodata | awk '{print $5 " " $6 }' | sort -r | head -20
0000002c memp_pools
00000013 SDK_VERSION
00000011 CSWTCH$31
00000010 tcp_pcb_lists
00000010 CSWTCH$34
00000010 CSWTCH$10
0000000d tcp_backoff
0000000c _C_numeric_locale
00000007 tcp_persist_backoff
00000006 ethzero
00000006 ethbroadcast_LWIP2
00000006 ethbroadcast
00000005 dhcp_discover_request_options
00000004 _ZN10DhcpServer12magic_cookieE
00000004 ip_addr_broadcast
00000004 ip_addr_any
00000004 _global_impure_ptr
00000004 dns_mquery_v4group
00000002 memp_UDP_PCB
00000002 memp_TCP_SEG
````

So, looks like you have some runtime issue.  The ELF won't show that, of course.  Statically, all looks fine here between 2.7.4 and 3.0.0.