esp8266/Arduino #8130: Updater generating String error messages

earlephilhower, 2021-06-16:
### Basic Infos

- [x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x] I have tested that the issue is present in current master branch (aka latest git).
- [x] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [x] I have filled out all fields below.

#### Platform

- Hardware:					[all]
- Core Version:			[3.0.0 + git master]
- Development Env:		[Arduino IDE|
- Operating System:		[all]

### Settings in IDE

- Module:  			[Wemos D1 mini r2]
- Flash Mode:			[dio]
- Flash Size: 		[4MB]
- lwip Variant:   [v2 Lower Memory]
- Reset Method:		[ck|nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz|160MHz]
- Upload Using:		[SERIAL]
- Upload Speed:   [115200|other] 

### Problem Description

While debugging #8079 I am getting error messages from `String`.  Need to figure out if this is a `String` problem, an `Updater` problem, or just a poorly worded debug message:

12:33:23.841 -> [HTTP-Client][begin] url: https://apps.95wd.com.cn/httpUpdate.ino.generic.bin
12:33:23.841 -> [HTTP-Client][begin] host: apps.95wd.com.cn port: 443 url: /httpUpdate.ino.generic.bin
12:33:23.841 -> [offending String op 134->165 ('x-ESP8266- ...  1662976
12:33:23.841 -> ')]
12:33:23.841 -> [offending String op 165->221 ('x-ESP8266- ... : 431296
12:33:23.841 -> ')]
12:33:23.841 -> [offending String op 221->251 ('x-ESP8266- ... 84e66870
12:33:23.874 -> ')]
12:33:23.874 -> [offending String op 251->294 ('x-ESP8266- ...  4194304
12:33:23.874 -> ')]
12:33:23.874 -> [offending String op 294->318 ('x-ESP8266- ... 38a443e)
12:33:23.874 -> ')]
12:33:23.874 -> [HTTP-Client][sendRequest] type: 'GET' redirCount: 0
12:33:23.874 -> [hostByName] request IP for: apps.95wd.com.cn
12:33:23.874 -> [hostByName] Host: apps.95wd.com.cn IP: 47.100.169.50
12:33:23.874 -> :ref 1

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

See #8079

d-a-v, 2021-06-16:
It's on us both #7908

earlephilhower, 2021-06-16:
Yes, I will do a PR with a reword as well as having the Updater pre-allocate the Strings to avoid the warning in the 1st place.  "offending String op xxx" doesn't mean much unless you go to the actual source line (where it is commented well!).

earlephilhower, 2021-06-17:
There's not a good way to pre-allocate the String in the httpUpdate class w/o changing the HTTPClient API which I don' t think is a good thing for a 3.0.1 bugfix, so #8131 simply uses a more descriptive warning.

PR #8131 (2021-06-17): Make String reallocation warning more descriptive
Fixes #8130
commit f1310c04 (2021-06-17): Make String reallocation warning more descriptive (#8131)

Fixes #8130
closed 2021-06-17 by earlephilhower