esp8266/Arduino #8135: Exception 28: LoadProhibited: A load referenced a page mapped with an attribute that does not permit loads

crbxy, 2021-06-17:
### Basic Infos

- [X] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [X] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [X] I have tested that the issue is present in current master branch (aka latest git).
- [X] I have searched the issue tracker for a similar issue.
- [X] If there is a stack dump, I have decoded it.
- [X] I have filled out all fields below.

#### Platform

- Hardware:					NodeMCU ESP-12E
- Core Version:			[latest git hash or date]
- Development Env:		Arduino IDE 1.8.15
- Operating System:		Windows

### Settings in IDE

- Module:  			NodeMCU
- Flash Mode:			[qio|dio|other]
- Flash Size: 		4MB
- lwip Variant:   v2 Lower Memory
- Reset Method:		nodemcu
- Flash Frequency:	40Mhz
- CPU Frequency:		80Mz
- Upload Using:		SERIAL
- Upload Speed:   115200

### Problem Description

Detailed problem description goes here.

Exception 28: LoadProhibited: A load referenced a page mapped with an attribute that does not permit loads
PC: 0x4000bf80
EXCVADDR: 0x0000008c

Decoding stack results
0x402019e0: coapClient::send(IPAddress, int, char*, COAP_TYPE, COAP_METHOD, unsigned char*, unsigned char, unsigned char*, unsigned int, unsigned char, unsigned char) at C:\Users\ugur_\OneDrive\Belgeler\Arduino\libraries\ESP-CoAP-master\coap_client.cpp line 81

0x40100100: ets_intr_lock() at C:\Users\ugur_\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.0\cores\esp8266\core_esp8266_main.cpp line 159

0x402035e6: uart_tx_fifo_available(int) at C:\Users\ugur_\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.0\cores\esp8266\uart.cpp line 499

0x40203670: uart_do_write_char(int, char) at C:\Users\ugur_\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.0\cores\esp8266\uart.cpp line 512

0x40201ea8: HardwareSerial::write(unsigned char) at C:\Users\ugur_\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.0\cores\esp8266/HardwareSerial.h line 189

0x40201a70: coapClient::loop() at C:\Users\ugur_\OneDrive\Belgeler\Arduino\libraries\ESP-CoAP-master\coap_client.cpp line 191

0x4020825b: rand at /workdir/repo/newlib/newlib/libc/stdlib/rand.c line 79

0x40201a3c: coapClient::get(IPAddress, int, char*) at C:\Users\ugur_\OneDrive\Belgeler\Arduino\libraries\ESP-CoAP-master\coap_client.cpp line 29

0x401008cc: malloc(size_t) at C:\Users\ugur_\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.0\cores\esp8266\umm_malloc\umm_malloc.cpp line 821

0x40100274: pvPortMalloc(size_t, char const*, int) at C:\Users\ugur_\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.0\cores\esp8266\umm_malloc/umm_heap_select.h line 85

0x40201a70: coapClient::loop() at C:\Users\ugur_\OneDrive\Belgeler\Arduino\libraries\ESP-CoAP-master\coap_client.cpp line 191

0x402021f8: Print::println() at C:\Users\ugur_\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.0\cores\esp8266/Print.h line 57

0x402017c7: coapClient::start() at C:\Users\ugur_\OneDrive\Belgeler\Arduino\libraries\ESP-CoAP-master\coap_client.cpp line 24

0x4020118a: setup() at C:\Users\ugur_\AppData\Local\Temp\arduino_modified_sketch_968430/coapclient.ino line 66

0x40202934: loop_wrapper() at C:\Users\ugur_\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.0\cores\esp8266\core_esp8266_main.cpp line 198

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp

/*
ESP-COAP Client
*/
#include <ESP8266WiFi.h>
#include "coap_client.h"

//instance for coapclient
coapClient coap;

//WiFi connection info
const char* ssid = "tt12";
const char* password = "12345";

//ip address and default port of coap server in which your interested in
IPAddress ip(127,0,0,1);//take ETH Zurich or coap.me server to run and check client 
int port =5683;

// coap client response callback
void callback_response(coapPacket &packet, IPAddress ip, int port);

// coap client response callback
void callback_response(coapPacket &packet, IPAddress ip, int port) {
    char p[packet.payloadlen + 1];
    memcpy(p, packet.payload, packet.payloadlen);
    p[packet.payloadlen] = NULL;

    //response from coap server
 if(packet.type==3 && packet.code==0){
      Serial.println("ping ok");
    }

    Serial.println(p);
}

void setup() {
   
    Serial.begin(115200);

    WiFi.begin(ssid, password);
    Serial.println(" ");

    // Connection info to WiFi network
    Serial.println();
    Serial.println();
    Serial.print("Connecting to ");
    Serial.println(ssid);
    WiFi.begin(ssid, password);
    while (WiFi.status() != WL_CONNECTED) {
    //delay(500);
    yield();
    Serial.print(".");
    }
    Serial.println("");
    Serial.println("WiFi connected");
    // Print the IP address of client
    Serial.println(WiFi.localIP());

    // client response callback.
    // this endpoint is single callback.
    coap.response(callback_response);

    // start coap client
    coap.start();

    //get request to server (arguments ip adrress of server,default port,resource(uri))
    int msgid = coap.get(ip,port,"light");

    //observe request (arguments ip adrress of server,deafult port,resource name,interger(0) ) 
    //int msgid= coap.observe(ip,port,"light",0);

    //reset observe cancel
    //int msgid=coap.observecancel(ip,port,"resoucename");
    
}
//int i=0;

void loop() {
    bool state;
 
    // Requests

    //get request
    //int msgid = coap.get(ip,port,"hello");

    //put request 
    //arguments server ip address,default port,resource name, payload,payloadlength
    //int msgid =coap.put(ip,port,"resourcename","0",strlen("0"));

    //post request
    //arguments server ip address,default port,resource name, payload,payloadlength
    //int msgid =coap.post(ip,port,"resourcename","0",strlen("0"));

    //delete request
    //int msgid = coap.delet(ip,port,"resourcename");

    //ping
    //int msgid=coap.ping(ip,port);
    
    // int msgid=coap.observe(ip,port,"obs",0);
  
    state= coap.loop();
    // Serial.print("state=");
    //Serial.println(state);
    //if(state==true)
          //i=i+1;
  
    //Serial.print("i=");
    //Serial.println(i);
      //if(i==3)
    //{
        //Serial.println("cancel observe");
        //coap.observeCancel(ip,port,"resourcename");
    //}
  
    //Serial.println(msgid);
    delay(1000);
}

```

### Debug Messages

```
--------------- CUT HERE FOR EXCEPTION DECODER ---------------

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 3460, room 16 
tail 4
chksum 0xcc
load 0x3fff20b8, len 40, room 4 
tail 4
chksum 0xc9
csum 0xc9
v00043a80
~ld

```

### This is the library i used [thingTronics ESP8266 12E Arduino Library](https://github.com/automote/ESP-CoAP)

### i've decoded the stack and posted everything, this is for my school project but i am a newb to arduino and these codes and have no idea what is causing this, please if anybody could help it would be appreciated.

Tech-TX, 2021-06-19:
The last stack point before the exception is a call inside that 3rd party library.  You should post this on _their_ issue tracker,  as this is not obviously  an issue with the Arduino ESP8266 core files or libraries.

holgerlembke, 2021-06-19:
Just fishing...

The sketch tries to connect to a coap-server at "IPAddress ip(127,0,0,1);".  127.0.0.1 is always (ok, very very most likely) the machine the sketch is running on itself, so your nodemcu-hardware.

You should have some IP there where a server is running. Like coap.me/134.102.218.18. ethz.ch does not exist any more...

crbxy, 2021-06-19:
Thank you both for replying

> Just fishing...
>
> The sketch tries to connect to a coap-server at "IPAddress ip(127,0,0,1);". 127.0.0.1 is always (ok, very very most likely) the machine the sketch is running on itself, so your [...]

i've started a coap server using californium eclipse and tested it with google chrome copper extension, coap://127.0.0.1 is open there as well as coap://192.168.X.X but as i said i am still new when it comes to these things, i've tried changing ip in the code to 134.102.218.18 as you suggested, still same error unfortunately

> The last stack point before the exception is a call inside that 3rd party library. You should post this on _their_ issue tracker, as this is not obviously an issue with the Arduino ESP8266 core files or libraries.

sorry about that this was my first post on github, but its issue tracker is really dead so i think it's very unlikely i will get a reply there, thought i could get help on here

holgerlembke, 2021-06-19:
@crbxy 127.0.0.1 can be a little confusing. /me fallen in that trap, too.

127.0.0.1 is the local computer. Lets name it Box A. So if your server is on Box A and your software runs also on Box A you can connect to your Box A's server via 127.0.0.1. Because 127.0.0.1 is always local. Thus its symbolic name **localhost** or more general called as a **loopback**. If you move your software to Box B (!) and leave your server on Box A, then you will connect to Box B if you use 127.0.0.1.

You need to find out the public IP of your Box A. Use _ipconfig_ in a command shell. Mostly it is something starting with 192.168..

It might be that the firewall of your Box A denies connections to that port from outside. You can try it with putty to make a raw connection to Box A's public ip and the coap protocool port (5683?). Take a second computer Box X and try to connect to Box A. It should at least result in an connection.

Frustrating stuff. Keep searching for errors at the basic levels (ping from A to B, ip connections from A to B and B to A) before blaming the higher levels... :-) Write down the IPs of your Boxes to keep an overview.

crbxy, 2021-06-19:
@holgerlembke i got the point, at least i think i do, thank you for the explanation, now i got the local ip (or router ip i don't know what its called), and tried to use it to check if the server still sends a response, i took a picture i guess the connection is done, i even tried to get putty but apparently it doesn't support UDP, used Packet Sender instead, now it makes a clear difference but on the nodemcu side, i am still getting the same error

earlephilhower, 2021-06-23:
Closing as it's showing a crash in a 3rd party library and the discussion has turned into a coding one.  Please use https://esp8266.com or https://gitter.im/esp8266/Arduino for discussions like this.

closed 2021-06-23 by earlephilhower