esp8266/Arduino #8139: ESP8266NetBIOS library only works in wifi STA mode, and not in wifi AP

pabloandresm, 2021-06-17:
Title says it all.

If you look at the code, file ESP8266NetBIOS.cpp, line 215, in the netbios answer, you can see:
	nbnsa.NBNSA_NODEADDRESS = WiFi.localIP(); // ulozime nasi IP adresu

Which works only if WiFi is in STA mode, but not AP mode.

The fix would be something like this:
	if (WiFi.getMode()==WIFI_STA)
		nbnsa.NBNSA_NODEADDRESS = WiFi.localIP(); // ulozime nasi IP adresu
	else
		nbnsa.NBNSA_NODEADDRESS = WiFi.softAPIP(); // ulozime nasi IP adresu

earlephilhower, 2021-06-17:
@pabloandresm you've found a problem and potential solution.  Why not try your hand at a [pull request](https://docs.github.com/en/github/collaborating-with-pull-requests/proposing-changes-to-your-work-with-pull-requests/about-pull-requests) so your work would be preserved in the `git` logs?

pabloandresm, 2021-06-17:
I'll do that right now. Thanks.

PR #8622 (2022-06-29): Use current netif address in NetBIOS response
fix for NetBIOS only working if WIFI in STA mode and not AP mode

resolve #8139
resolve #8140
closed 2022-06-29 by mcspr