esp8266/Arduino #8145: Please document the options on the Arduino Tools menu

technoblogy, 2021-06-19:
Many of the options on the Arduino IDE **Tools** menu for the ESP8266 boards aren't obvious, but I can't find them documented anywhere. If they are documented, please mention where to find this information in the main Arduino core ReadMe.

For example:

On the **Flash Size** submenu what does **FS** mean in **FS:2MB**?
What is **IwIP Variant**? What does "(no features)" mean?
What is **Vtables**?
What does **Stack Protection** do?

Searching in the **readthedocs** finds nothing for any of these. 

Note that some of the options _are_ documented, which is good. For example, the **MMU** and **Non-32-bit access** options are explained on the MMU page in **readthedocs**. However, I would prefer to have them all documented or referenced from one place, systematically explaining each option on the **Tools** menu.

d-a-v, 2021-06-21:
This will answer to your fair request: #8154

technoblogy, 2021-06-21:
This is great - exactly what I was looking for! Thanks, David

d-a-v, 2021-06-21:
You're welcome, David

PR #8154 (2021-06-22): documentation: Arduino IDE specific options for esp82xx
closes #8145

(english syntax or grammar, or description reviews are welcome, I wish someone else had done this doc :)
closed 2021-06-22 by earlephilhower