esp8266/Arduino #8148: Isn't the "stability problems may occur" message misleading

technoblogy, 2021-06-20:
When you compile a program in the Arduino IDE you get a message such as:

````text
Sketch uses 342821 bytes (32%) of program storage space. Maximum is 1044464 bytes.
Global variables use 61960 bytes (75%) of dynamic memory, leaving 19960 bytes for local variables. Maximum is 81920 bytes.
Low memory available, stability problems may occur.
````
Isn't this message misleading on the ESP8266?

On other platforms, such as AVR, the remaining 25% of dynamic memory is used for the stack, so reducing it will definitely reduce stability; highly recursive functions will cause a crash due to stack overflow.

I've been testing this (on an Adafruit ESP8266 Feather) by finding how deep I can call a recursive function before a reset occurs.

It turns out that it makes no difference if I allocate virtually all the dynamic memory to arrays in my program, so the message becomes:

````text
Sketch uses 342821 bytes (32%) of program storage space. Maximum is 1044464 bytes.
Global variables use 77960 bytes (95%) of dynamic memory, leaving 3960 bytes for local variables. Maximum is 81920 bytes.
Low memory available, stability problems may occur.
````
Reset occurs at the same depth of recursion.

There's no mention of the stack here:

https://github.com/esp8266/esp8266-wiki/wiki/Memory-Map

but from the forums I understand it's fixed at 4Kbytes, so not affected by the amount of dynamic memory available?

earlephilhower, 2021-06-20:
This is a general question and not following the (deleted) template, closing.

The stack is fixed size (4K) and preallocated.  This warning is correct and if you expect to get much done w/<4KB free heap, good luck.  There most definitely will be issues in almost all sketches due to memory fragmentation and OOM.

closed 2021-06-20 by earlephilhower

technoblogy, 2021-06-20:
Thanks for the explanation, and sorry if my post was inappropriate.