esp8266/Arduino #8159: How to make RF calibration behave like in core 2.4.2?

TD-er, 2021-06-22:
### Basic Infos

- [x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x] I have tested that the issue is present in current master branch (aka latest git).
- [x] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [ ] I have filled out all fields below.

#### Platform

- Hardware:					other
- Core Version:			[latest git hash or date]
- Development Env:		[Platformio]
- Operating System:		[Windows]

### Settings in IDE

- Module:  			[Generic ESP8266 Module]
- Flash Mode:			[dio]
- Flash Size: 		[4MB]
- lwip Variant:   [v1.4|v2 Lower Memory|Higher Bandwidth]
- Reset Method:		[nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz]
- Upload Using:		[SERIAL]
- Upload Speed:   [115200] (serial upload only)

### Problem Description

I do have a module here which does work well with a build of ESPEasy of 2019/02/26, but not on later builds.  (core 2.4.2)
Symptoms are:
- Able to connect to AP
- Most of the time it is able to receive an IP
- Hardly any IP traffic is possible (pings do not even get answered)
- Access point kicks off node after set timeout in AP (On a Mikrotik you can extend this, so I am sure it is the AP that kicks off the node)

After lots and lots of debugging I found out that the specific board seems to have an antenna which is not matched very well to the WiFi frequency.
If I hold the board between thumb and index finger on either the begin or end of the PCB antenna trace, the node suddenly starts answering to pings.
This means the resonance frequency of the antenna is too high.
N.B. the antenna also feels warm.

A user sent me this board, which is part of a larger batch which all show the same symptoms.

Normally I would toss this away as it is clearly something wrong in the antenna tuning.
However, on core 2.4.2 the board runs fine.
So I wonder if it is possible to fix this in software?

Apparently the RF calibration used in core 2.4.2 was able to adjust parameters a bit further to compensate for this compared to what later core revisions would allow.

While debugging, I noticed quite a few other issues which also may sound very familiar to anyone who has had issues with WiFi since core 2.5.0.
Using very sophisticated tools like my fingers, ping and live RSSI charts in the MikroTik web interface I was able to 'fine tune' the antenna of the board and noticed the following:

- RF calibration can correct the tuning if the antenna is somewhat in the acceptable range when starting. -> RSSI values are significantly higher on the AP, compared to when reconnecting with fingers on the antenna after boot.
- By slightly moving my fingers or changing pressure I could change responsiveness of the web UI and ping times.
- Reducing TX power does lower temperature of antenna, but also "tuning" using my fingers. (no crashes or other strange issues by the way caused by RF reflected signals back into the ESP)

These findings were very well reproducible, so I do feel confident it was not just a fluke.

This looks like a lot of ESP devices out there may have antennas which are not optimal and maybe just on the edge of what the RF calibration in core 2.5.0 and later can correct.
N.B. enclosures or presence of other materials close to the antenna can cause an antenna to shift its resonance frequency.

Is there a way to change the RF calibration to allow it to accept a larger range to correct an antenna?
It was working in core 2.4.2, so the ESP is capable of correcting for it.

jandrassy, 2021-06-22:
2.5.0 was the version with SDK pre-3 and many users reported problems with WiFi

TD-er, 2021-06-22:
@jandrassy Yep, the file was changed, but not its value.
What I meant with the suggested setting is, would it be possible to tell the RF calibration to correct for these symptoms by changing this byte to allow for using a frequency offset?

closed 2021-06-23 by earlephilhower