esp8266/Arduino #8164: Esp8266 wemos d1 interrupt not working and crashing. Same code works fine on esp32 and arduino nano. why does esp8266 crash ?

paxsipornax, 2021-06-23:
volatile int revs = 0;
unsigned long rpm = 0;
unsigned long simdiki_zaman = 0;
unsigned long onceki_zaman = 0;
#define rpmPin D5

unsigned long yenilemehizi = 250;
void setup() {

  Serial.begin(115200);

  pinMode(rpmPin, INPUT);

  attachInterrupt(digitalPinToInterrupt(rpmPin), rpm_engine, FALLING);
}
void loop() {
  simdiki_zaman = millis();
  
  if (simdiki_zaman - onceki_zaman >= yenilemehizi) {
    onceki_zaman = simdiki_zaman;
    detachInterrupt(digitalPinToInterrupt(rpmPin));

    rpm = revs * 120;

    onceki_zaman = simdiki_zaman;
    revs = 0;
    Serial.print("RPM: ");
    Serial.println(rpm);
    Serial.flush();
    
    attachInterrupt(digitalPinToInterrupt(rpmPin), rpm_engine, FALLING);
    
  }
}
void rpm_engine() {
  revs++;
}

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Abort called

>>>stack>>>

ctx: cont
sp: 3fffff00 end: 3fffffc0 offset: 0000
3fffff00:  feefeffe feefeffe feefeffe 3ffef2fc  
3fffff10:  000000fe 00000000 00000000 00000000  
3fffff20:  00000000 00000000 00000000 00ff0000  
3fffff30:  5ffffe00 5ffffe00 0001c200 3ffee570  
3fffff40:  00000000 00000002 0000000e 40201e6a  
3fffff50:  4010049d 00000001 3ffee508 40201e7c  
3fffff60:  00000000 00000001 0000000e 4020237d  
3fffff70:  00000000 feefeffe feefeffe feefeffe  
3fffff80:  3fffdad0 00000000 3ffee55c 4020241c  
3fffff90:  3fffdad0 00000000 3ffee55c 40201069  
3fffffa0:  feefeffe feefeffe feefeffe 402019f8  
3fffffb0:  feefeffe feefeffe 3ffe85dc 40100d59  
<<<stack<<<

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

 ets Jan  8 2013,rst cause:2, boot mode:(3,3)

load 0x4010f000, len 3460, room 16 
tail 4
chksum 0xcc
load 0x3fff20b8, len 40, room 4 
tail 4
chksum 0xc9
csum 0xc9
v000420c0
~ld
ISR not in IRAM!

User exception (panic/abort/assert)

sh-user, 2021-06-23:
try adding `void ICACHE_RAM_ATTR rpm_engine();`

paxsipornax, 2021-06-23:
> try adding `void ICACHE_RAM_ATTR rpm_engine();`

thank you so much it worked cheers 👍

closed 2021-06-23 by paxsipornax