esp8266/Arduino #8169: strnlen_P causes crash when used on object allocated on 2nd heap

TD-er, 2021-06-24:
### Basic Infos

- [x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x] I have tested that the issue is present in current master branch (aka latest git).
- [x] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [ ] I have filled out all fields below.

#### Platform

- Hardware:					[ESP-12|ESP-01|ESP-07|ESP8285 device|other]
- Core Version:			[latest git hash or date]
- Development Env:		[Platformio]
- Operating System:		[Windows]

### Settings in IDE

- Module:  			[Generic ESP8266 Module]
- Flash Mode:			[dio]
- Flash Size: 		[4MB]

PIO flags:
```
                            -DPIO_FRAMEWORK_ARDUINO_MMU_CUSTOM                            
                            -DMMU_IRAM_SIZE=0xC000 
                            -DMMU_ICACHE_SIZE=0x4000 
                            -DMMU_IRAM_HEAP
```

### Problem Description

I have been working on this new (very nice) feature of using the 2nd heap to store some run time data.
For example I keep a queue of my events (`String` objects) and fetch them.

This did cause a crash on my project which eventually pointed to the use of `strnlen_P` (mind the 'n') which is not used in my code, but it is used in the `String` class.
The crash was about the loadprohibited errors you typically get when trying to use strlen like calls on flash strings:

```
LoadProhibited: A load referenced a page mapped with an attribute that does not permit loads
  epc1=0x40283620 in strnlen_P at /workdir/repo/newlib/newlib/libc/sys/xtensa/string_pgmspace.c:51
```

But now the really strange issue here and that's why I can't (yet) include a MCVE...
It does occur when interacting with data from the 2nd heap, but to fix it I have to alter calls to functions which may expect a `const String&` which and being called with a flash string.
For example when serving a web page, I enter code like this:

```c++
void Web_StreamingBuffer::sendHeaderBlocking(bool json, const String& origin) {
  #ifndef BUILD_NO_RAM_TRACKER
  checkRAM(String(F("sendHeaderBlocking")));
  #endif
```

This already includes the 'fix' to wrap the flash string in a `String` constructor. But this function `checkRAM` does have a function definition which accepts a flash string.

So it 'feels' like this web serving call is happening right when data was being stored in my event queue which does swap the heap to the iram heap.

I will try to see if I can also fix it by explicitly switching to the DRAM heap when serving a web page, but it does feel a bit tricky to say the least.

Also I am having a hard time in getting this 2nd heap thing to work reliable.
It feels like random if the 2nd heap is active or not when compiling it using PlatformIO.
Using `PIO_FRAMEWORK_ARDUINO_MMU_CACHE16_IRAM48_SECHEAP_SHARED` as PIO define does randomly have it compiled in.
When it doesn't work, calling `ESP.getFreeHeap()` returns the same value always regardless the selected heap.

d-a-v, 2021-06-24:
> Using PIO_FRAMEWORK_ARDUINO_MMU_CACHE16_IRAM48_SECHEAP_SHARED as PIO define does randomly have it compiled in.
> When it doesn't work, calling ESP.getFreeHeap() returns the same value always regardless the selected heap.

You need to clean every object file if you change this configuration between two builds.

TD-er, 2021-06-24:
> > Using PIO_FRAMEWORK_ARDUINO_MMU_CACHE16_IRAM48_SECHEAP_SHARED as PIO define does randomly have it compiled in.
> > When it doesn't work, calling ESP.getFreeHeap() returns the same value always regardless the selected heap.
> 
> You need to clean every object file (and library/archive files) if you change this configuration between two builds.

Been there done that, waiting for the T-shirt.
I even had several Explorer windows open on the folders that needed to be wiped.
I am even thinking of creating a batch script for it.

earlephilhower, 2021-06-24:
@TD-er can you post the full crash dump?  The EXECVADDR is important as it says what the actual faulting read  address was.  If it's in 0x00000 + epsilon, you have a nullptr problem.  If it's inside the IRAM area, then it's something else.

The referenced line of code in the crash is a 32b read which should work just fine in IRAM, but will crap out if you have, say, an odd address or are in an illegal region (i.e. 0x00000...).

TD-er, 2021-06-24:
I just have another one...
```
 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 3460, room 16
tail 4
chksum 0xcc
load 0x3fff20b8, len 40, room 4
tail 4
chksum 0xc9
csum 0xc9
v000e1b70
~ld
�U343 : Info   : 

INIT : Booting version: 'feature/build_core_3_0_0_c934304' (ESP82xx Core efb0341a, NONOS SDK 3.0.0-dev(c0f7b44), LWIP: 2.1.2 PUYA support)
344 : Info   : INIT : Free RAM:28992
346 : Info   : INIT : Exception #108 Last Action before Reboot: Background Task Last systime: 1624562651 - Restart Reason: Exception
347 : Info   : FS   : Mounting...
384 : Info   : FS   : Mount successful, used 83332 bytes of 957314
415 : Info   : CRC  : SecuritySettings CRC   ...OK
420 : Info   : WiFi : Start network scan all channels
513 : Info   : INIT : Free RAM:25720
514 : Info   : INIT : I2C
515 : Info   : INIT : SPI not enabled
--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Exception (28):
epc1=0x4021d415 epc2=0x00000000 epc3=0x00000000 excvaddr=0x13840000 depc=0x00000000

LoadProhibited: A load referenced a page mapped with an attribute that does not permit loads
  epc1=0x4021d415 in CPlugin_003(CPlugin::Function, EventStruct*, String&) at ??:?

>>>stack>>>

ctx: cont
sp: 3fff2d60 end: 3fff30c0 offset: 0190
3fff2ef0:  00000054 00000054 3ffe8760 40101713
3fff2f00:  00000000 00000012 3fff1d2c 3fff3020
3fff2f10:  4026cb80 00000020 3fff3cfc 40101b16  
3fff2f20:  00000001 00000028 00000002 4023b2c9
3fff2f30:  00000051 00000051 3ffe8760 40101713
3fff2f40:  00000000 0012001f 00000000 3fff3020
3fff2f50:  00006238 0007ec0b 3fff2fe0 40101b16  
3fff2f60:  00000000 00000002 3fff2fe0 3fff3020
3fff2f70:  3fff0504 00000002 3fff2f80 4023b480 <
3fff2f80:  00000000 3fff3bf2 00000000 00000000
3fff2f90:  00000030 00000000 00000000 40101713
3fff2fa0:  00000000 00000000 00000002 00000000
3fff2fb0:  00000000 3fff0520 00000000 00000000
3fff2fc0:  00000000 00000000 00000000 00000000
3fff2fd0:  00000000 00000000 03030c00 4000ff00
3fff2fe0:  00000000 000e000f 00000000 0000000d
3fff2ff0:  0000000a 3fff051c 3fff3ccc 4023b65e
3fff3000:  4026cb80 3fff3cb4 4bc6a7f0 00000001
3fff3010:  3fff04d8 3fff04b4 3fff04dc 4023b5a9
3fff3020:  00000000 00000020 00000000 00000001  
3fff3030:  3fff04d8 3fff04b4 3fff04dc 4021c480
3fff3040:  3fff06c8 3fff17f8 3fff1288 4023a81c
3fff3050:  00000000 0015001f 00000000 00000000
3fff3060:  0075007f 00000000 feefeffe feefeffe
3fff3070:  feefeffe feefeffe feefeffe feefeffe
3fff3080:  feefeffe feefeffe feefeffe 3fff310c
3fff3090:  3fffdad0 00000000 3fff30f0 40211e80
3fff30a0:  feefeffe feefeffe feefeffe 4027166f
3fff30b0:  feefeffe feefeffe 3ffe875c 40100689
<<<stack<<<

0x40101713 in umm_free_core at umm_malloc.cpp:?
0x4026cb80 in EspClass::getFreeHeap() at ??:?
0x40101b16 in free at ??:?
0x4023b2c9 in CPluginCall(unsigned char, CPlugin::Function, EventStruct*, String&) at ??:?
0x40101713 in umm_free_core at umm_malloc.cpp:?
0x40101b16 in free at ??:?
0x40101713 in umm_free_core at umm_malloc.cpp:?
0x4023b65e in addCPlugin(unsigned char, unsigned char) at ??:?
0x4026cb80 in EspClass::getFreeHeap() at ??:?
0x4023b5a9 in CPluginCall(CPlugin::Function, EventStruct*) at ??:?
0x4021c480 in CPluginInit() at ??:?
0x4023a81c in ESPEasy_setup() at ??:?
0x40211e80 in setup at ??:?
0x4027166f in loop_wrapper() at core_esp8266_main.cpp:?
0x40100689 in cont_wrapper at ??:?

--------------- CUT HERE FOR EXCEPTION DECODER ---------------
```

Not sure if it is any usable, but just to be sure I saved the [.pio build output](https://www.dropbox.com/s/7tu8z2nk7fm2ctr/normal_300_ESP8266_4M1M.rar?dl=0)
It is essentially built on this source: https://github.com/letscontrolit/ESPEasy/pull/3680/commits/c934304871e8c1dcdfca3379ebbabef75640a911
I was testing something else that may be put in this 2nd heap and that caused too many complications, so I just reverted the code to this last commit and cleaned + rebuild.
And to make matters worse, the previous build I had running (on that PR code) was running stable for hours but this build just keeps crashing with the crash I showed above.

earlephilhower, 2021-06-24:
That's a different signature than your original report.  Looks like random memory corruption and not String related here.  Address `0x13840000` is definitely not valid and not in IRAM or the IRAM-heap in any way.

TD-er, 2021-06-24:
Yep it is indeed a different signature and the more I think of it, I guess it may indeed be some kind of corruption.
All previous fixes I made were fixable by wrapping the flash string in a `String()` constructor though.
So I guess this crash dump (which just happened as I read your post, so I immediately copy/pasted it) may confuse the issue here.

TD-er, 2021-06-25:
Do you have a list of what addresses can be considered valid?

```
 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 3460, room 16
tail 4
chksum 0xcc
load 0x3fff20b8, len 40, room 4
tail 4
chksum 0xc9
csum 0xc9
v000ee2b0
~ld
�U348 : Info   : 

INIT : Booting version: 'feature/build_core_3_0_0_c934304' (ESP82xx Core 04c23227, NONOS SDK 3.0.0-dev(c0f7b44), LWIP: 2.1.2 PUYA support)
349 : Info   : INIT : Free RAM:29048
351 : Info   : INIT : Exception #1388 - Restart Reason: Exception
352 : Info   : FS   : Mounting...
391 : Info   : FS   : Mount successful, used 83332 bytes of 957314
420 : Info   : CRC  : SecuritySettings CRC   ...OK
425 : Info   : WiFi : Start network scan all channels
503 : Info   : INIT : Free RAM:26456
504 : Info   : INIT : I2C
504 : Info   : INIT : SPI not enabled
929 : Info   : INFO : Plugins: 77 [Normal] [Testing] [TEST_A] (ESP82xx Core 04c23227, NONOS SDK 3.0.0-dev(c0f7b44), LWIP: 2.1.2 PUYA support)
931 : Info   : EVENT: System#Wake
1086 : Info   : Webserver: start
1089 : Info   : Time set to 1624606468.000
1092 : Info   : Current Time Zone:  DST time start: 2021-03-28 02:00:00 offset: 120 min STD time start: 2021-10-31 03:00:00 offset: 60 min
1096 : Info   : Local time: 2021-06-25 09:34:28
1098 : Info   : EVENT: System#Boot

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Exception (28):
epc1=0x40293b65 epc2=0x00000000 epc3=0x4024910d excvaddr=0x0c5c0000 depc=0x00000000

LoadProhibited: A load referenced a page mapped with an attribute that does not permit loads
  epc1=0x40293b65 in pgm_read_byte_inlined at /workdir/repo/newlib/newlib/libc/sys/xtensa/sys/pgmspace.h:100
  epc3=0x4024910d in rulesProcessingFile(String const&, String const&) at ??:?

>>>stack>>>

ctx: cont
sp: 3fff2da0 end: 3fff3120 offset: 0190
3fff2f30:  402dbab0 0000000e 3fff2f88 4027eca8
3fff2f40:  3fff302c 00000000 3fff2f88 4027fa8d
3fff2f50:  0c5c0000 3fff300c 3fff302d 4027eddd
3fff2f60:  3fff302f 3fff300c 3fff302d 40248ae6
3fff2f70:  00000000 000e006e 00000000 00000000
3fff2f80:  3fff739c 00000000 3fff739c 000e000f  
3fff2f90:  80000000 00000002 3fff300c 4027ed34
3fff2fa0:  3fff3028 3fff30b0 3fff30b0 00000001
3fff2fb0:  0000000e 3fff300c 3fff3018 3fff3024
3fff2fc0:  3fff308c 3fff2fe0 3fff3028 40249099
3fff2fd0:  3fff3024 3fff302d 3fff302c 3fff302c
3fff2fe0:  4028dc40 00000000 000003e8 40252e76
3fff2ff0:  00000000 3fff7330 3fff7324 40282510
3fff3000:  00000000 00000000 3fff35a0 3fff73b4
3fff3010:  000e000f 80000000 3fff741c 000e000f
3fff3020:  80000000 3fff3050 00000002 01010000
3fff3030:  402d5501 00000001 3fff3080 4027f33d
3fff3040:  00000024 3fff76c4 00000001 3fff30b0
3fff3050:  00000040 3fff302c 3fff302d 00000001
3fff3060:  3fff0800 00000000 3fff3080 00000001
3fff3070:  3fff0800 3fff30b0 00000000 402491c7
3fff3080:  656c7572 742e3173 0a007478 4027eff4
3fff3090:  3fff0800 3ffe878c 3fff18bc 4027f01d  
[... 71 lines cut ...]
3fff3040:  00000000 00000000 00000001 00000000
3fff3050:  00000000 3fff4044 00000000 00000000
3fff3060:  00000000 00000000 00000000 00000000
3fff3070:  00000000 00000000 03030c00 00000001
3fff3080:  00000000 3fff1274 3fff1214 4022d5e6  
3fff3090:  3fff0800 3fff1470 00000000 4022d469  
3fff30a0:  3fff0800 3fff1470 3fff18bc 4024d230
3fff30b0:  00000000 0015001f 00000000 00000000
3fff30c0:  0030003f 00000000 feefeffe feefeffe  
3fff30d0:  feefeffe feefeffe feefeffe feefeffe
3fff30e0:  feefeffe feefeffe feefeffe 3fff316c
3fff30f0:  3fffdad0 00000000 3fff3150 4021d7f8
3fff3100:  feefeffe feefeffe feefeffe 40280517
3fff3110:  feefeffe feefeffe 3ffe87d8 40100689
<<<stack<<<

0x4022d5e6 in PluginInit() at ??:?
0x4022d469 in NPluginInit() at ??:?
0x4024d230 in ESPEasy_setup() at ??:?
0x4021d7f8 in setup at ??:?
0x40280517 in loop_wrapper() at core_esp8266_main.cpp:?
0x40100689 in cont_wrapper at ??:?

--------------- CUT HERE FOR EXCEPTION DECODER ---------------
```

TD-er, 2021-06-25:
OK, it does not seem to be related to the 2nd heap, but more something in the current staging code as I now also have it happening when using this PIO environment:

```
[core_stage]
extends                   = esp82xx_3_0_x
platform                  = https://github.com/platformio/platform-espressif8266.git
build_flags               = ${esp82xx_3_0_x.build_flags}
                            -DPIO_FRAMEWORK_ARDUINO_ESPRESSIF_SDK22x_191122
                            -DPHASE_LOCKED_WAVEFORM
                            

platform_packages = 
    platformio/framework-arduinoespressif8266 @ https://github.com/esp8266/Arduino.git
    mcspr/toolchain-xtensa @ 5.100200.201223
```

```

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 3460, room 16
tail 4
chksum 0xcc
load 0x3fff20b8, len 40, room 4
tail 4
chksum 0xc9
csum 0xc9
v000ec990
~ld
�cl`s�ol�U328 : Info   :

INIT : Booting version: 'feature/build_core_3_0_0_247db6e' (ESP82xx Core d99dd5d3, NONOS SDK 2.2.2-dev(a58da79), LWIP: 2.1.2 PUYA support)
329 : Info   : INIT : Free RAM:29240
330 : Info   : INIT : Exception #108 - Restart Reason: Exception
331 : Info   : FS   : Mounting...
356 : Info   : FS   : Mount successful, used 83332 bytes of 957314
379 : Info   : CRC  : SecuritySettings CRC   ...OK
381 : Info   : WiFi : Start network scan all channels
440 : Info   : INIT : Free RAM:26736
441 : Info   : INIT : I2C
441 : Info   : INIT : SPI not enabled
707 : Info   : INFO : Plugins: 77 [Normal] [Testing] [TEST_A] (ESP82xx Core d99dd5d3, NONOS SDK 2.2.2-dev(a58da79), LWIP: 2.1.2 PUYA support)
708 : Info   : EVENT: System#Wake
777 : Info   : Webserver: start
779 : Info   : Time set to 1624616479.000
781 : Info   : Current Time Zone:  DST time start: 2021-03-28 02:00:00 offset: 120 min STD time start: 2021-10-31 03:00:00 offset: 60 min
784 : Info   : Local time: 2021-06-25 12:21:19
786 : Info   : EVENT: System#Boot
790 : Info   : ACT  : Monitor,GPIO,2
793 : Info   : GPIO port #2: added to monitor list.
796 : Info   : ACT  : Monitor,GPIO,16
798 : Info   : GPIO port #16: added to monitor list.
800 : Info   : ACT  : Monitor,GPIO,12
802 : Info   : GPIO port #12: added to monitor list.
804 : Info   : ACT  : Monitor,GPIO,13
808 : Info   : GPIO port #13: added to monitor list.
811 : Info   : ACT  : Monitor,GPIO,14
813 : Info   : GPIO port #14: added to monitor list.
873 : Info   : SYS  : 0.00,20104.00,100.00,0.00
940 : Info   : EVENT: GPIO#2=0
982 : Info   : EVENT: WiFi#APmodeDisabled
1068 : Info   : EVENT: TaskInit#test=1,1
1157 : Info   : EVENT: Time#Initialized
1243 : Info   : EVENT: test#All=0.00,20104.00,100.00,0.00
1597 : Info   : SYS  : 0.00,19616.00,100.00,0.00
1653 : Info   : EVENT: test#All=0.00,19616.00,100.00,0.00
1716 : Info   : EVENT: Clock#Time=Fri,12:21
1755 : Info   : ACT  : logentry,"12:21:19 < 07:00:00"
1757 : Debug  : Command: logentry
1758 : Info   : 12:21:19 < 07:00:00
[... 110 lines cut ...]
0x40244c83 in Web_StreamingBuffer::startStream() at ??:?
0x40279ea0 in handle_sysinfo() at ??:?
0x4028bc58 in std::_Function_handler<void (String&, String&, String const&, int, int, int, int), esp8266webserver::storeArgHandler<WiFiServer> >::_M_manager(std::_Any_data&, std::_Any_data const&, std::_Manager_operation) at ??:?
0x40206655 in WiFiClient::flush(unsigned int) at ??:?
0x40206668 in WiFiClient::flush() at ??:?
0x4027f5b4 in String::equals(String const&) const at ??:?
0x4028bb9e in std::_Function_handler<void (), void (*)()>::_M_invoke(std::_Any_data const&) at ??:?
0x401000e1 in std::function<void ()>::operator()() const at ??:?
0x40272365 in esp8266webserver::FunctionRequestHandler<WiFiServer>::handle(esp8266webserver::ESP8266WebServerTemplate<WiFiServer>&, HTTPMethod, String const&) at ??:?
0x402043be in ESP8266WiFiGenericClass::setOutputPower(float) at ??:?
0x4024c85c in esp8266webserver::ESP8266WebServerTemplate<WiFiServer>::_handleRequest() at ??:?
0x402c4a9d in ic_get_rssi at ??:?
0x4024935e in SetWiFiTXpower(float, float) at ??:?
0x40100e65 in millis at ??:?
0x4024ca36 in esp8266webserver::ESP8266WebServerTemplate<WiFiServer>::handleClient() at ??:?
0x4028d4f0 in spiffs_cache_page_get_by_fd at ??:?
0x4024cb34 in backgroundtasks() at ??:?
0x4024cd04 in ESPEasy_loop() at ??:?
0x4010072c in ets_post at ??:?
0x4021d748 in loop at ??:?
0x4028044b in loop_wrapper() at core_esp8266_main.cpp:?
0x40100689 in cont_wrapper at ??:?

--------------- CUT HERE FOR EXCEPTION DECODER ---------------
```

TD-er, 2021-06-25:
@earlephilhower I just posted some thoughts too in this issue: https://github.com/platformio/platform-espressif8266/issues/252#issuecomment-868421793

I'm not entirely sure what is causing this, but for sure there is something wrong in how the code is being built.
It does seem to be related to the latest code in esp8266/Arduino and how PIO is building it.
And the flash string related code does pop up in the crash stack traces quite a lot, but not exclusively.

earlephilhower, 2021-06-25:
@TD-er I'm not sure what to say.  It looks to me like a runtime data corruption issue, not al linker one.  You've posted quite a few traces (please include the EPC= and EXECVADDR line or it's kind of useless) with varying PCs.

All the execvaddr that's been posted are fishy and of the form 0xAAAA0000 where the AAAA part is random garbage and the lower 16bits are 0.  My first impression is that you've got something in your code that's writing  past the end of its buffer somewhere and due to the way the heap/stack are now it's become fatal.

W/o a consistent failure and MCVE, there's not much we can do here, unfortunately.

mcspr, 2021-06-25:
> ... It looks to me like a runtime data corruption issue, ...

Having dealt with a oob array access (with vector operator[], where it also does not check the bounds), adding -DDEBUG_ESP_OOM and -DUMM_POISON_CHECK to the Core build flags is a pretty good way to find the root cause, since it will crash at a much earlier time when the offending allocation happens, and not some time later when it happens to create something else in that memory region and show nonsense stack traces.

With PIO, it is build_flags = ..., Arduino IDE has an option in the menu for OOM debug

d-a-v, 2021-06-25:
Given this new page, https://arduino-esp8266.readthedocs.io/en/latest/ideoptions.html
Would it be interesting to add the equivalent core defines and related PIO defines, for a reference ?
Or if there's already a PIO page mentioning these options, it would be nice to add it to this local documentation.

> to the Core build flags is a pretty good way to find the root cause

`-fstack-protector` is also a good option available in the arduino IDE that I'm not sure whether it is available for PIO (I can't see it in the pio builder script).

@valeros

TD-er, 2021-06-25:
@mcspr Just had pointed out something in my code which could cause some excessive allocations. I am testing it now, so maybe that would make these crashes a lot less likely... (fingers crossed)

See: https://github.com/letscontrolit/ESPEasy/issues/3693

And if there is something to help me debug these OOM issues or stack overflows, then I'm all ears.
Would save me a lot of time and frustrated users.

TD-er, 2021-06-25:
Nope... still happening, but the web server does respond even faster than it did before, so that issue at least did help a bit, but not in stability.

```

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Exception (28):
epc1=0x4029396c epc2=0x00000000 epc3=0x4025ca1a excvaddr=0x0f440000 depc=0x00000000

LoadProhibited: A load referenced a page mapped with an attribute that does not permit loads
  epc1=0x4029396c in strnlen_P at /workdir/repo/newlib/newlib/libc/sys/xtensa/string_pgmspace.c:51
  epc3=0x4025ca1a in htmlEscape(String&, char) at ??:?

>>>stack>>>

ctx: cont
sp: 3fff2b80 end: 3fff3050 offset: 0190
3fff2d10:  4027b83e 402da7a5 3fff2d40 4026d0f1
3fff2d20:  00000001 0f440000 3fff2d68 4027ee33
3fff2d30:  000008cd 402e1e98 402e1e98 4027ee65
3fff2d40:  00000001 3fff2d80 00000001 4026ec00
3fff2d50:  00000001 3fff2d80 00000001 4026d188
3fff2d60:  00000000 00000099 00000000 4027ee65
3fff2d70:  00000000 00000000 00000001 00000000  
3fff2d80:  00000046 00000000 00000000 4026cb1b
3fff2d90:  006a007f 00000000 00000000 4026ec6c
3fff2da0:  00000a1e 00000a1e 3ffe87dc 40274bb2
3fff2db0:  402de9ab 00000007 3fff2e1c 0000018c
3fff2dc0:  40205495 3fff2e00 3fff2e48 401017aa
3fff2dd0:  730aa8c0 00ffffff fe0aa8c0 4027e931
3fff2de0:  00000000 00000183 3fff2e00 40247036
3fff2df0:  4028da0c 00000000 3fff2e30 4027eaf0
3fff2e00:  00000001 3fff2014 3fff2e30 4027ebfc
3fff2e10:  00000001 3fff2e50 00000000 4027ec25
3fff2e20:  4028d3f8 00000000 00000000 40272a40
3fff2e30:  00000000 00000005 00000000 4027eb7c
3fff2e40:  00000001 00000000 00000000 40272aa0
3fff2e50:  00000000 00000000 00000001 4027ee65
3fff2e60:  3ffe868c 402dcc45 00000000 40272b6c
3fff2e70:  00000000 00650078 00000000 00000000
3fff2e80:  00000001 00000000 00000000 4027b285
3fff2e90:  00000000 ffffffff 00000000 4028148e
3fff2ea0:  00077250 00000b68 3fff748c 402060ee
3fff2eb0:  0000012c 3fff16e8 ffffffff 4024b6ad
3fff2ec0:  0000000a 3fff170c 4028bb60 00000000
3fff2ed0:  3fff170c 00000000 3fff16e8 40206655
3fff2ee0:  3fff170c 00000000 3fff2f00 40206668
3fff2ef0:  3fff1730 3fff6398 00000000 4027f4c0
3fff2f00:  00000000 69746365 00000000 3fff309c
3fff2f10:  3ffe868c 00000001 3fff641c 4028baa6
3fff2f20:  00637067 00000000 00000000 401000e1
3fff2f30:  00000000 3fff0748 00000000 40272281
3fff2f40:  402043be 3fff1730 3fff16e8 4024c7b8
3fff2f50:  402c49a5 00000000 00000000 402492ba
3fff2f60:  00000000 00000000 3fff05e0 00000000
3fff2f70:  00000000 73656369 00000000 00000000
3fff2f80:  40205840 00000000 00000000 69746365
3fff2f90:  00000000 3fff05e0 00000000 402492ee  
3fff2fa0:  40282c20 ffffffff 00000000 00000001
3fff2fb0:  3fff47e4 00000000 3fff1ec0 3fff309c
3fff2fc0:  3ffe868c 3fff170c 3fff16e8 4024c992
3fff2fd0:  3ffe868c 00000000 00000000 3fff309c
3fff2fe0:  3ffe868c 00000000 3fff05e1 3fff309c
3fff2ff0:  3ffe868c 00000000 3fff05e1 4024ca90
3fff3000:  3ffe868c 00000000 00000000 4024cc60
3fff3010:  00000000 00000000 00000000 4010072c
3fff3020:  3fffdad0 00000000 3fff3080 4021d744
3fff3030:  feefeffe feefeffe feefeffe 40280357
3fff3040:  feefeffe feefeffe 3ffe87d8 40100689
<<<stack<<<
[... 61 lines cut ...]
tail 4
chksum 0xcc
load 0x3fff20b8, len 40, room 4
tail 4
chksum 0xc9
csum 0xc9
v000ec8d0
~ld
�cd`r�nl�U325 : Info   :

INIT : Booting version: 'feature/build_core_3_0_0_c3e684a' (ESP82xx Core d99dd5d3, NONOS SDK 2.2.2-dev(a58da79), LWIP: 2.1.2 PUYA support)
326 : Info   : INIT : Free RAM:29240
328 : Info   : INIT : Exception #2 - Restart Reason: Exception
328 : Info   : FS   : Mounting...
357 : Info   : FS   : Mount successful, used 83332 bytes of 957314
381 : Info   : CRC  : SecuritySettings CRC   ...OK
384 : Info   : WiFi : Start network scan all channels
443 : Info   : INIT : Free RAM:26664
444 : Info   : INIT : I2C
444 : Info   : INIT : SPI not enabled
705 : Info   : INFO : Plugins: 77 [Normal] [Testing] [TEST_A] (ESP82xx Core d99dd5d3, NONOS SDK 2.2.2-dev(a58da79), LWIP: 2.1.2 PUYA support)
707 : Info   : EVENT: System#Wake
```

N.B. this is staging code, SDK222 and thus no 2nd heap or other MMU tweaks.

mcspr, 2021-06-25:
> `-fstack-protector` is also a good option available in the arduino IDE that I'm not sure whether it is available for PIO (I can't see it in the pio builder script).

@d-a-v see https://docs.platformio.org/en/latest/platforms/espressif8266.html#debug-level, PIO's platformio.ini is a project-level config file that can change the build flags for Core + libs (build_flags) or project files (src_build_flags), so we only need to know which flags to use for what purpose

paulocsanz, 2021-06-27:
I don't know if it's related, but we noticed some of our usage of PROGMEM strings break after 3.0.

From the top of my head __PRETTY_FUNCTION__ used to work as char*, but now is PROGMEM. Also we used to store header names as PROGMEM string and this stopped working in 3.0 (I opened a PR for one case, but I'm pretty sure it happens in other locations).

It produced a few hard to debug crashes.

TD-er, 2021-06-27:
@paulocsanz Do you have some concrete example of code that's now causing issues?

paulocsanz, 2021-06-27:
@TD-er

paulocsanz, 2021-06-27:
@TD-er This line used to work, https://github.com/internet-of-plants/embedded/blob/35cec6755d951f1aa560024f6e5c0060c74141c3/include/core/string/view.hpp#L12, In 3.0 we replaced it for iop::StaticString (wraps const __FlashStringHelper* for type safety).

Don't mind the noise, this line basically creates a string_view of a char*. But maybe it always was PROGMEM data and our use wasn't accessing it in a unaligned way.

Another example was https://github.com/internet-of-plants/embedded/blob/35cec6755d951f1aa560024f6e5c0060c74141c3/src/core/network.cpp#L175, which can't actually be PROGMEM string, Idk if it never was supposed to, and the method just didn't access it unaligned, but it started crashing after the 3.0 update.

TD-er, 2021-06-28:
OK, I've squashed a number of bugs including a rather fishy assembly implementation of a mutex I used. (found by @mcspr )
But I still have some issues with crashes.

For example:
```

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Exception (28):
epc1=0x4025f1ef epc2=0x00000000 epc3=0x00000000 excvaddr=0x06c303e0 depc=0x00000000

LoadProhibited: A load referenced a page mapped with an attribute that does not permit loads
  epc1=0x4025f1ef in findTaskIndexByName(String const&) at ??:?

>>>stack>>>

ctx: cont
sp: 3fff2a90 end: 3fff3120 offset: 0190
3fff2c20:  00000014 3fff2cf0 3fff2d70 4025f1d4
3fff2c30:  00000014 402d37ac 3fff2c60 4027f450
3fff2c40:  00000014 3fff2d7c 3fff2d70 00000000  
3fff2c50:  00000014 3fff2d7c 3fff2d70 4025f465
3fff2c60:  00000000 00006e69 00000000 00000073
3fff2c70:  3fff5b0c 3fff6a87 0000006e 3fff2d70
3fff2c80:  3fff6d64 3fff6a64 00000000 00000030
3fff2c90:  2e676966 00746164 00000000 00000000
3fff2ca0:  00000000 00000020 00000020 40101a6f
3fff2cb0:  00000001 00002000 3fff2d70 40101aa8
3fff2cc0:  00000000 00000050 00000000 40101a6f
3fff2cd0:  3fff5174 00000008 00000000 40101aa8
3fff2ce0:  00000000 69747075 4000656d 06000000
3fff2cf0:  74736574 40103e00 04000000 0000000c
3fff2d00:  00000000 3fff5a18 3fff5a08 00000000
3fff2d10:  00000000 00000000 3fff5aa4 40101a32
3fff2d20:  3fff6a74 3fff2d7c 3fff2d7c 00000000
3fff2d30:  3fff6a64 3fff6d64 3fff2d70 4025f5d4
3fff2d40:  3fff6a64 3fff6d64 3fff2d70 40268da3
3fff2d50:  3fff5280 3fff6d64 3fff6964 40268e83
3fff2d60:  00000000 00000000 0000001f 4010072c
3fff2d70:  3fff5aa4 0000001f 80000000 3fff5afc
3fff2d80:  0012001f 80000000 7fffffff 00000000
3fff2d90:  3fff2fb0 0000000c 00040000 ffffffff
3fff2da0:  3ffeec50 40103fce 00000000 00000110  
3fff2db0:  00000000 00000000 00000000 3fff3020
3fff2dc0:  00000001 3fff6964 3fff2fb0 40217eb9
3fff2dd0:  3fff2de0 3fff2f90 00000000 00000030
3fff2de0:  40243197 00000030 0000001e ffffffff
3fff2df0:  4024efab 3fff2ee0 00000000 ffffff00
3fff2e00:  00000000 00000000 00000000 fffffffe
3fff2e10:  000000d0 3fffc6fc 0000002e 3fff2fb0
3fff2e20:  00000003 3fff3020 00000000 00000030
3fff2e30:  4028e09c 00000000 000003e8 4025328e
3fff2e40:  00000000 00000000 00000000 4028296c
3fff2e50:  00000000 00000000 3fff35a0 00000000
3fff2e60:  007a1200 6edb0ad5 00000000 642e6769
3fff2e70:  00000000 00000000 3fff2ef0 4027f104
3fff2e80:  00000000 00000000 00000000 00000020
3fff2e90:  00000000 00000000 3fff2f10 4027f104
3fff2ea0:  3fff3020 3fff2fe0 3fff2f10 4027f210
3fff2eb0:  3fff2f24 3fff2fb0 3fff2ee0 402430e4
3fff2ec0:  3fff3020 00000003 3fff2fb0 00000020
3fff2ed0:  3fff3020 00000003 3fff2fb0 4024f36c
3fff2ee0:  00000000 00000000 00000000 00000000
3fff2ef0:  00000000 00000000 00000000 00000000
3fff2f00:  00000000 00000000 00000000 00000000
3fff2f10:  00000000 00000000 00000000 00000000
3fff2f20:  00000000 00000000 00000000 00000000
3fff2f30:  00000000 00000000 03030000 3f000000
3fff2f40:  3fff2f3c 00000000 3fff2f48 3fff2f48  
3fff2f50:  00000000 3fff2f54 3fff2f54 00000000
[... 75 lines cut ...]

load 0x4010f000, len 3460, room 16
tail 4
chksum 0xcc
load 0x3fff20b8, len 40, room 4
tail 4
chksum 0xc9
csum 0xc9
v000ec950
~ld
�U343 : Info   : 

INIT : Booting version: 'feature/build_core_3_0_0_d27b2a8' (ESP82xx Core 421d02ee, NONOS SDK 3.0.0-dev(c0f7b44), LWIP: 2.1.2 PUYA support)
345 : Info   : INIT : Free RAM:29048
346 : Info   : INIT : Exception #124 - Restart Reason: Exception
347 : Info   : FS   : Mounting...
380 : Info   : FS   : Mount successful, used 78061 bytes of 957314
414 : Info   : CRC  : SecuritySettings CRC   ...OK
418 : Info   : WiFi : Start network scan all channels
515 : Info   : INIT : Free RAM:26544
516 : Info   : INIT : I2C
```

This function is called with a `const String&` argument and this is used in a `std::map<String, ...>` object in calling `find`
I can of course enable the DRAM stack for these and wrap the `String` argument in a `String()` constructor, but that's rather trial and error to see if I found all.

Gizmodo, 2021-06-28:
@TD-er  Please, give me a code to try it on my ESP8266
Few days ago I also faced with "A load referenced a page mapped with an" and after long investigations - i've got code error maid by myself.

TD-er, 2021-06-28:
Not tried it (as I'm not behind a computer with access to a compiler), but it should be something like this:

```c++
typedef std::map<String, int> myMap_t;

myMap_t myMap;

void crashtest() {
  {
    HeapSelectDram ephemeral;
    myMap[F("bla")] = 1;
  }    
  {
    HeapSelectIram ephemeral;
    auto it= myMap.find(F("bla"));
    if (it != myMap.end()) // Make sure the compiler doesn't remove the find call
      myMap[F("bla2")] = 1;
  }    
  {
    HeapSelectDram ephemeral;
    auto it= myMap.find(F("bla2"));
    if (it != myMap.end()) // Make sure the compiler doesn't remove the find call
      myMap[F("bla")] = 1;
  }    
}
```

Gizmodo, 2021-06-28:
It works without errors

TD-er, 2021-06-30:
I was wondering as it seems that this is crashing seemingly random, if it may be related to this build flag:
```
PSTR_ALIGN=1
```

`strlen_P` is an alias for `strnlen_P` with the size parameter set to irrelevant.
So even though the `strnlen_P` is not found in my code, it surely is being executed. (a lot!)
How is this function trying to read the flash string?
Reading 32 bits at once and then trying to read the separate bytes may read a chunk which is outside the range assigned to these flash strings. (either at the beginning or the end)
I can imagine it may not know how long a string is, so reading until it reads a `0` may defeat the purpose of the intended speedup. So if my theory holds, it is probably crashing at the end.

closed 2021-06-30 by earlephilhower