esp8266/Arduino #8171: Watchdog triggered with Adafruit_ST7735 library

Vinicius-FF, 2021-06-24:
### Basic Infos

- [x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x] I have tested that the issue is present in current master branch (aka latest git).
- [x] I have searched the issue tracker for a similar issue.
- [x] If there is a stack dump, I have decoded it.
- [x] I have filled out all fields below.

#### Platform

- Hardware:					[ESP-12]
- Core Version:			[3.0.0]
- Development Env:		[Arduino IDE]
- Operating System:		[Windows]

### Settings in IDE

- Module:  			[Nodemcu 1.0 (ESP12-E Module)]
- Flash Mode:			[Arduino IDE board manager default (couldn't find the information)]
- Flash Size: 		[4MB]
- lwip Variant:   [v2 Lower Memory]
- Reset Method:		[nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz]
- Upload Using:		[SERIAL]
- Upload Speed:   [115200] (serial upload only)

### Problem Description

Soft WDT is triggered when using the Adafruit ST7735 and ST7789 Library (1.7.3, latest available on library manager) when using core 3.0.0
With core 2.7.4 it works fine.

Using a ST7735 display.

Decoding stack results:
```
0x40202f4d: Adafruit_SPITFT::writeColor(unsigned short, unsigned int) at C:\Users\Vinicius\Documents\Arduino\libraries\Adafruit_GFX_Library\Adafruit_SPITFT.cpp line 1333
0x40202d29: Adafruit_SPITFT::writeCommand(unsigned char) at C:\Users\Vinicius\Documents\Arduino\libraries\Adafruit_GFX_Library\Adafruit_SPITFT.cpp line 2093
0x40202fcd: Adafruit_SPITFT::writeFillRectPreclipped(short, short, short, short, unsigned short) at C:\Users\Vinicius\Documents\Arduino\libraries\Adafruit_GFX_Library\Adafruit_SPITFT.cpp line 1566
0x40203225: Adafruit_SPITFT::fillRect(short, short, short, short, unsigned short) at C:\Users\Vinicius\Documents\Arduino\libraries\Adafruit_GFX_Library\Adafruit_SPITFT.cpp line 1650
0x40207b1e: Adafruit_GFX::fillScreen(unsigned short) at C:\Users\Vinicius\Documents\Arduino\libraries\Adafruit_GFX_Library\Adafruit_GFX.cpp line 317
0x40201313: tftPrintTest() at C:\Users\Vinicius\Documents\Arduino\libraries\Adafruit_GFX_Library/Adafruit_GFX.h line 141
0x40205739: __delay(unsigned long) at C:\Users\Vinicius\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.0\cores\esp8266\core_esp8266_wiring.cpp line 55
0x4020191d: setup() at C:\Users\Vinicius\AppData\Local\Temp\arduino_modified_sketch_657864/graphicstest.ino line 130
0x40204e98: loop_wrapper() at C:\Users\Vinicius\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.0\cores\esp8266\core_esp8266_main.cpp line 198
```

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

graphicstest example from the Adafruit ST7735 and ST7789 Library using this call:
```cpp
// For ST7735-based displays, we will use this call
Adafruit_ST7735 tft = Adafruit_ST7735(TFT_CS, TFT_DC, TFT_MOSI, TFT_SCLK, TFT_RST);
```
### Debug Messages

```
 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 3460, room 16 
tail 4
chksum 0xcc
load 0x3fff20b8, len 40, room 4 
tail 4
chksum 0xc9
csum 0xc9
v000487e0
~ld

SDK:2.2.2-dev(38a443e)/Core:3.0.0=30000000/lwIP:STABLE-2_1_2_RELEASE/glue:1.2-48-g7421258/BearSSL:c0b69df

```

Tech-TX, 2021-06-25:
Bodmer's library doesn't blow chunks...  I haven't used the Adafruit library since finding his improved version.

Vinicius-FF, 2021-06-25:
I'll give his libray a try, thanks for the reply @Tech-TX! 
Posted this issue because I managed to isolate the problem but couldn't find informations about it anywhere.
Now if someone has the same problem, at least they can find this issue report, along with your recomendation of the library and maybe scratch their head for less time than I did...

Also, since it worked with with 2.7.4, maybe some bug or something hidden, IDK.

dok-net, 2021-06-27:
Which version of Adafruit-GFX-Library are you running?

Vinicius-FF, 2021-06-28:
Version 1.10.10, the latest available on the Arduino IDE library manager, and IDE version 1.8.13. 
Will update to version 1.8.15, but I'm not sure that could be the issue. I'll report the result of the test with the 1.8.15 version of the Arduino IDE

dok-net, 2021-06-28:
@earlephilhower Can you explain how line 1333 of `Adafruit_SPITFT.cpp` from version 1.10.10 of `Adafruit_GFX_Library` can be a correct stack dump artifact, when the sources are:
```
1317 #if defined(ESP8266)
1318     do {
1319       uint32_t pixelsThisPass = len;
1320       if (pixelsThisPass > 50000)
1321         pixelsThisPass = 50000;
1322       len -= pixelsThisPass;
1323       yield(); // Periodic yield() on long fills
1324       while (pixelsThisPass--) {
1325         hwspi._spi->write(hi);
1326         hwspi._spi->write(lo);
1327       }
1328     } while (len);
1329 #else // !ESP8266
1330     while (len--) {
1331 #if defined(__AVR__)
1332       AVR_WRITESPI(hi);
1333       AVR_WRITESPI(lo);
1334 #elif defined(ESP32)
1335       hwspi._spi->write(hi);
1336       hwspi._spi->write(lo);
1337 #else
1338       hwspi._spi->transfer(hi);
1339       hwspi._spi->transfer(lo);
1340 #endif
1341     }
1342 #endif // end !ESP8266
```
I've given up on trusting the dump long ago :-(

TD-er, 2021-07-02:
@dok-net I have seen issues with line numbers if various line endings are mixed.

Anyway, this number of "50000" does seem rather "experimental based".
If the writing process is slightly lower, then you would hit the WDT timer.
Also I do remember to have seen changes to `yield()` lately. Can it still be used to feed the dog?

dok-net, 2021-07-02:
@TD-er Runs off 3.0.1 forever:
```
void setup()
{
	Serial.begin(74880);
	delay(500);
}

void loop()
{
	Serial.println();
	Serial.println("yyield");
	auto started = ESP.getCycleCount();
	for (;;)
	{
		const auto now = ESP.getCycleCount();
		if (now - started > ESP.getCpuFreqMHz() * 1000 * 1560)
		{
			yield();
			started = ESP.getCycleCount();
		}
	}
}
```
Commenting out the `yield();`, `Soft WDT reset`
Stack dump:
```
   Error
   0x40201068 esp_get_cycle_count
   (inlined by) EspClass
   :\Users\dok\Documents\Arduino\yyield/yyield.ino:11
   0x40201274 loop_wrapper()
   0x40100c29 cont_wrapper   
```

TD-er, 2021-07-02:
Thanks good to know :)

dok-net, 2021-07-02:
@Vinicius-FF @TD-er Now, if either of you has the hardware and an actual sketch to trigger the WDT - I surmise it's not the constructor snippet from above alone - could you insert some timing reports into the section in question and find out if it's the 50000 that is at fault? This is all becoming rather theoretical now :-)

Vinicius-FF, 2021-07-02:
@dok-net I can try, I have the hardware. But I'm rather rusty in programming, and my knoledge is more on the hardware side of things. 
One thing I just recalled is that among the test I've made (was troubleshooting project that we made last year and it stopped working after programming the same code with the newer version of the core), is that I've tried disabling the watchdog and then it hit the hardware watchdog instead.

dok-net, 2021-07-02:
@Vinicius-FF At the least you can tinker with the 50000 and see if reducing that fixes the WDT issue.