esp8266/Arduino #8183: -DNO_GLOBAL_INSTANCES breaks ESP8266HttpClient

paulocsanz, 2021-06-27:
### Basic Infos

- [X] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [X] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [X] I have tested that the issue is present in current master branch (aka latest git).
- [X] I have searched the issue tracker for a similar issue.
- [X] If there is a stack dump, I have decoded it.
- [X] I have filled out all fields below.

#### Platform

- Hardware:				ESP-12
- Core Version:			master
- Development Env:		Platformio
- Operating System:		Ubuntu

### Problem Description

Defining `NO_GLOBAL_INSTANCES` breaks `ESP8266HttpClient` because it depends on `devnull` being defined. But it seems that if it's needed, it shouldn't be disable by that option.

```cpp
/src/ESP8266HTTPClient.cpp:850:32: error: 'devnull' was not declared in this scope
  850 |         _client->sendAvailable(devnull); // clear _client's output (all of it, no timeout)
      |                                ^~~~~~~
*** [.pio/build/release/libc18/ESP8266HTTPClient/ESP8266HTTPClient.cpp.o] Error 1
```

paulocsanz, 2021-06-27:
`Serial` is always declared as external in the header, but not defined when NO_GLOBAL_INSTANCES is defined. But other globals aren't even declared in the headers if NO_GLOBAL_INSTANCES is defined.

Should it also not be defined in that case?

dok-net, 2021-07-04:
@paulocsanz Both `Serial` and `Serial1`, to be precise. Now while you are at it, there shouldn't be a problem enclosing that in `#ifdef`, given that your results indicate it passes the linker phase just fine. From my POV, go ahead and add the `#ifdef` to your PR.

PR #8184 (2021-07-17): Fix NO_GLOBAL_INSTANCES
Fixes #8183 

It's not clear to me if this is the best solution. This is just a first draft so the community can debate on a better solution.

But it seems a fairly simple thing to fix. But something needed, if something is used internally it should be declared + defined.

It's not clear to me if we should create it when `NO_GLOBAL_STREAMDEV` is specifically defined.
closed 2021-07-17 by earlephilhower