esp8266/Arduino #8193: ambigious definition of byte

nablabla, 2021-07-04:
disclaimer: If I cannot do anything, it is hard to get into this topic, so I have read not everything I am supposed to, I am sorry

### Basic Infos

- [i have no idea] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [this is a bit much sorry] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [i don't now how to do that since these files are buried in my file system] I have tested that the issue is present in current master branch (aka latest git).
- [I did] I have searched the issue tracker for a similar issue.
- [n/a] If there is a stack dump, I have decoded it.
- [ ] I have filled out all fields below.

#### Platform

- Hardware:					wemos [ESP-12-S
- Core Version:			[3.0.1
- Development Env:		Arduino IDE
- Operating System:		Windows 

### Settings in IDE

- Module:  			Wemos D1 mini r2 (or clone, not shure which is correct for me, the problem is in both)
- Flash Mode:			dout
- Flash Size: 		4MB
- lwip Variant:   v2 Lower Memory
- Reset Method:		i am blind or this does not exist
- Flash Frequency:	40Mhz
- CPU Frequency:		80Mhz
- Upload Using:		I can not find this settings either, but i am not doing OTA
- Upload Speed:   now i am really doubting my sanity, i can not find this setting either

### Problem Description

i get this error on compiling: reference to 'byte' is ambiguous.
It seems to me that there are two definitions of "byte" in hardware definition and tools. Only when using std namespace.

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp

using namespace std;
byte currentMode = 0x0;

void loop() {
currentMode++;
}

```

### Debug Messages

sketch_jul04b:2:1: error: reference to 'byte' is ambiguous
    2 | byte currentMode = 0x0;
      | ^~~~
In file included from c:\users\<<<username>>>\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\3.0.3-gcc10.3-9bcba0b\xtensa-lx106-elf\include\c++\10.3.0\cmath:42,
                 from c:\users\<<<username>>>\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\3.0.3-gcc10.3-9bcba0b\xtensa-lx106-elf\include\c++\10.3.0\math.h:36,
                 from C:\Users\<<<username>>>\AppData\Local\arduino15\packages\esp8266\hardware\esp8266\3.0.1\cores\esp8266/Arduino.h:34,
                 from sketch\sketch_jul04b.ino.cpp:1:
c:\users\<<<username>>>\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\3.0.3-gcc10.3-9bcba0b\xtensa-lx106-elf\include\c++\10.3.0\bits\cpp_type_traits.h:404:30: note: candidates are: 'enum class std::byte'
  404 |   enum class byte : unsigned char;
      |                              ^~~~
In file included from sketch\sketch_jul04b.ino.cpp:1:
C:\Users\<<<username>>>\AppData\Local\arduino15\packages\esp8266\hardware\esp8266\3.0.1\cores\esp8266/Arduino.h:160:17: note:                 'typedef uint8_t byte'
  160 | typedef uint8_t byte;
      |                 ^~~~
reference to 'byte' is ambiguous

TD-er, 2021-07-04:
Looks like it might be related to this pending PR: https://github.com/esp8266/Arduino/pull/8090

d-a-v, 2021-07-04:
`using namespace std;` is currently  incompatible with `byte` with c++17 and Arduino.

devyte, 2021-07-04:
Previous comment is correct. Don't do that, there are many reasons against `using namespace std`, including that old frameworks may clash with the STL. The Arduino reference is old. 
Closing due to user error.

closed 2021-07-04 by devyte