esp8266/Arduino #8197: ESP8266 shows up in DHCP list only when connected to internet

sh-user, 2021-07-07:
When I browse into the dhcp devices tab in my router I can't see the ESP8266 device with the IP it got from the router. ESP8266 shows up in DHCP list only when connected to internet and sending some data when connected. I think that the lease time for the IP address is not set, I do not have enough knowledge to check it.

Tech-TX, 2021-07-09:
That's an issue with your router,  not the core files or libraries.  My router shows the ESP as soon as the connection is established.

d-a-v, 2021-07-09:
@sh-user did you try with a smartphone acting as AP ?

sh-user, 2021-07-10:
Indeed, the problem is in the TP-LINK TL-WR841N v14 router, it displays the ESP8266 in the DHCP device list only when there is traffic, and then saves the IP address lease. For example, there is no such problem on a Tenda router, it displays the IP address immediately after connecting. The access point created by the smartphone only shows the MAC address.

closed 2021-07-10 by sh-user