esp8266/Arduino #8200: String::concat(char*, size_t) doesn't support non-null terminated views

paulocsanz, 2021-07-07:
### Basic Infos

- [X] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [X] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [X] I have tested that the issue is present in current master branch (aka latest git).
- [X] I have searched the issue tracker for a similar issue.
- [X] If there is a stack dump, I have decoded it.
- [ ] I have filled out all fields below.

### Problem Description

If a non-null terminated char* view (like `std::array<char, 32> md5` as storage) is passed to String::concat it currently crashes, because it's triggering UB by reading after the end of the buffer. This is the only way the current String API could allow adding a non-null terminated string to `String`, the rest all depend on strlen. So it's kinda important to the view usage.

mcspr, 2021-07-07:
> If a non-null terminated char* view (like std::array<char, 32> md5 as storage) is passed to String::concat it currently crashes

Could you share the specific crash condition though? idk if this has anything to do with nullchars, as we can read whatever with memcpy, *if* it is in the valid memory range? e.g. if array is at 0x40000000 - 32, reading past 0x40000000 will cause the crash.

paulocsanz, 2021-07-08:
I'm not sure, but I was doing something close to that, we don't use the stack at the 4kb unused sys stack, so we put some std::arrays in its place and the array that seemed to be the problem was the last one in the struct, but it didn't even reach 0x3FFFF000, so it may just be a misdiagnose of mine.

PR #8198 (2021-07-08): Avoid copying null in concat, unused + breaks views
We use a few non-null terminated views and it seems String::concat(char*, size_t) breaks those because it always copies the null terminator, but since the function immediately overwrites it, it doesn't need to.

This change makes String::concat support non-null terminated char* without any side effect.

Fixes #8200
closed 2021-07-08 by earlephilhower